espressif/arduino-esp32 #1747: wifi can not reach STA_CONNECTED randomly (wifi: Set status to INIT. )

lzw7603, 2018-08-11:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 DevKItC
Core Installation/update date:	        11/Aug/2018
IDE name:						Arduino 1.8.5(ESP32 Dev Module)
Flash Frequency:					80Mhz/DOUT
Upload Speed:					921600

### Description:
wifi can not connect randomly .

**when success:**
Connecting to SC
I (1029) wifi: wifi driver task: 3ffb7240, prio:23, stack:4096, core=0
I (2071) wifi: wifi firmware version: 4fd6a2c
I (2072) wifi: config NVS flash: enabled
I (2074) wifi: config nano formating: disabled
I (2089) wifi: Init dynamic tx buffer num: 32
I (2089) wifi: Init data frame dynamic rx buffer num: 10
I (2089) wifi: Init management frame dynamic rx buffer num: 10
I (2093) wifi: Init static rx buffer size: 1600
I (2097) wifi: Init static rx buffer num: 4
I (2101) wifi: Init dynamic rx buffer num: 10
I (2162) wifi: mode : sta (30:ae:a4:17:a7:d4)
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
I (3335) wifi: n:3 0, o:1 0, ap:255 255, sta:3 0, prof:1
.I (4068) wifi: state: init -> auth (b0)
I (4072) wifi: state: auth -> assoc (0)
I (4080) wifi: state: assoc -> run (10)
.I (4641) wifi: connected with SC, channel 3
I (4645) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
.[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.50.174, MASK: 255.255.252.0, GW: 192.168.51.254
.

**when failure:**
Connecting to SC
I (1028) wifi: wifi driver task: 3ffb7240, prio:23, stack:4096, core=0
I (2069) wifi: wifi firmware version: 4fd6a2c
I (2070) wifi: config NVS flash: enabled
I (2072) wifi: config nano formating: disabled
I (2087) wifi: Init dynamic tx buffer num: 32
I (2087) wifi: Init data frame dynamic rx buffer num: 10
I (2087) wifi: Init management frame dynamic rx buffer num: 10
I (2091) wifi: Init static rx buffer size: 1600
I (2095) wifi: Init static rx buffer num: 4
I (2099) wifi: Init dynamic rx buffer num: 10
I (2158) wifi: mode : sta (30:ae:a4:17:a7:d4)
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
I (3296) wifi: n:3 0, o:1 0, ap:255 255, sta:3 0, prof:1
.I (4029) wifi: state: init -> auth (b0)
I (4033) wifi: state: auth -> assoc (0)
E (4041) wifi: Set status to INIT.
I (4041) wifi: state: assoc -> init (11)
I (4041) wifi: n:3 0, o:3 0, ap:255 255, sta:3 0, prof:1
..............................................................................

### Sketch:
```cpp

#include <WiFi.h>

const char* ssid     = "SC";
const char* password = "SC12345678";

const char* host = "www.lesso.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void WiFiEvent(WiFiEvent_t event) {
    //Serial.printf("[WiFi-event] event: %d\n", event);
    
}

void setup()
{
    esp_log_level_set("wifi", ESP_LOG_VERBOSE); 
    Serial.setDebugOutput(true);
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.disconnect(true);
    delay(1000);
    WiFi.mode(WIFI_STA);
    delay(1000);
    
    WiFi.begin(ssid, password);

    int nTimes = 0;
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.println(WiFi.BSSIDstr());
}

int value = 0;

void loop()
{
    char buf[10];
    memset(buf, 0x00, 10 * sizeof(char));

    sprintf(buf, "wifi is %d", WiFi.status());
    
    Serial.println(buf);
    delay(5000);
}

dmk99, 2018-09-11:
Hi,

I've been encountering a similar issue where the device connects fine the first time and then after a reboot/restart it won't connect. I get the same statuses as you, when it fails it gets stuck on `WIFI_READY`. I think I've find a "solution", it's not ideal but it does seem to work at least in my case.

In `setup` do the following, the delays are at `1000 ms` but you can experiment with different delays. The main thing I think here is that I'm beginning a connection, disconnecting and then beginning another connection, this seems to get the WiFi chip into the correct state to then get an IP:

```
void setup(void) {
  Serial.begin(115200);

  WiFi.begin(ssid, password);
  delay(1000);
  WiFi.disconnect();
  delay(1000);
  WiFi.begin(ssid, password);
  delay(1000);
...
}
```

I'd be interested to know if this works for you.

bradshawlupton, 2019-02-08:
this worked for Bradshaw in Buzzards Bay!  thank you.

Imara90, 2019-05-01:
This did not work for me. The esp is currently stuck in immediately in:
3064 - WIFI CONNECTING: SSID 
E (6803) wifi: Set status to INIT.

Did anyone have any luck for restricting the set status to INIT at all? 
It will only go out of this state, if I completely flash new firmware on it.

closed 2019-08-14 by stale[bot]

phpdrill, 2020-12-24:
Had the same issue. The solution posted by dmk99 resolved the issue.