espressif/arduino-esp32 #1751: ESP32 WiFi stops working after some time

szerwi, 2018-08-12:
### Hardware:
Board: ESP32 Dev Module
Core Installation/update date: 11.08.2018
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed: 115200

### Description:
I'm using this library to send data to Influxdb database: https://github.com/teebr/Influx-Arduino. After some time (15 minutes - 6 hours) the esp32 is stopping sending data to database. I've enabled debug level verbose and here are my logs:

### Debug Messages:
When data is sent properly:
```
[D][HTTPClient.cpp:192] begin(): host: 192.168.1.92 port: 8086 uri: /write?db=Stacja_pogodowa
ESP32 Wind_speed=0.0,Wind_direction=270.0
[D][HTTPClient.cpp:850] connect():  connected to 192.168.1.92:8086
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'HTTP/1.1 204 No Content'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Type: application/json'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Request-Id: 1a136962-9e55-11e8-b7ee-000000000000'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'X-Influxdb-Build: OSS'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'X-Influxdb-Version: 1.5.2'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'X-Request-Id: 1a136962-9e55-11e8-b7ee-000000000000'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Date: Sun, 12 Aug 2018 17:28:18 GMT'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:968] handleHeaderResponse(): code: 204
[D][WiFiClient.cpp:452] connected(): Disconnected: RES: 0, ERR: 128
[V][HTTPClient.cpp:244] end(): tcp is closed
```

When data is not sent properly:
```
[D][HTTPClient.cpp:192] begin(): host: 192.168.1.92 port: 8086 uri: /write?db=Stacja_pogodowa
ESP32 Wind_speed=0.0,Wind_direction=270.0
[E][WiFiClient.cpp:213] connect(): lwip_connect_r: 104
[D][HTTPClient.cpp:846] connect(): failed connect to 192.168.1.92:8086
[W][HTTPClient.cpp:1124] returnError(): error(-1): connection refused
[V][HTTPClient.cpp:244] end(): tcp is closed
```

What does 'lwip_connect_r: 104' error mean? How to repair it?

lbernstone, 2018-08-12:
Looks like the connection is refused by the server.  Look at the server logs.

szerwi, 2018-08-12:
@lbernstone Hmm, I don't know how to enable any logs :D It is OrangePi with Linux. But the WiFi in OrangePi is working, so it would be strange if it refuses connection with ESP without any reason.

lbernstone, 2018-08-13:
The database server logs.

szerwi, 2018-08-13:
@lbernstone I know, but I don't know how to check any logs :D

szerwi, 2018-08-13:
I found it. There are no requests from ESP32. So the problem is on ESP32 side.

lbernstone, 2018-08-13:
Then you will need to Wireshark the connection to see the full transaction.

szerwi, 2018-08-13:
@lbernstone I don't know why, but I can't see any data coming from ESP32 in Wireshark. Is it because ESP32 is connected to the repeater?

szerwi, 2018-08-14:
@lbernstone I can see only Telnet data coming from ESP to my PC. There is no tracking from ESP to database. Do you know why?

falconLucas, 2018-08-20:
I'm having the same kind of issue. But for me, the esp32 loses its IP address after a few minutes running a webserver. I'm getting this message:
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 14 - MIC_FAILURE
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 8 - STA_LOST_IP

I've tried to solve this issue for a couple of days now. I have tried the following:
- Reinstalling everything and reprogramming the esp32
- Restarting the wifi router
- Adding a 22uf capacitor between 3v3 and GND to reduce power losses

closed 2019-08-15 by stale[bot]