espressif/arduino-esp32 #1752: Unable to change ESP32 CPU frequency in sdkconfig and boards.txt

usrdes, 2018-08-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			12/Aug/2018
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					80Mhz
PSRAM:                                                      Enabled
Upload Speed:						921600

### Description:
Any Arduino sketch that depends on CPU frequency other than 240 MHz, appears to fail.
Changing the sdkconfig as follows makes no difference:
#
# ESP32-specific
#
CONFIG_ESP32_DEFAULT_CPU_FREQ_80=
CONFIG_ESP32_DEFAULT_CPU_FREQ_160=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=160
CONFIG_SPIRAM_SUPPORT=y

AND making changes in board.tx file as follows shows no change:
esp32.build.f_cpu=160000000L

### Sketch:
Example from thus URL:
https://github.com/earlephilhower/ESP8266Audio

Modified to just access a web radio. You will need to add your ssid and password in this file:

[StreamMP3FromHTTP.txt](https://github.com/espressif/arduino-esp32/files/2281604/StreamMP3FromHTTP.txt)

```

### Debug Messages:
The audio playback is obviously much faster. 
Other than that the sketch works fine.
(one compile note about i2c call being deprecated, but the sketch works fine)

lbernstone, 2018-08-13:
The sdkconfig is only used if you compile the sdk libraries. https://github.com/lbernstone/arduino-esp32-80MHz
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

usrdes, 2018-08-13:
@lbernstone 
Thank you, I looked at your github links *before * I posted. 
I'd like to use the Arduino IDE,  & not the ESP-IDF. 
For me ( a non software guy) Arduino is a much easier user interface to use.
It would help if the Arduino IDE -> Tools -> had a CPU freq select.
Thanks for your links

closed 2019-08-15 by stale[bot]