espressif/arduino-esp32 #1754: wifi task not resetting task watchdog and hanging entire board

Sauro98, 2018-08-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			latest
IDE name:							VSCode
Flash Frequency:					       80Mhz
Upload Speed:						115200

### Description:

Sometimes when reconnectring to the wifi my esp32 board hangs endlessly with hte messages shown below. It sometimes blocks at the first try but usually it goes on for 1-2 days reconnecting a handful of times every day and the suddenly it stops. 

If ESP.restart() is called under these circumstances the board never restarts.

### Sketch:
```cpp

void connectToWiFI(){
  Serial.println(F("Reconnectiong"));
  WiFi.mode(WIFI_STA);
  Serial.println(String(wifi_configs.ssid));
  Serial.println(String(wifi_configs.password));
  WiFi.disconnect();
  Serial.println(F("Disconnected"));
  WiFi.begin(wifi_configs.ssid, wifi_configs.password);
}
```

### Debug Messages:
```
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
Reconnecting
**ssid**
**password**
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE

.... continuing forever
```

lbernstone, 2018-08-14:
Does WiFi.reconnect() work?  It does essentially the same thing, but maybe a bit less memory usage.  They are just calling esp_wifi_disconnect(), which goes to the hardware libraries.

Sauro98, 2018-08-14:
I Will try in the next days because It works forum days and then suddenly
halt

Sauro98, 2018-08-17:
Seems like connecting and then disconnecting at the begin and end of every HTTP request fixed it and greatly improved BLE performance for the sketch. I will wait some days before closing this to be sure that it is actually working.

Sauro98, 2018-09-04:
In the end my solution was to not use WiFi together with BLE. Devices using LoRa have shown no signs of hanging in weeks. I guess WiFi + BLE is just too much for some boards

closed 2018-09-04 by Sauro98