espressif/arduino-esp32 #1756: Ibeacon and wifi in esp32

hanadjerbi1, 2018-08-14:
I'm working with esp32 as ibeacon generator to be detected on other devices and with wifi to send data via mqtt to google cloud..but I have not found a way to make them work together ... suggestions?

Sauro98, 2018-08-14:
I am using WiFi + BLE scanning for IBeacons in my project and it mostly works. The only problems I am facing are:
* at random times the WiFi task will enter an infinite loop of not resetting the task watchdog, which causes the board to hang and a reboot is needed
* BLE scanning seems to be noticeably slower when WiFi is on

Of course, I had to resize the board's partitions, otherwise the sketch was just too big to load.

hanadjerbi1, 2018-08-14:
I tried to synchronize between wifi and ibeacon. First I connect to the wifi and then disconnect to ensure ibeacon connection but it does not reconnect to wifi unless after reboot...
about the repair.I made the change in the default.cvs file as follows
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,
storage,  data, fat,     0x300000, 1M,
still not working

Sauro98, 2018-08-14:
You also have to modify `upload.maximum_size` in boards.txt, under the name of your board.
For the record my default.cvs looks like this:
```

# Name,   Type, SubType,   Offset,  Size, Flags
--
nvs,      data, nvs,     0x9000,    0x5000,
otadata,  data, ota,     0xe000,    0x2000,
app0,     app,  ota_0,   0x10000,   0x190000,
app1,     app,  ota_1,   0x1A0000,  0x190000,
eeprom,   data, 0x99,    0x330000,  0x1000,
spiffs,   data, spiffs,  0x331000,  0x0CF000,
```
And in boards.txt I have

```
esp32.name=ESP32 Dev Module

esp32.upload.tool=esptool
esp32.upload.maximum_size=1638400
```

About the WiFI, I connect and disconnect it multiple times with little issues, does it happen every time to you?

hanadjerbi1, 2018-08-14:
yes unfortunately...

Sauro98, 2018-08-14:
Are you using `WiFi.reconnect()`? I have to disconnect and reconnect to the WiFi at each send, otherwise the BLE scan would be just too slow. 
I am using this code and it works for me almost every time (it hangs after running for some days):
```
// Before sending data via WiFi
WiFi.begin(ssid, password);

// Sending the actual data here

// After sending the data
WiFi.disconnect();
```
For some reason `WiFi.reconnect()` did not work for me, but using `WiFi.begin` every time seems to work

hanadjerbi1, 2018-08-14:
almost the same
//to disconnect
  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
//to connect
  WiFi.enableSTA(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

Sauro98, 2018-08-14:
(I edited my comment above)

I'll try to replicate the issue and let you know what I get

hanadjerbi1, 2018-08-14:
okay..i'm working to find a solution as soon as possible

Sauro98, 2018-08-14:
I tried using the code you posted here and it reconnects every single time without any issues. Bear in mind that, at least for me, it won't reconnect immediately. You might have to wait a little until it is connected. I have set a 40 seconds timeout and usually all connections happen in less than 15 secondes.

By the way how much heap is still free in your board at execution time? Heve you freed the memory used by bluetooth functionalities that you don't use?

Sauro98, 2018-08-14:
I'm using the code you posted here in the loop function of my WiFi+BLE program. It takes a lot of space for me too, but luckily you can free some of it by calling 

```
ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
```
before initializing the BLE. It frees the resources needed for classic Bluetooth, which are useless if you are only gonna use BLE

hanadjerbi1, 2018-08-14:
i have this error 
E (142272) phy_init: failed to allocate memory for RF calibration data

Sauro98, 2018-08-14:
When and where? Could you add a little more detail?

hanadjerbi1, 2018-08-14:
Connecting to WiFi
192.168.1.12
Connecting to mqtt.googleapis.com
Waiting on time sync...
MQTT connecting ...
projects/sage-archway-210516/locations/europe-west1/registries/IDregistre/devices/my-esp32-device
connected
/devices/my-esp32-device/config
mqtt done
mqtt disconnect
ibeacon done
ibeacon stop
Connecting to WiFi
192.168.1.12
Connecting to mqtt.googleapis.com
Waiting on time sync...
MQTT connecting ...
projects/sage-archway-210516/locations/europe-west1/registries/IDregistre/devices/my-esp32-device
failed, status code =-2 try again in 5 seconds

E (128892) phy_init: failed to allocate memory for RF calibration data
Backtrace: 0x40095ca8:0x3ffe31f0 0x40095eab:0x3ffe3210 0x40157d51:0x3ffe3230 0x401aceae:0x3ffe3260 0x401acf48:0x3ffe3290 0x401ad236:0x3ffe32c0 0x4
01ad3af:0x3ffe32f0 0x4008e12c:0x3ffe3310
Rebooting...

Sauro98, 2018-08-14:
Could you post the code that provides these messages?

hanadjerbi1, 2018-08-14:
void ibeacon()
{
  BLEDevice::init("ESP32");
  BLEDevice::setPower(ESP_PWR_LVL_P7);
  pServer = BLEDevice::createServer();
  myBeacon.setManufacturerId(0x4c00);
  myBeacon.setMajor(5);
  myBeacon.setMinor(88);
  myBeacon.setProximityUUID(MY_UUID);
  myBeacon.setSignalPower(0xb3);
  advertisementData.setFlags(ESP_BLE_ADV_FLAG_LIMIT_DISC | ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT);
  advertisementData.setManufacturerData(myBeacon.getData());
  pAdvertising = pServer->getAdvertising();
  pAdvertising->setAdvertisementData(advertisementData);
  pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
}
void mqtt_connect() {
  WiFi.enableSTA(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("Connecting to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.println(WiFi.localIP());
  Serial.println("Connecting to mqtt.googleapis.com");
  client.setCACert(root_cert);
  mqttClient.setServer(host, httpsPort);
  mqttClient.setCallback(callback);
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  configTime(0, 0, "pool.ntp.org", "time.nist.gov");
  Serial.println("Waiting on time sync...");
  while (time(nullptr) < 1510644967) {
  delay(10);
  }
  while (!client.connected()) {
    Serial.println("MQTT connecting ...");
    String pass = getJwt();
    Serial.println(pass.c_str());
    const char *user = "unused";
    String clientId = get_client_id();
    Serial.println(clientId.c_str());
    if (mqttClient.connect(clientId.c_str(), user, pass.c_str())) {
      Serial.println("connected");
      String configTopic = get_config_topic(device_id);
      Serial.println(configTopic.c_str());
      mqttClient.setCallback(callback);
      mqttClient.subscribe(configTopic.c_str(), 0);
    } else {
      Serial.print("failed, status code =");
      Serial.print(mqttClient.state());
      Serial.println(" try again in 5 seconds");
      delay(5000);
    }
  }         
}
esp_err_t event_handler(void *ctx, system_event_t *event)
{
    return ESP_OK;
}
void setup() {
  #if defined (__AVR_ATtiny85__)
    if (F_CPU == 16000000) clock_prescale_set(clock_div_1);
  #endif
  strip.begin();
  strip.show();
  Wire.begin();
  pinMode(LED_PIN, OUTPUT);
  fill_priv_key(private_key_str);
  Serial.begin(115200);
  cc.init_CCS811();
  bme680.init();
  mpu.init_MPU();
  gp2y10.init();
  tsl.init();
  Serial.println(ESP.getFreeHeap());
}
void loop() {
  esp_err_t ret = nvs_flash_init();
  if (ret == ESP_ERR_NVS_NO_FREE_PAGES)
  {
      ESP_ERROR_CHECK(nvs_flash_erase());
  }
  delay(3000);
  Serial.println(ESP.getFreeHeap());
  double outputV =gp2y10.getOutputV();
  double ugm3 =gp2y10.getDustDensity(outputV);
  double dust_score =gp2y10.getAQI(ugm3);
  float noise=i2s.init_I2S();
  float lum= tsl.Light();
  float co2=cc.para_CO2();
  float cov=cc.para_TCOV();
  float co = mq_7.getMQ7();
  String localisation = mpu.para(ax,ay,az);
  float t=bme680.CalculateTemp();//done
  float h=bme680.CalculateHumi();//done
  float p=bme680.CalculatePressure()/100 ;//done
  float g=bme680.CalculateGasResistance();//done
  float TScore=bme680.CalculateTScore(t);
  float HScore=bme680.CalculateHScore(h);
  float GScore=bme680.CalculateGScore(g);
  float TotalScore=bme680.CalculateTotalScore(TScore,HScore,GScore);
  String iaq_bme =IAQ_BME680(TotalScore);
  String iaq_co =IAQ_CO(co);
  String iaq_co2 =IAQ_CO2(co2);
  String iaq_dust =IAQ_dust(dust_score);
  if (!mqttClient.connected()) {
    mqtt_connect();
    delay(100);
  }
  mqttClient.loop();
  long now = millis();
  if (now - lastMsg > 3000) {
   lastMsg = now;
   String tmes ="{" "\"Temperature\":";
          tmes +=t;
          tmes += "}";
  tmes.toCharArray(pubMessage,500);
  String eventsTopic = get_events_topic(device_id);
  mqttClient.publish(eventsTopic.c_str(),pubMessage);
}
  Firebase.setFloat(" test / Temperature", t);
  Serial.println("mqtt done");
  WiFi.disconnect();
  mqttClient.disconnect();
  Serial.println(ESP.getFreeHeap());
  Serial.println("mqtt disconnect");
  delay(15000);
  ibeacon();
  Serial.println("ibeacon done");
  delay(15000);
  pAdvertising->stop();
  while (ret == ESP_ERR_NVS_NO_FREE_PAGES)
  {
      ESP_ERROR_CHECK(nvs_flash_erase());
      tcpip_adapter_init();
  }
  ESP_ERROR_CHECK( ret );
  Serial.println(ESP.getFreeHeap());
  delay(3000);
}

Sauro98, 2018-08-14:
From your logs it looks like that it is connecting just fine and instead `mqttClient.connect` is the one failing. Which mqtt library are you using?

hanadjerbi1, 2018-08-14:
WiFiClientSecure.h and PubSubClient.h

Sauro98, 2018-08-14:
What is `getJwt()`?
Also, I read of people with the same issue solving this problem by freeing the Bluetooth memory the did not need. Have you tried using the function i gave you above to free your BT classic memory?

hanadjerbi1, 2018-08-15:
sorry for being late to answer you .. yes, I used it but when I want to know the free memory after each instruction, even using all the functions you told me...I get this
projects/sage-archway-210516/locations/europe-west1/registries/IDregistre/devices/my-esp32-device
connected
/devices/my-esp32-device/config
60900 (memory )
59844(memory )
mqtt done
103332(memory )
mqtt disconnect
ibeacon done
24560(memory )
25016(memory )
Connecting to WiFi
192.168.1.12
Connecting to mqtt.googleapis.com
MQTT connecting ...
projects/sage-archway-210516/locations/europe-west1/registries/IDregistre/devices/my-esp32-device
-2 try again in 5 seconds

hanadjerbi1, 2018-08-15:
when i use   ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)); my esp32 reboot

Sauro98, 2018-08-15:
Are you calling it from `setup()` or from `loop()`? I am not using nkolban's library for BLE but my setup looks like this:
```c++
    nvs_flash_init();
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_bt_controller_init(&bt_cfg));
    ESP_ERROR_CHECK(esp_bt_controller_enable(ESP_BT_MODE_BLE));
```
Which is basically what you do when you call `BLEDevice::init`, which you could call just at the beginning of the program instead than at every loop

hanadjerbi1, 2018-08-15:
finally it works

hanadjerbi1, 2018-08-15:
finally it works but just for a while

closed 2018-08-15 by hanadjerbi1