espressif/arduino-esp32 #1757: BLEDevice: esp_bt_controller_init: rc=259 Invalid state

SoyTecnopata, 2018-08-15:
### Hardware:
Board:							ESP-WROOM-32 
Core Installation/update date:		14/Aug/2018  `fff178304644a93f72e9c280b52a8c4d9a7ef828`
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:					115200?
ESP32_BLE_Arduino installation: 		23/Jun/2018 `7951347ed68313d75c367e1f2cce763cb56d1eb2`

### Description:
Using Example BLE_server
I can compile it and Upload it
When I open the serial port there is a problem initiating BLE device
I have tried with different previous commits of the ESP32_BLE library but none solved the problem

### Sketch:
```cpp

/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "[4fafc201-1fb5-459e-8fcc-c5c9c331914b](https://github.com/espressif/arduino-esp32/commit/fff178304644a93f72e9c280b52a8c4d9a7ef828)"
#define CHARACTERISTIC_UUID "[beb5483e-36e1-4688-b7f5-ea07361b26a8](https://github.com/nkolban/ESP32_BLE_Arduino/commit/7951347ed68313d75c367e1f2cce763cb56d1eb2)"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("MyESP32");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

### Debug Messages:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Starting BLE work!
**E (12) BLEDevice: esp_bt_controller_init: rc=259 Invalid state**

```

lbernstone, 2018-08-15:
Where in the world did you get those UUID strings?  They should just be the hexadecimal numbers. https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/examples/BLE_server/BLE_server.ino

SoyTecnopata, 2018-08-15:
That was a mistake in mi comment the links were supposed to be in the hardware version of the issue request
I used that example, from : File -> Examples -> Examples for ESP32 -> ESP32 BLE -> BLE_server

lbernstone, 2018-08-15:
Set your board to ESP32 Dev Module and then turn debugging to verbose.  I haven't heard of other problems with this, so you may have a hardware problem.

Bascy, 2018-08-15:
Same here.
It looks like BT_CONTROLLER_INIT_CONFIG_DEFAULT is not defined because it doesnt find the definition of CONFIG_BT_ENABLED. 
There is a line in sdkconfig.h defining CONFIG_BT_ENABLED as 1.

I've tried this with an Arduino-esp32 library in Eclipse downloaded on May, 19th and ti works, but on the master HEAD it doesnt compile anymore

rmhomecouk, 2018-08-18:
I want to confirm that I also have this issue while I was testing out BLE integration with the Blynk App. Reverting to an older arduino-esp32 also resolved the issue for me.

Referenced here (and the commit that I reverted to)
[https://github.com/nkolban/esp32-snippets/issues/592](url)

geoffreyottoy, 2018-08-21:
Same problem here. Working on previous versions, but also get the error:
BLEDevice: esp_bt_controller_init: rc=259 Invalid state

SoyTecnopata, 2018-08-24:
I've been digging in my previous version and installations.
And I found a way to make it work for windows users:

In order to get rid of the `E (12) BLEDevice: esp_bt_controller_init: rc=259 Invalid state` message I uninstalled/deleted the arduino-esp32 folder from `C:/Users/[YOUR_USER_NAME]/Documents/Arduino `, where the documentation recommends to clone.

I also uninstalled Arduino and reinstalled older versions (tested on 1.6.13 and 1.8.0) but I don't think this is super necessary. Reboot the computer.

I followed the [Steps to install Arduino ESP32 support on Windows ](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md) but cloning in `C:\Program Files (x86)\Arduino\hardware`. I did this with a **Administrator mode** terminal. Also ran as administrator the get.exe file.

In the Arduino 1.6.13 version I  had to use the [previous arduino-esp32 repo version](https://github.com/espressif/arduino-esp32/commit/30b3eeb).
With `git checkout  30b3eeb`

Bascy, 2018-09-03:
I seem to have solved this by adding -DARDUINO_ARCH_ESP32 to the C++ compiler options

rmhomecouk, 2018-09-04:
Shouldn't that option be already included as part of the compilation (at least if you're doing from the Arduino IDE)? Do you know what commit you are using as I can confirm an older commit works as does others?

Bascy, 2018-09-04:
I'm using Eclipse IDE from SLoeber
Im at the lastest commit of BLE and of arduino-esp32

seannaes, 2019-06-11:
不知道你们解决这个问题了吗？我也遇到同样的问题了，求助！

closed 2019-08-24 by stale[bot]

SanchD, 2021-06-24:
@Bascy It does work Thank you! 

Guide: https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L51