espressif/arduino-esp32 #1761: Not working with MPU-9250 IMU, reads fail

MiloMindbender, 2018-08-16:
### Hardware:
Board:							Adafruit ESP32 feather
Core Installation/update date:			1.0.0
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
I noticed in some closed issues there were problems with the MPU-9250 IMU that were apparently resolved, but it appears these problems  may have come back.

My ESP32 is unable to communicate with the https://www.sparkfun.com/products/13762 9250 i2c breakout board.  When I got this it seemed to be running fine but with the latest release of ESP32 arduino support this doesn't seem to work any more.

Samples from this repository
https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library
or this one 
https://github.com/sparkfun/SparkFun_MPU-9250_Breakout_Arduino_Library

Both fail in similar ways, the writes appear to go through and be acked and all the reads fail with a nak according to my Saleae logic analyzer.  It is not even possible to retrieve the "who am I" value.

I have tried adding 3.3k pullups to the setup, but this does not make any difference.
I have tried using 100,000 and 400,000 speeds on Wire, no difference.

I have also double-checked the wires, only 4 (2 power and SDA/SCL, and they are correct.  Everything is powered from the 3.3v regulator on the Adafruit ESP32 Feather Board.

On one of my computers at home these samples DO seem to work, but I think that machine might have an older release of the ESP32 arduino libraries, I will check that and report back soon.

If it would be helpful, I can post a dump of the I2C traffic from my logic analyzer (software to view it is free).  If anyone wants it, let me know, I need to re-capture the data to shrink it's size a bit.

stickbreaker, 2018-08-16:
@MiloMindbender there is a known problem with Release 1.0.0, a fix has been applied to the main branch.  Until a new release has been compiled you can apply the fix by substituting the four i2c related files from the github source to your Arduino package location.

The Four files are:
* \libraries\Wire\src\Wire.h
* \libraries\Wire\src\Wire.cpp
* \cores\esp32\esp32-hal-i2c.h
* \cores\esp32\esp32-hal-i2c.c

under Windows 7 pro the Arduino Board Manager installs the package into:
>C:\Users\\{user}\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0

Chuck.

MiloMindbender, 2018-08-16:
I will give this a try.  Last night I did regression testing and found the MPU-9250 worked with 1.0.0-rc3 but failed on anything newer than that.

Also, I've noticed the i2c frequency on ESP32 is off-spec.  it's running 98khz when set to 100 and 382khz when set to 400.

stickbreaker, 2018-08-16:
@MiloMindbender The ESP32 hardware uses clock cycle counting to control bus frequency.  When it is sourcing SCL a single SCL cycle is constructed of four parts:
* first it pulls SCL Low
* second it detects when SCL is low and starts counting the pre-programmed CPU clock count for the low period.
* third it releases SCL (the pullup resistor and bus capacitance controls the time it takes SCL to reach the HIGH level).
* forth, it then detects the high level and starts counting the pre-programmed number of CPU clock cycles for the high period.
 
So, the bus capacitance and the pullup values distort the bus frequency from the pre-programmed values.  `Wire.setClock()` can set the pre-programmed values from 10KHz to over 1.5MHz.  If you need the bus frequency to be a specific period, you could use `Wire.setClock()` to adjust for you specific circuit. But, do realize that because of the delayed positive leading edge, your high/low periods will be asymmetric with a noticeably short HIGH component.

Chuck

MiloMindbender, 2018-08-16:
I just replaced the 4 files, it works better but is intermittant.  Some MPU9250 samples will start up ok the first time and maybe a time or two after pressing reset, then they start to fail.  A full power off is required to get it working again.

Some of the samples such as MPU9250BasicAHRS_I2C still fail every time.  These are trying to read the WHO_AM_I register and always fail at that point.

I tried inserting a 2 second delay after Wire.begin to give things a chance to stabalize, but this did not help.

stickbreaker, 2018-08-16:
@MiloMindbender are you seeing any debug output?

MiloMindbender, 2018-08-16:
The only output I get is the usual stuff when you press the reset button.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4

Do I need to set anything to get more debugging output?

stickbreaker, 2018-08-16:
@MiloMindbender set the Core Debug Level to at least DEBUG and run you code,  if the i2c system has any problems it will vomit out what it detects.

Also, inside \cores\esp32\esp32-hal-i2c.c there is a #define that will increase debug support and verbosity of messages;
line 45 [here](https://github.com/espressif/arduino-esp32/blob/172802b18e3540e39371d5125ba4dc70d07046e7/cores/esp32/esp32-hal-i2c.c#L45)

Chuck.

MiloMindbender, 2018-08-16:
More info.  I removed the 3.3k pullups and now it takes 6-7 tries to get a freeze.

I also tried clipping the logic analyzer to the MPU-9250 pins, getting the setup into a frozen state and then disconnecting SDA/SCL from the ESP, this showed SDA low and SCL high also.  So it appears the 9250 has gotten into a state where it's pulling SDA low for some reason.

Since the ESP can't communicate with it, a software reset of the 9250 isn't possible, so it remains stuck this way till you power off...not sure why this happens.

stickbreaker, 2018-08-16:
@MiloMindbender you can use `Wire.begin()` as a bus reset function.  The newest Github with PR#1717 changed `Wire.begin()` to a function to will cycle the bus if necessary.

If you can detect the lockup, try:
```c++
if(lockedUP){
  uint32_t timeout =millis();
  while((millis()-timeout<1000)&&(!Wire.begin())){ // twiddle thumb
   delay(5); // maybe IMU is busy? give it a chance to finish
  }
}
If(!Wire.begin()){ 
   Serial.println(" I2C bus did not Init, Fatal")
} else {
  // bus is good, back to regularly scheduled programming.
}
```

Chuck.

MiloMindbender, 2018-08-16:
Not sure if this is it or not...In the 9250 datasheet they mention

 "Each byte transferred must be followed by an acknowledge bit. To acknowledge, the slave device pulls the SDA line LOW and keeps it LOW for the high period of the SCL line. "

When it fails on a reset of the ESP32 it appears that SCL is high and SDA has just gone low and it stays this way.  It is doing a "setup read".  So maybe the 9250 was in the middle of an ACK when reset was pushed and is waiting for SCL to go low, but it never does?

Across a reset, should the SCL/SDA lines go low and back to high or just stay high?

Not an expert, just guessing.

MiloMindbender, 2018-08-16:
Will try to get your test code in shortly.  In the meantime maybe this logic analyzer trace is helpful?  
[image]

This shows the bus working ok, then reset is pressed and it  in the low/high state forever.

Hope I explained myself correctly earlier, what I meant was that maybe when the ESP32 was reset the 9250 was working on an ACK and had just pulled SDA low, but because of the reset SCL is now high.  the 9250 is waiting on an SCL low pulse that never comes, so it just stays stuck in that state?

Like I said, not an expert, does the logic analyzer trace help?

MiloMindbender, 2018-08-16:
Just to be totally clear, this lockup only happens when reset is pressed and you often have to reset 7 or more times before it ends up in this state.  The I2C does not appear to lock up at any other time.  I'm letting one of the test programs run for a while to be sure.

Also to be totally clear, in the logic analyzer trace above, where the signal stops is where reset was pressed and there is no activity on I2C once the ESP32 finishes its reset cycle.

Also, just checking, would you recommend I pull down a zip of the latest code here and install it before putting in and testing your patch above?

stickbreaker, 2018-08-16:
Grab the latest github main branch. I'll look at your trace, brb

stickbreaker, 2018-08-16:
This snip of your trace is questionable, the i2c decode is showing a STOP, but the raw voltage trace does not show SDA going high?  So, no stop was generated?

Can you add the `Wire.begin()` reset code and capture the scope output?  Each time `Wire.begin()` is called it will stimulate the bus, attempting to recover the locked up bus.  I would be interested in seeing how the IMU responds.

I did a Google search on the IMU and saw other with i2c SDA issues. (AVR STM and even Intel Edison)  So, the IMU is the locus, but, I haven't seen the triggering factors.

Chuck.

[image]

stickbreaker, 2018-08-16:
One guy said he could recover the IMU by connecting Vdd to SDA directly!  BAD! BAD! Worse than BAD!

stickbreaker, 2018-08-16:
>Also to be totally clear, in the logic analyzer trace above, where the signal stops is where reset was pressed and there is no activity on I2C once the ESP32 finishes its reset cycle.

If the ESP32 resets, It should boot the Arduino code and `Wire.begin()` should happen at least once,  So you should see at least a 9bit pulse of SCL if SDA is low.  If SCL and SDA are high, no pulse packet is sent.  The reset sequence is only sent if either SDA or SCL is low when `Wire.begin()` is called.  Or, when, a transaction is attempted.  When a `Wire.endTransmission()` or `Wire.requestFrom()` is executed, the first thing it does is verify the bus is idle (SDA & SCL HIGH). If the bus is not idle, it attempts to reset it.

You would see this reset pulse on the scope.

Chuck.

MiloMindbender, 2018-08-16:
Ok, getting really confusing,  This is what I see now, I press reset and it hangs on restart.  This is what the bus looks like.  I think I need to try and tie a trace to the reset button so I can see exactly where on this trace the reset is occurring.

[image]

stickbreaker, 2018-08-16:
All I'm seeing is that the transaction is aborted. that last read operation is up to the 6th bit of the first byte. Then the reset of the ESP32 happens. the Hardware reset stops the SCL clocking, but the IMU is in the middle of a byte read.  From it's point of view it is just waiting for the next SCL cycles.

During a i2c read, the master(esp32) supplys the SCL and the slave(IMU) controls SDA except for the 9th bit ACK signal. Every 9th SCL cycle the IMU allow the master to either ACK(low) the byte, with causes the IMU to again exclusively control the SDA pin for the next 8 clock cycles.  To Terminate a read cycle the last byte the master(esp32) wants to read needs to be NAK'd(HIGH).  The current `i2cCheckLineState()` only sends the equivilent of an ACK (LOW).  That new code may need additional changes.  If my understanding of what is happening is true.

```c++
  if(!digitalRead(sda) || !digitalRead(scl)) { // bus in busy state
        log_w("invalid state sda=%d, scl=%d\n", digitalRead(sda), digitalRead(scl));
        digitalWrite(sda, HIGH);
        digitalWrite(scl, HIGH);
        delayMicroseconds(5);
        digitalWrite(sda, HIGH);
        for(uint8_t a=0; a<9; a++) {
             delayMicroseconds(5);
            digitalWrite(scl, LOW);
            delayMicroseconds(5);
            digitalWrite(scl, HIGH);
            if(digitalRead(sda)){ // bus recovered, all done. resync'd with slave
                break;
            }
        }
    }
```

If we are out of sequence with the bit count of the IMU this code should resync.

Chuck.

MiloMindbender, 2018-08-16:
Yes, very confused.... you can see channel 2 is the reset button.  The hang up happens exactly when the reset button is pressed.
[image]

However when the reset button is released (not shown here), I see nothing at all happening to sda/scl.
I get the "invalid state" log message which shows the new i2cCheckLineState is running, but nothing at all on the logic analyzer, it just stays low/high like before.

I also tried adding back the 3.3k pullups which squared up the analog signals a bit, but no change in what it was doing

MiloMindbender, 2018-08-16:
Sorry, we wrote at the same time...the above trace is before the new code you just posted....I'll retest in a minute.

stickbreaker, 2018-08-16:
lengthen your capture window, we need to see what happens after EN(reset) go high for at least 1 second.

stickbreaker, 2018-08-16:
change your trigger to trigger when EN go high.  That is where the interesting stuff is.  What you are showing is the failure, I want to see the recovery attempts.

Chuck.

MiloMindbender, 2018-08-16:
I am capturing out 4 seconds after the reset goes high, well past the time when your "i2cCheckLineState" message appears.  I just didn't show you it because SDA and SCL remain perfectly flat, not so much as a bobble even in the analog.  Here is another capture of the fail and when reset goes high SDA and SCL stay low/high for at least 4 seconds.
[image]

MiloMindbender, 2018-08-16:
I ran the capture out 10 seconds after the reset, still nothing at all after the freeze occurs.

Gotta run for a bit, be back in about 2 hours if you have any more suggestions.

stickbreaker, 2018-08-16:
is channel 2 EN?  I want to see the LOW->HIGH transition of EN.  That is were the ESP32 boots, and after the boot loader start Arduino.  maybe 500ms after EN low->high is where the recover should show up.

This capture is showing the reset, everything is correct.  It is the recovery that is failing.

If SCL is high and SDA is low and you are seeing:
>[W][esp32-hal-i2c.c:1363] i2cCheckLineState(): invalid state sda=0, scl=1
>
>[E][esp32-hal-i2c.c:1379] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init

Without any SCL activity something is not correct. 
The code should have reset the I2C hardware and be manually stimulating the i2c bus.

I'll do some thinking and see if I can add some useful debugging statements to track down this problem

Chuck.

MiloMindbender, 2018-08-17:
Channel 2 is connected to a pin labeled reset on my board, which I assume is same as EN.  And yes, for 10 seconds after reset goes high there is absolutely no change in SDA (low) and SCL (high)

I see the "invalid state" messages almost the instant reset goes high, so that code is being executed, but it doesn't seem to be doing anything to SDA and SDL, both the analog and digital versions are dead flat for at least 10 seconds.

I am pretty sure your modified code is being executed because the first version you gave me had a typo I had to correct, so I'm it is being compiled in.

I could put some code in i2cCheckLineState that would flip some unused GPIOs for my logic analizer so we can see the exact timing of when the code runs, if that would help.

At the moment though, it looks like that code is running but there is no change in state of the SDA/SCL lines when it does.

stickbreaker, 2018-08-17:
@MiloMindbender I have spent the last few hours setting up tests, and they fail.  The code in `i2cCheckLineState()` does not reset slave devices.  I will have to compare it to my last working code and try to figure out what changed.  I am not seeing any of the manual stimulation of the bus being express at the pins.  It did work in my fork. This code is a modified version of my fork. So I just have to work through the equivalent functions to find the problem.

It will probably take a couple of days to figure out what went sideways. 

The reproducible error sequence is to reset the ESP32 during a i2c read, at the exact moment when the next bit to be output by the slave is a zero.  This causes the fatal cascade.

Hopefully I can resurrect my stickbreaker fork.  I have been working on slave mode and my local copy is broken.  I'll have to redownload the last working copy from github.

It will be at least tomorrow, maybe Saturday before I get a chance to do this, I'm done for the day. I'm working on 5hr sleep in the last 48hr. just ran out of steam.

Chuck.

MiloMindbender, 2018-08-17:
Thanks for looking at this, I can continue to work with the code in it's current state.

One question, do you think it is something in the ESC32 that actually causes the lock up?  Or is the ESC32 issue just that it isn't able to reset the bus to a good state?

stickbreaker, 2018-08-17:
@MiloMindbender Just posted a PR that fixes this problem, try it. #1767

Chuck.

xvinny, 2018-08-23:
Replacing the Wire.cpp, Wire.h, esp32-hal-i2c.c and esp32-hal-i2c.h? Does Espressif wants to be trustable? They should be careful when releasing SDK updates :-/

stickbreaker, 2018-08-23:
@xvinny You get what you pay for, If it doesn't meet your specification, You can provide the fix.  This is an open source project.  Instead of complaining, use that effort to find and solve problems.

Chuck.

xvinny, 2018-08-23:
@stickbreaker you are right, I'm already doing this, but what I'm talking is about the releases and not the current branch. I believe that Espressif pays someone(s) to control the release approval.

stickbreaker, 2018-09-04:
@MiloMindbender yes that is correct procedure.  
If you have already installed the `arduino-esp32` Release 1.0.0 using Board-Manager, using this procedure will add a second copy (github development version).  BUT, it is difficult to distinguish which version you are using.

After you follow the "update" procedure, I recommend you do open small modification:
 
in C:\Users\{your user name}\Documents\Arduino\hardware\espressif\esp32\platform.txt

change:
>name=ESP32 Arduino

to
>name=ESP32 Arduino GitHub Version

When you want to select the github version go into Tools -> Board -> 
Scroll down, until you see the board you use.  make sure you select your board under "ESP32 Arduino GitHub Version"  not "ESP32 Arduino"

Chuck.

MiloMindbender, 2018-09-06:
Hmmm....used the boards manager to uninstall the RC1.0.0, the "github version" is still showing up on the menu but not when I just choose "boards manager"

The code compiles, but it is behaving like it's still running the RC version (ie: I2C isn't even starting)

I think something is definately messed up, I'm going to try and clear out the remaining ESP32 files and reinstall just the github version.

MiloMindbender, 2018-09-06:
@stickbreaker looks like SUCCESS.  Cleaned out everything and re-installed Github ESP32 from scratch.  Everything builds ok now.

The Recovery process appears to be working...the debug verbose output looks like this on a hang
`[V][esp32-hal-i2c.c:1436] i2cInit(): num=0 sda=23 scl=22 freq=0
[V][esp32-hal-i2c.c:1621] i2cSetFrequency(): Fifo threshold=3
[W][esp32-hal-i2c.c:1363] i2cCheckLineState(): invalid state sda(23)=0, scl(22)=1
[D][esp32-hal-i2c.c:1371] i2cCheckLineState(): Recovered after 1 Cycles
`
I have seen the "recovered" message go as high as 6 cycles but it always seems to recover.

The logic analyzer sees this

[image]

stickbreaker, 2018-09-06:
The release version is in `appdata\local\arduino15`. When you remove it, `appdata\local\arduino15\packages\esp32` should disappear. 

The github development version resides in `{sketchdirectory}\hardware\espressif\esp32`

chuck.

stickbreaker, 2018-09-06:
The recovered cycle count is the number of SCL clocks required for the slave to release SDA.  It has to synchronize with the slave. It should always be less than 8.  Because, a slave can only output 8 bits before it waits for the ACK bit.  When the recovery sequence see the ACK bit it can now issue a STOP to abort the in progress transaction.

MiloMindbender, 2018-09-06:
@stickbreaker Caught one that went 8 cycles, it looked like this

[image]

Now that the reset issue is fixed I will be running some more tests with the MPU-9250 today, if I don't encounter any other problems communicating with it I'll close the issue.

Thanks for fixing this!

P.S. Does the ESP32 native dev environment us this same code for I2C?

stickbreaker, 2018-09-06:
The release 1.0.0 version has a BUG with ReSTART operations (`Wire.endTransmission(false)`). It will not correctly execute the succeeding `Wire.requestFrom()` call.  The `requestfrom()` will always fail.

You should use the GitHub dev version until me-no-dev creates a new release.

Chuck.

stickbreaker, 2018-09-06:
in your scope example, the Slave was emitting a '0', eight lows.  And, the i2c bus cycle happened to be at bit zero when the interrupt or reset happened. 

>P.S. Does the ESP32 native dev environment us this same code for I2C?

The github dev version is the most up to date.  

Release 1.0.0 does not have the correct reset code.

MiloMindbender, 2018-09-07:
I had a ton of issues today, but none of them were I2C, they were all bugs in people's MPU-9250 libraries!

So I will close now.

By the way when I asked about the native dev environment  I was referring to the Native ESP-IDF environment that runs FreeRTOS and NOT arduino.

I was just wondering if they use your code too or if they have something different.

closed 2018-09-07 by MiloMindbender