espressif/arduino-esp32 #1762: ETH static IP assigns, but can't connect

wcpettus, 2018-08-16:
### Hardware:
Board:							ESP32-Gateway (Olimex)
Core Installation/update date:		14/August/2018 (commit e346f20aa9f455f3147d9d018e09dd7c05818951)
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					80MHz
Upload Speed:						115200

### Description:
(The Olimex board has a built-in ethernet port, so there I was very lazy with the hardware)

Ethernet connection breaks when assigning static IP.  When adding an ETH.config() line to assign static IP after ETH.begin(), all connections fail, although the static IP is successfully set.  Here is the Serial monitor:
```
ETH Started
ETH MAC: 30:AE:A4:1C:43:53, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH MAC: 30:AE:A4:1C:43:53, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps

connecting to google.com
connection failed
```

If I tried in a different order with ETH.config() before ETH.begin(), it immediately fails and continuously reboots the device:
```
E (44) tcpip_adapter: tcpip_adapter is not initialized!
abort() was called at PC 0x400d83ab on core 1

Backtrace: 0x4008f858:0x3ffb1e80 0x4008fa5b:0x3ffb1ea0 0x400d83ab:0x3ffb1ec0 0x400d1b9d:0x3ffb1f00 0x400d19b2:0x3ffb1f40 0x4011e4a3:0x3ffb1fa0

Rebooting...
```

If I add a ~10 sec delay and before running the ETH.config(), it at leasts connects and verifies that it works, before failing on repeat:
```
ETH Started
ETH Connected
ETH MAC: 30:AE:A4:1C:43:53, IPv4: 192.168.0.7, FULL_DUPLEX, 100Mbps
ETH MAC: 30:AE:A4:1C:43:53, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps

connecting to google.com
connection failed
```

And finally if I add a ~10 sec delay and call testClient once before running the ETH.config(), it continues connecting successfully for ~20 times before failing on repeat.

### Sketch:
Sketch is just a minimally modified version of ETH_LAN8720.ino from the examples.  Line ETH.config(...) added in setup (this is certainly not a minimal script to reproduce, but it is a minimal modification).
```cpp
/*
    This sketch shows the Ethernet event usage

*/

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
  ETH.config(IPAddress(192, 168, 0, 90),IPAddress(192, 168, 0, 1),IPAddress(255, 255, 255, 0));
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

### Debug Messages:

core debug appears to be unavailable for the ESP32-Gateway?

### Reference

The ETH_LAN8720 example works nicely, lots of successful test connections at google; the log looks like this, but repeating:
```
ETH Started
ETH Connected
ETH MAC: 30:AE:A4:1C:43:53, IPv4: 192.168.0.7, FULL_DUPLEX, 100Mbps

connecting to google.com
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Thu, 16 Aug 2018 05:03:44 GMT
Expires: Sat, 15 Sep 2018 05:03:44 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection
```

lbernstone, 2018-08-17:
You really need to see what error config is throwing there.  Switch to the esp32 dev module to turn on core debugging, and throw a `Serial.setDebugOutput(true)` in your setup.

pipi61, 2018-08-17:
Hi!
Try set the dns address too.
in arduino-esp32/libraries/WiFi/src/ETH.cpp: 
bool ETHClass::config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1, IPAddress dns2)

me-no-dev, 2018-08-17:
`ETH.config(IPAddress(192, 168, 0, 90),IPAddress(192, 168, 0, 1),IPAddress(255, 255, 255, 0),IPAddress(192, 168, 0, 1));` << you need to add the DNS server. If you would have enabled the log, you would have seen it :)
```
connecting to google.com
[E][WiFiGeneric.cpp:662] hostByName(): DNS Failed for google.com
connection failed
```

wcpettus, 2018-08-18:
Thanks for the advice, I can confirm that setting the DNS explicitly solves my problem (so much for trying to minimize my first test).

Naively I assumed it would work because the config arguments allow you to bypass the DNS.  In hindsight that was silly for an immediate static config which bypasses DHCP.

I'm not sure I understand what happens if I let DHCP the connection and then 10 sec later overwrite the IP.  I see the DNS still gets zeroed, which wasn't what I was expecting:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/ETH.cpp#L152
This is even weirder in the case where I made a single testClient connection after establishing DHCP, then did the config - where I get ~10-20 connections before it dies.  Is it somehow using the now erased DNS setting?

Now I'm definitely asking questions in the weeds.  Thanks again for the helpful replies.

closed 2019-02-07 by wcpettus

maheshchimpanna, 2019-04-08:
hi we are also using the same code but we are not getting ETH.macAddress in the serial port.we are getting "Eth Started" "Ethernet connected" only after dumping the code.what may be the issue?

dienvx1997bn, 2021-04-07:
setup function here:
void setup() {
    Serial.begin(115200);
    WiFi.onEvent(WiFiEvent);
    ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
    ETH.config(IPAddress(192, 168, 9, 125), IPAddress(192, 168, 9, 1), IPAddress(255, 255, 255, 0), IPAddress(192, 168, 9, 1));
}
and result here:
23:19:24.846 -> ETH Started
23:19:28.849 -> ETH MAC: 24:6F:28:88:D0:A3, IPv4: 192.168.9.125, FULL_DUPLEX, 100Mbps
23:19:28.849 -> ETH MAC: 24:6F:28:88:D0:A3, IPv4: 192.168.9.125, FULL_DUPLEX, 100Mbps
23:19:29.864 -> 
23:19:29.864 -> connecting to google.com
23:19:29.966 -> HTTP/1.1 301 Moved Permanently
23:19:29.966 -> Location: http://www.google.com/
23:19:29.966 -> Content-Type: text/html; charset=UTF-8
23:19:29.966 -> Date: Wed, 07 Apr 2021 16:19:29 GMT
23:19:29.966 -> Expires: Fri, 07 May 2021 16:19:29 GMT
23:19:29.966 -> Cache-Control: public, max-age=2592000
23:19:29.966 -> Server: gws
23:19:29.966 -> Content-Length: 219
23:19:29.966 -> X-XSS-Protection: 0
23:19:29.966 -> X-Frame-Options: SAMEORIGIN
23:19:30.000 -> 
23:19:30.000 -> <HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
23:19:30.000 -> <TITLE>301 Moved</TITLE></HEAD><BODY>
23:19:30.000 -> <H1>301 Moved</H1>
23:19:30.000 -> The document has moved
23:19:30.000 -> <A HREF="http://www.google.com/">here</A>.
23:19:30.000 -> </BODY></HTML>
23:19:30.000 -> closing connection

josefgull, 2021-12-01:
i wanted to ask if the code can be changed working with dhcp instead of static ips. ?