espressif/arduino-esp32 #1763: ESP32 SPIFFS stops write files after some period of time

dppddd, 2018-08-16:
### Hardware:
Board:							Nodemcu Esp32 Development Board (ESP32 Dev Module) 
Core Installation/update date:	        latest update
IDE name:						Arduino 1.8.5.
Flash Frequency:					40-80 Mhz (both settings was used)
Upload Speed:					115200 (uploaded via OTA)

### Description:
I am writing short files using SPIFFS. The new files are written using SaveVariablesToDisk after a previous file (beginning with "/a0") is deleted using SPIFFS.remove. Or, at seldom conditions (occurring once in 10 or 100 or even more minutes), the number of stored files is increased by using SaveVariablesToDisk after a previous file (beginning with "/a0") is renamed using SPIFFS.rename, as shown in the code below. This code works fine and as expected.

The problem is that after many hours of execution (8-12 hours), the function ConfigFile.write in the SaveVariablesToDisk function stops write data and returns 'false', which crashes my code. I cannot understand why .write function which works fine all the time, stops work as expected after many SPIFFS read-write cycles and the only way to start proper execution again is to format SPIFFS (or remove all the files from it). Is this a bug of core or something wrong with my code?

### Sketch:
In function executed periodically:

File root = SPIFFS.open("/");
if(!root)
  {
        return res;
  }
if(!root.isDirectory())
  {
        return res;
  }
File file = root.openNextFile();
String CurrentName="";
while(file)
  {
      if(!file.isDirectory())
        {
          String Name=file.name();
          int FileNumberPos=Name.indexOf("a");
          int FromPos=Name.indexOf("from");
          int ToPos=Name.indexOf("to");
          int DotPos=Name.indexOf(".");
          int EndPos=Name.indexOf("e");

          if(FileNumberPos>-1&&FromPos>-1&&ToPos>-1&&DotPos>-1&&EndPos>-1)
            {
              long FileNumber=Name.substring(FileNumberPos+1,FromPos).toInt();
              float From=Name.substring(FromPos+4,ToPos).toInt();
              From=From/100;
              float To=Name.substring(ToPos+2,DotPos).toInt();
              To=To/100;
              if(FileNumber==0)
                {
                  CurrentName=Name.substring(FromPos, EndPos+1);
                  TM=Name.substring(DotPos+1,EndPos).toInt();
                }
              else
                {
                  AllFiles=AllFiles+Name;
                  if(MinFileNumber==-1&&MaxFileNumber==-1)
                    {
                        MinFileNumber=FileNumber;
                        MaxFileNumber=FileNumber;
                    }
                  if(MinFileNumber>FileNumber)MinFileNumber=FileNumber;
                  if(MaxFileNumber<FileNumber)MaxFileNumber=FileNumber;
                }
            }
        }
        //file.close(); - tried, not helped
        file = root.openNextFile();
        //delay(100); - tried, not helped
  }
//root.close(); - tried, not helped
if(CurrentName!=""&&TM!=Time)
  {
     if(MinFileNumber==-1&&MaxFileNumber==-1)
      {
        MaxFileNumber=1;
        MinFileNumber=1;
      }
     else MaxFileNumber++;
     String NewName="/a"+String(MaxFileNumber)+CurrentName;
    if(!SPIFFS.rename("/a0"+CurrentName, NewName))logstr=logstr+"<br/>cannot rename from "+"/a0"+CurrentName+" to "+NewName;
    else logstr=logstr+"<br/>rename from "+"/a0"+CurrentName+" to "+NewName+" is successfull";
    AllFiles=AllFiles+NewName;
  }
else 
  {
    if(!SPIFFS.remove("/a0"+CurrentName))logstr=logstr+"<br/>cannot remove "+"/a0"+CurrentName;
    else logstr=logstr+"<br/>remove "+"/a0"+CurrentName+" is successfull";
  }
if(MinFileNumber==-1&&MaxFileNumber==-1)return "not found!";  
for(int i=MaxFileNumber;i>=MinFileNumber;i--)
  {
    long StartFilePos=AllFiles.indexOf("/a"+String(i)+"from"); 
    if(StartFilePos==-1)continue; 
    long EndFilePos=AllFiles.indexOf("e",StartFilePos+1);  
    String FileName=AllFiles.substring(StartFilePos,EndFilePos+1);

    String Contents=ReadFromFileToVariable(FileName);
   .....
   ..... 
  }

bool SaveVariablesToDisk(String FilePath,String Var)
{
File ConfigFile = SPIFFS.open(FilePath, FILE_WRITE);
if (!ConfigFile) return false;
for (int i = 0; i < Var.length(); i++)
  {
    if(!ConfigFile.write(Var[i]))
      {
        ConfigFile.close();
        return false;
      }
  }
ConfigFile.flush();
ConfigFile.close();
return true;
}

In another function executed periodically:
float Min=some changable value;
float Max=some changable value;
long TM=some changable value;
FileName="/a0from"+String(int(Min*100))+"to"+String(int(Max*100))+"."+String(TM)+"e";
FileContents="some contents is here";
if(!SaveVariablesToDisk(FileName,FileContents))logstr=logstr+"<br/>cannot save "+FileName;
else logstr=logstr+"<br/>Save "+FileName+" is successfull";
            

### Debug Messages:
```
Code uploaded by OTA 
```

dppddd, 2018-08-17:
Nobody can suggest a solution of the issue? Are "bad sectors" possible on ESP32 memory, that prevents SPIFFS write operation? Also is it any way to check SPIFFS free space available? However I do not think that this is free space problem as when the problem occurs, the number of files does not exceed 20 and the size of each file does not exceed 700 bytes.

Also, does ESP32 use external pins during SPIFFS operations? I am using the following hardware connected to the ESP32:

pins 32, 33, 13, 25 - relay module
pins 18, 19 - sensors
pin 23 - hall sensor with interrupt attached on it
pins 15, 27 - buttons, pulled up
pins 2, 4 - leds

Is any conflict between my hardware and SPIFFS write operation is possible?

lbernstone, 2018-08-17:
It is very possible nobody wants to try to read your message due to formatting problems.  Try doing a SPIFFS.end() and begin() every 1000 files or so. Look at the preferences library, which may be a better solution for storing small bits of information.  There is a total and used bytes method in spiffs, which will give you free space. SPI flash uses pins 6-11.  Pins 18 and 19 are a poor choice for sensors, as ADC2 is not available when WiFi is running.  Use the pins in the 30s for sensors, and pins below for gpio.

me-no-dev, 2018-08-17:
I would also suggest you do add a counter to each operation (open, write, rename) and see if it always stops at the same number (could be a very good clue). But TBH you are using SPIFFS not really in the way it was meant to be used... flash chips have a relatively low number of write/erase cycles and you will exhaust it quite fast, which will break it to a point where it will be unusable ;)

Maybe get an SD-Card?

dppddd, 2018-08-21:
I set counters on SPIFFS read - write operations and used totalBytes() and usedBytes() functions as advised. It seems like the problem is not depends on the number of read and write operations. But it definitely depends on used bytes. When the problem occurs, I have 49196 bytes total with 42921 bytes used. When I format the SPIFFS and start the process again I have 49196 bytes total with 6275 bytes used. 
Is the 48 KB - normal size for SPIFFS? Can this size be increased? 

Also, when I write 700 bytes file, does it takes several kilobytes of real SPIFFS space? Because it seems like I have some memory leakage.

Later, I will consider using the preferences library or SD-Card as advised. But now I want to complete the project on SPIFFS as it is almost ready and changing to preferences library or SD-Card requires some time of re-coding.

stickbreaker, 2018-08-21:
@dppddd SPIFFS is configured with 256byte pages, a 'file' needs one page for a header, and filesize/(pagesize - metadata) pages for data.  MetaData is overhead that SPIFFs uses to organize the data, 12 to 32 bytes. 
So, for your 700 byte file it uses (1+4)256 bytes or 1280 bytes. and SPIFFS needs two Blocks (4k) free at all times for updates.  49196 -42921 = 6,275.  all done no more room!

Chuck.

dppddd, 2018-08-23:
Thank you for the explanation. Does this mean that storing small data in one file is more preferred method to save free space, comparing to storing the same small data in many different files?

I have also found memory leakage in my code. When the new data processed, I have saved to SPIFFS not only 700 bytes data files, but also all data sent to update my HTML page, including 2 HTML tables as is, always increasing with the new data, and also other raw data. After modifying the code, now I have only 10 KB SPIFFS usedbytes. I have 2 MB data total written to SPIFFS (with 4500 file flush operations) in 14 hours, and almost 3 MB data read from it in the same time. What will be the lifetime of my SPIFFS memory with such read-write productivity? 

My another question is regarding preferences library. Does it use the same memory with the same amount of free space as SPIFFS?

lbernstone, 2018-08-23:
Preferences uses the [nvs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html) partition and functions under the covers.  This is a key-pair storage system.  It uses the same flash storage on the backend, but is record based, so is more efficient for storing small bits of data (<2k), in the same way that a database is more efficient than writing a file for each line of data.  It will in fact likely use more bytes than writing files to SPIFFS, but will typically be faster.  A database like [EDB](https://github.com/jwhiddon/EDB) may be appropriate for your use case.

Flash storage longevity is typically measured in write cycles. A conservative estimate would be about 10000 cycles.  You can do the math for yourself, but writing to an sd card makes that a replaceable part rather than the whole unit.

dppddd, 2018-08-28:
Before changing from SPIFFS to preferences, I've decided to optimize my code in a point of flash write operations. I have achieved 10 KB per hour, in contrast to 150 KB, written each hour before the optimization. Once I have uploaded the sketch through USB, and later via OTA, but since that moment, I have noticed that the SPIFFS total Bytes changed from 49196 to 1374476, that is for some reasons, the SPIFFS size was increased from 48 KB to more than 1 MB. What are the factors that affect the SPIFFS total size? Does it depends on upload method (USB vs OTA)? Or does it depends on the size of my final compiled code? BTW, my not compiled source code after the optimization become even larger than before.

lbernstone, 2018-08-28:
When uploading from Arduino IDE, you can choose a partition scheme.  OTA uploads don't touch the SPIFFS partitions. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ota.html

dppddd, 2018-08-28:
@lbernstone , thanks. My initial problem with SPIFFS free space was because I didn't know that the Partition Scheme setting is useless for OTA uploads. Most likely that I uploaded my first sketch via USB with "Minimal SPIFFS" partition scheme, and later when I tried to change it to default, this had not effect as all subsequent sketches were uploaded via OTA . Nevertheless this discussion was helpful for me and as a result I greatly improved the code.

closed 2019-08-15 by stale[bot]