espressif/arduino-esp32 #1764: https get request issue

muradali362, 2018-08-16:
### Hardware:
Board:							        ESP32 Dev Module
Core Installation/update date:			8/16/2018
IDE name:							Arduino IDE
Flash Frequency:					        40Mhz
Upload Speed:						115200

### Description:
i am trying to run aws greengrass discovery using wificlientsecure example for esp32 
successfully connected to secure server 
but of get request cant get desired output 
please help

### Sketch:
wificlientsecure.cpp

 client.setCACert(test_root_ca);
 client.setCertificate(test_client_cert); // for client verification
 client.setPrivateKey(test_client_key);	// for client verification

  Serial.println("\nStarting connection to server...");
  if (!client.connect(server, 8443))
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
    client.println("GET https://greengrass.iot.us-west-2.amazonaws.com/greengrass/discover/thing/cert-device_0 HTTP/0.9");
 
    client.println();

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
```

### Debug Messages:
```

Starting connection to server...
Connected to server!
headers received
{"message":"Not Found","traceId":"df1ebb67-3096-a1c6-53cf-d78c45fb7238"}⸮
```

beegee-tokyo, 2018-08-16:
`if (line == "\r")` cannot work. You compare the whole received String against a single character.

Sauro98, 2018-08-17:
I think they're searching for a line that is composed of just `"\r\n"` that separates the headers from the body

muradali362, 2018-08-17:
THANK YOU IT WORKED 
URL WAS WRONG AND WAS NOT GETTING COMPLETE RESPONSE DUE TO (line == "\r") CHECK 

Thanks you @beegee-tokyo  and  @Sauro98  for your help

copercini, 2018-08-19:
Closed as solved =)

closed 2018-08-19 by copercini