espressif/arduino-esp32 #1765: matching esp-idf

dsptech, 2018-08-17:
Not an issue.

Hi, 
I'm working to add the arduino support to an already done esp-idf project. I followed the instructions and now the arduino-esp32 library is installed as component in that project.

The esp-idf project now is compiling fine, and our own arduino libraries too, but I noticed different behaviors of the arduino-esp32 libraries (The first one: a bigger latency in the WiFi class). 

I'm now investigating on these behaviors, so, I need an information: 
What is the esp-idf version that match the current the arduino-esp32 framework ? (aaf1239 ?)
Is there a way to get that information by inspecting the published code ?

Thank you.
Regards.

lbernstone, 2018-08-17:
In arduino-esp32, `git log | grep -i "update idf"`
Then, in esp32 `git log | grep <commit number from above>`
`git checkout <commit>`
Unfortunately, no way to figure that out without going through the git logs.

me-no-dev, 2018-08-17:
Code:
```cpp
Serial.print("ESP32 SDK: ");
Serial.println(ESP.getSdkVersion());
```

Serial:
```
ESP32 SDK: v3.2-dev-518-g020ade652
```

😝 😝 😝

dsptech, 2018-08-17:
Hi,
I retrieved the id: aaf1239. This confirm to me that the observed behaviors are not version matching related.
(sigh) I'll continue my investigation.

Thank you.
Regards.

closed 2018-08-17 by dsptech

dsptech, 2018-08-17:
for @me-no-dev :
I think that ESP.getSdkVersion() called in a esp-idf project return the currently installed sdk version instead of the sdk version used to build the arduino framework.
ESP.getSdkVersion() need to be called in a arduino environment to produce the wanted result. Is it correct ?
Regards

dsptech, 2018-08-17:
Ok.
Thank you for the support.
Regards.