espressif/arduino-esp32 #1771: Sketch too big error

gabovich, 2018-08-17:
### Hardware:
Board:							NodeMCU-32s
Core Installation/update date:	        11/jul/2018
IDE name:					       Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:			         	115200

### Description:
Hey,
I've got a memory problem while i'm facing ble and wifi libraries. I understood that they consume lot of storage. 
According to the error the maximum is 1310720 bytes and i'm exceed this (about 109%, and it's just the beginning of the project).
According to the datasheet the memory of the NodeMCU-32s is 16MB Flash, 520KB SRAM.
Something I don't understand - the arduino ide compiling use the flash memory? Or something else?

I read about partition that can be made to increase storage available for the code but i didn't understand how to do this. 
How can I edit the default.csv? I can open the file only by EXCEL.
There is another solution? 

I'm just in the beginning of my project and will need a lot more than 1310720 bytes.
Help please.

moijerem59, 2018-08-27:
bonsoir , dans outils partitions choisi minimal spiffs

lbernstone, 2018-08-27:
You can choose a different partition scheme, or edit the csv file using a text editor.  The partition sizes are in hex for convenience with the math. The ESP32 has SRAM and flash RAM.  The flash is storage.  Most boards have 4MB, but 16MB modules are readily available now.

closed 2019-08-15 by stale[bot]

drmpf, 2020-02-03:
I am getting the same issue.
I tried to edit the default.csv partition size BUT the Arduino ESP32 build process for the default partition has the memory size hard coded and the upload is NOT run.
The error shown is

```
Sketch uses 1337334 bytes (102%) of program storage space. Maximum is 1310720 bytes.
Global variables use 61608 bytes (18%) of dynamic memory, leaving 266072 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board SparkFun ESP32 Thing.
```
A work around is to select a non-default partition setting like "no OTA", the build/upload process then completes.
I noticed a possible commit to allow partition.csv file in the sketch dir
https://github.com/espressif/arduino-esp32/commit/9ef3e2d2a6b79c951f894dba1dd71c811317ca6e
 but that did not work for me using version 1.0.2

Tip. You need to reload you SPIFFS after a partition change.