espressif/arduino-esp32 #1772: How to use matrix touch buttons for esp32

ahmed847, 2018-08-17:
### Hardware:
Board:							ESP32 Devkit v1
Core Installation/update date:			18/June/2018
IDE name:						Sloeber
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I am working with touch of esp32. I want to use matrix touch like in the following link: https://github.com/espressif/esp-iot-solution/blob/master/examples/touch_pad_evb/main/evb_touch_matrix.cpp

But the problem is that "arduino-esp32" does not have the functions required for matrix touch. Please tell me how I can achieve this on "arduino-esp32".

Thanks in advance.

closed 2019-08-15 by stale[bot]

jwktje, 2021-06-01:
If possible I'd like to see this issue reopened as I also would like to use the touch matrix in Arduino. Specifically this example with the ESP32-S2: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/touch_element/touch_matrix

lbernstone, 2021-06-01:
This example should work in 2.0.0alpha release.

jwktje, 2021-06-01:
Could I use the exact same code without issue inside the Arduino Core instead of ESP-IDF when using 2.0.0alpha?

jwktje, 2021-06-02:
@Ibernston

> This example should work in 2.0.0alpha release.

Would I [be able to use this directly](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/touch_element/touch_matrix/main/touch_matrix_example_main.c) in Arduino or would it need some rewriting to work?