espressif/arduino-esp32 #1774: PSRAM in WRover again

manecolooper, 2018-08-19:
i'm trying to write and read 900000 bytes in psram , with this simple sketch:

uint8_t* sDelayBuffer0 = (uint8_t*)ps_malloc(900000);

//uint8_t* sDelayBuffer0 = (uint8_t*)malloc(90000);

unsigned long DelayCounter = 0;
byte leido ;
byte escrito = 0 ;

void setup() {
   Serial.begin(115200);
   
}

void loop()
{

   sDelayBuffer0[DelayCounter] = escrito ;
   escrito++ ;
   if(escrito >= 255)
         {
          escrito = 0;
         }
   
   leido = sDelayBuffer0[DelayCounter] ; 
   Serial.println(leido, HEX);
   Serial.println(DelayCounter, HEX);
   DelayCounter++ ;
       if(DelayCounter >= 900000)
         {
          DelayCounter = 0; 
   
          //digitalWrite(RLED, HIGH);
          
         }

}

it crashes , but if i use malloc with a size of up to 90000 , ten times smaller, it works fine

how to use PSRAM ? 

thanks!

stickbreaker, 2018-08-19:
What/whom is crashes? (Bet you cannot answer my question.) 

Please post the actual program output and any error messages.

Just saying 'it crashes' doesn't help us solve your problem.

Chuck.

manecolooper, 2018-08-19:
thanks, this is the output
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0af9  PS      : 0x00060330  A0      : 0x800e4a11  A1      : 0x3ffb1f80  
A2      : 0x3ffc23e8  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffb2034  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc23ec  
A10     : 0x3ffc23e4  A11     : 0x00000000  A12     : 0x0800001c  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0af9:0x3ffb1f80 0x400e4a0e:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0af9  PS      : 0x00060330  A0      : 0x800e4a11  A1      : 0x3ffb1f80  
A2      : 0x3ffc23e8  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffb2034  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc23ec  
A10     : 0x3ffc23e4  A11     : 0x00000000  A12     : 0x0800001c  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0af9:0x3ffb1f80 0x400e4a0e:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310

stickbreaker, 2018-08-20:
Based on the crash report, 
>uint8_t* sDelayBuffer0 = (uint8_t*)ps_malloc(900000);

is returning NULL.  I don't think PSRAM is functioning on your board.
see What this prints:
```c++
void setup(){
  Serial.begin(115200);
Serial.printf("Internal Total heap %d, internal Free Heap %d\n",ESP.getHeapSize(),ESP.getFreeHeap());
    //Internal RAM
    //SPI RAM
  Serial.printf("SPIRam Total heap %d, SPIRam Free Heap %d\n",ESP.getPsramSize(),ESP.getFreePsram());
 
  Serial.printf("ChipRevision %d, Cpu Freq %d, SDK Version %s\n",ESP.getChipRevision(), ESP.getCpuFreqMHz(), ESP.getSdkVersion());

  Serial.printf(" Flash Size %d, Flash Speed %d\n",ESP.getFlashChipSize(), ESP.getFlashChipSpeed());
 }
```

stickbreaker, 2018-08-20:
You must not be using the current github main branch.

stickbreaker, 2018-08-20:
Here is the current ESP.h https://github.com/espressif/arduino-esp32/blob/80c110ece70b179ddfe686e8ee45b6c808779454/cores/esp32/Esp.h#L53

manecolooper, 2018-08-20:
now i updated 

Internal Total heap 396836, internal Free Heap 371908
SPIRam Total heap 4194252, SPIRam Free Heap 4194252
ChipRevision 1, Cpu Freq 240, SDK Version v3.2-dev-39-gaaf12390
 Flash Size 4194304, Flash Speed 80000000

stickbreaker, 2018-08-20:
Are you still getting the same error?

Try: 
```c++
byte * ps=byte * ps_malloc(900000);

Serial.printf("ps=%p\n",ps);
```

manecolooper, 2018-08-20:
Thanks, error still persists, i tried those two lines 

result:

Internal Total heap 396712, internal Free Heap 371784
SPIRam Total heap 4194252, SPIRam Free Heap 4194252
ChipRevision 1, Cpu Freq 240, SDK Version v3.2-dev-39-gaaf12390
 Flash Size 4194304, Flash Speed 80000000
ps=0x3f800024

me-no-dev, 2018-08-20:
How about this:

```cpp
uint8_t* sDelayBuffer0 = NULL;
unsigned long DelayCounter = 0;
byte leido ;
byte escrito = 0 ;

void setup() {
   Serial.begin(115200);
   sDelayBuffer0 = (uint8_t*)ps_malloc(900000);
}

void loop(){
   sDelayBuffer0[DelayCounter] = escrito ;
   escrito++ ;
   if(escrito >= 255){
      escrito = 0;
   }
   leido = sDelayBuffer0[DelayCounter] ; 
   Serial.println(leido, HEX);
   Serial.println(DelayCounter, HEX);
   DelayCounter++ ;
   if(DelayCounter >= 900000){
      DelayCounter = 0; 
   }
}
```

Do not use ps_malloc out in the wild like this! Use it inside functions. PSRAM is not even available when you are trying to use it...

manecolooper, 2018-08-20:
That worked!!! thanks a million!!!!

closed 2018-08-20 by manecolooper

pipi61, 2018-08-20:
Only comment:
byte escrito how to  >= 255 ??