espressif/arduino-esp32 #1778: Conflict between accelerometer MPU-9265 and Oled SSD1306

joseandres15, 2018-08-20:
### Hardware:
Board:	WEMOS TTGO ESP-32 Wifi Bluetooth 0.96" OLED
Core Installation/update date:			20/aug/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Hi! I'm trying to solve a problem on my project but I can't find a solution for it.

I want to get data from an accelerometer (MPU-92/65) connected to the board by means of I2C on pins 21 (SDA) and 22 (SCL). Then, I would like to show the angles data obtained by the accelerometer on the OLED display integrated on the board. The OLED also communicates with the board by I2C, and it is internally connected to pins 4 (SCL) and 5 (SDA).

The angles readings are obtained fine. On the other hand, I can show any message on the screen. My problem comes when I try to set them on the same program. When I try this, the angles readings are obtained fine, but the screen turns off. Trying to fix it, I realized that if I comment the line "Wire.begin()" on the "setup()" function, then the screen finally turns on but the program is not able to read data from the accelerometer. I'm not sure if the problem is on the function "Wire.begin()" or is further than this fact.

I've been looking for a function which turns the I2C communication off, like "Wire.end()" but what I found didn't work for me. I would like to make them work simultaneously because I'm sure that it is possible. But if I couldn't make it, I would like to read data for an instant and then turn off the communication and show the data on the screen.

The libraries used are "SSD1306.h" and "OLEDDisplayUi.h" for the OLED display and "Wire.h" standard for I2C communication on Arduino.

Could anybody help me? I will go mad soon.

Thanks.

PS: There are probably some inconsistencies in the code. I am new using Arduino and some things I have copied and pasted into my program.

### Sketch:
include <Wire.h>   // Librería para comunicarse con el acelerómetro por I2C.
#include "SSD1306.h"      // Librería para utilizar la pantalla OLED.
#include "OLEDDisplayUi.h"

SSD1306 display(0x3c, 5, 4);          // Definimos la pantalla.
OLEDDisplayUi ui ( &display );

#define    OLED_ADDRESS              0x3c // Dirección hexadecimal de la pantalla.
#define    OLED_RST                  16   // GPIO16 Pin reset de la pantalla.

// DEFINICIÓN DE ACELERÓMETRO
#define    MPU9265_ADDRESS           0x68

#define    ACC_FULL_SCALE_4_G        0x08

void I2Cread(uint8_t Address, uint8_t Register, uint8_t Nbytes, uint8_t* Data){
    Wire.beginTransmission(Address);
    Wire.write(Register);
    Wire.endTransmission();
    Wire.requestFrom(Address, Nbytes);
    uint8_t index = 0;
    while (Wire.available())
        Data[index++] = Wire.read();
}

void I2CwriteByte(uint8_t Address, uint8_t Register, uint8_t Data){
    Wire.beginTransmission(Address);
    Wire.write(Register);
    Wire.write(Data);
    Wire.endTransmission();
}

void setup() {
  Serial.begin(9600);
  Wire.begin();

  pinMode(OLED_RST,OUTPUT);
  digitalWrite(OLED_RST, LOW); // low to reset OLED
  delay(50); 
  digitalWrite(OLED_RST, HIGH); // must be high to turn on OLED
  
  display.init(); // Inicializamos pantalla.
  display.setContrast(255);
  display.setFont(ArialMT_Plain_16); // Estilo.
  display.setTextAlignment(TEXT_ALIGN_LEFT); // Origen en la izquierda.

  I2CwriteByte(MPU9265_ADDRESS, 28, ACC_FULL_SCALE_4_G);
}

void loop() {

 // LECTURAS DEL ACELERÓMETRO EN LOS EJES X,Y,Z.
   uint8_t Buf[14];
   I2Cread(MPU9265_ADDRESS, 0x3B, 14, Buf);
  
  int16_t ax = -(Buf[0] << 8 | Buf[1]);    
  int16_t ay = -(Buf[2] << 8 | Buf[3]);
  int16_t az = -(Buf[4] << 8 | Buf[5]);

  // CONVERSIÓN DE LAS LECTURAS.
    // Conversión a aceleración (m/s^2)
      float ax_conv=-ax*9.80665/8207.066225+0.027654;
      float ay_conv=-ay*9.80665/8226.456689-0.170296;
      float az_conv=az*9.80665/8185.788079-0.144666;
      if(ax_conv<0.3){
        ax_conv=round(ax_conv);
      }
      if(ay_conv<0.3){
        ay_conv=round(ay_conv);
      }
      if(az_conv<0.3){
        az_conv=round(az_conv);
      }
      
    // Conversión de aceleración a orientación
      float Y = RAD_TO_DEG * (atan2(-ay_conv, -az_conv));  
      float X = RAD_TO_DEG * (atan2(-ax_conv, -az_conv));  
      float Z = RAD_TO_DEG * (atan2(-ay_conv, -ax_conv));
    
      int a = RAD_TO_DEG * (atan2(az_conv, ax_conv));  
      int b = RAD_TO_DEG * (atan2(ay_conv, ax_conv));

 // Mostrar datos por pantalla
  display.clear(); // Limpiar pantalla
  display.display(); // Muestra lo que haya en buffer
  display.drawString(0, 5, "Alpha= " + String(a));
  display.display(); // Muestra lo que haya en buffer
  display.drawString(0, 25, "Beta= " + String(b));
  display.display(); // Muestra lo que haya en buffer
}

stickbreaker, 2018-08-20:
@joseandres15 you have three choices:
* First rewire your IMU or SSD1306 to use the same SDA, SCL pins
* Second Switch between 4,5 and 21,22
Wire.begin(5,4); // before you access the SSD1306
Wire.begin(21,22); // before you access the IMU
* Third use both i2c peripherals and change either your SSD1306 or IMU library to use the second peripheral
Wire1.begin(secondSDA,SecondSCL);
The library would need all `Wire()` calls changed to `Wire1()`

Your Choice.

Chuck.

joseandres15, 2018-08-21:
The first option would be the most logical, because I know i2c support different devices on the same pins and manage them without problems. But when I try the i2c scanner, it doesn't recognize the screen address on pins 4 and 5. It only finds the address 0x68 of the accelerometer on pins 21 and 22. Due to this reason, I don't know if to wire accelerometer on pins 4 and 5 would work. I'm right or it doesn't matter?. However, despite the address is not recognized on the i2c scanner, screen works without problems (if "Wire.begin()" is commented and it doesn't affect to the code, of course).

The second option would be the simplest. I have tried it, but it doesn't work.
I have wrote "Wire.begin(5,4);" before "uint8_t Buf[14];" and "Wire.begin(21,22);" before "display.clear();" but nothing changes.

Third option... I don't understand it and I don't know how to apply it. Could you give me some details more?

stickbreaker, 2018-08-21:
Remember the SSD1306 has a reset pin, It will not answer i2c calls unless the reset is clear.
```c++
     digitalWrite(rst, HIGH);
    // VDD (3.3V) goes high at start, lets just chill for a ms
    delay(1);
    // bring reset low
    digitalWrite(rst, LOW);
    // wait 10ms
    delay(10);
    // bring out of reset
    digitalWrite(rst, HIGH);
```

joseandres15, 2018-08-21:
@stickbreaker I had already declared the reset pin and did reset this pin in the "setup" function. Furthermore, the reset pin is clear ("high") when it leaves the "setup" function. Should I do reset it in another place in the code?

I have tried with the same code which you have written for the reset, but nothing changes.

stickbreaker, 2018-08-21:
I use a ssd1306 with 6 24lc512 eeproms, 2 MCP23017 IO expanders without any problems. Verify you have SDA, SCL and RST correct.

stickbreaker, 2018-08-21:
Are you sure SCL is 5 not 15

joseandres15, 2018-08-21:
@stickbreaker This image shows that pins 4 and 5 are I2C pins of the screen.

[image]

On the other hand, although the image does not show it, pins 21 and 22 are also I2C and separately (pins 4/5 and 21/22) work fine. But together they don't.

What I am beginning to think is that pin 16 is not really the OLED reset pin. I copied it from another code that it would be, but I think it is not for my board.  The problem is that I can't find which is my OLED reset pin...

copercini, 2018-09-21:
Closed as defective hardware

closed 2018-09-21 by copercini