espressif/arduino-esp32 #1779: FireBeetle - LED_BUILDIN causes reset on a battery.

kalvis84, 2018-08-20:
### Hardware:
Board:							FireBeetle ESP32 V2.0
Core Installation/update date:			30/aug/20178
IDE name:							Arduino IDE 
Flash Frequency:				  80Mhz
Upload Speed:						951600

### Description:
Board has connected USB and Battery. When, after programming, I disconnect USB, board resets and works on battery. But if in the code there is a line that activates output for LED_BUILDIN (D9/IO02) then ESP32 resets after that command. 
Line
 ``` rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) ```
suggests that there was software reset?
It can be a hardware issue with the board. I also have a different issue on that board #829.

### Sketch:
```cpp

/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  delay(1000);
  Serial.println("Begin");
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
  Serial.println("Blinking...");
}
```

### Serial print with    "pinMode(LED_BUILTIN, OUTPUT);" not commented:
```
rst:0x1 (POWERON_RESETY,boot:0x13 (SPI_FAST_FLASH_BOOTY

configsip: 0, SPIWP:0xee

clk_Hrv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:808

load:0x40078000,len:6084

load:0x40080000,len:6696

entry 0x400802e4

ŰyŚ’ß€llâĂNěĂOŕĂOěűnßÄ	bßĆňnăbŰnă’Ü€€€ÜľnplśÄ°l#p‚nâűNŕĂnŚăn|bßÉrÜČnnßbßŚ	~’ÜŔ‚ß±ňnßÄĎÜÇÄÉânŘÜľ‚Ü¸’Üâb#``~bß‰ÄŔŚ’ÜŹn|’ÜĆÄ’ßąßöl`’	<0><0><0>Ś<0>ŕ‚ÇŔ€ě8ÄŽ<0>$Ž0<0><0><0>Ś<0>ŕÇŔś€ěŽ<0>l €<0><0><0>Ś<0>ŕÇŔ€€€<0>l ‚<0><0><0>Ś<0>ŕrÇŽâÜ‡ňÇÎŚ8€<0>l`6Ä<0><0><0>Ś<0>ŕrÇŽâÜ‡ňÇČŚŽŔ<0>l Ž<0><0><0>Ś<0>ŕrÇŽ‚ÇŔlŽŚ8€<0>l <0><0><0>Ś<0>ŕÇŔ€€l<0>l ’<0><0><0>Ś<0>ŕrÇŽâÜ‡ÇÉě8€<0>l`6Ü<0><0><0>Ś<0>ŕrÇŽâÜ‡Ç‡ÇČ<0>l ž<0><0><0>Ś<0>ŕÇŔ€€‚<0>l ŕ<0><0><0>Ś<0>ŕÇŔ€€Ž<0>l â<0><0><0>Ś<0>ŕÇŔ€€l8’<0>l`6Ă<0><0><0>Ś<0>ŕÇŔ€€l8’<0>l ‚l<0><0><0>Ś<0>ŕrÇŽâÜ‡ňÇČŚŽŔ<0>l ‚Ś<0><0><0>Ś<0>ŕ€ślśŚ8<0>Ű8Ž<0><0>ě€?ŽŚśrÜŔlŽă<0>rŰ8<0><0>ě€?€€€€<0>Ű8’<0><0>ě€?€€€€<0>ŽŰ@Ś0<0><0>ě€?€€€‚Ś8<0>NŰ`ŰIŽŰAě€?€€€€<0>rN’	‰lě€?€€€€<0>p`Ű<0><0>ě€?€€€€<0>pŰA|A<0><0>ě€?€€€€<0>pŰ`~ŰI|I<0>ě€?€€€€<0>rrŰprnäbßrÜ‰b€?€bÇđěěŕrÇŽâÜ‡ňÇÎŚ8€€?žŚśâÜŽâÜŽ‚ÜöŕrÇŽâÜ‡ňÇÎśŔ€?€śě8ÇöŕrÇŽâÜ‡ňÇÎŚśŔ€?€śě8ŕŚŽäŕrÇŽâÜ‡ňÇđśŕl`¶`F’bÜżňnänßňÜľŚbß¶`ţets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:808

load:0x40078000,len:6084

load:0x40080000,len:6696

entry 0x400802e4

ŰpÜ¸Ä‰rÜ‰ßČnnßÄ°bŚ<0>l#p‚nâűNŕĂnŚăn|bßÉrÜČnnß’`¶śŔrÜÁlŕ‚ÇŔ€ŚŽŕŽÇbŔ¶śŔrÜĆlŕÇŔ€€€bŔ¶śŔrÜÇlŕÇŔ€€€bŔ¶śŕrÜÎÜ‚ÜĆ’ŕÇŔ€€€bŔ†śß‡’ŕÇŔ€€€l`ţ

```

stickbreaker, 2018-08-21:
Do you have a hardware fault?  does the led flash?

kalvis84, 2018-08-21:
LED works ok on USB. I currently check if there will be a similar problem on low battery.

closed 2019-08-15 by stale[bot]