espressif/arduino-esp32 #1781: Using hardware timer causes SPIFFS to crash

marcmerlin, 2018-08-21:
Using arduino IDE 1.8.6 beta with an ESP32 dev board

I have git master for esp-32 and esp-idf, this did not fix my problem.
Accessing SPIFFS in various ways while having a timer interrupt, causes SPIFFS to crash as shown below.
I can work around the problem by disabling the interrupt before talking to SPIFFS:
timerAlarmDisable(timer);
spiffs_stuff_here
timerAlarmEnable(timer);

Timer definition:
https://github.com/z3t0/Arduino-IRremote/blob/master/irRecv.cpp#L5
https://github.com/z3t0/Arduino-IRremote/blob/master/irRecv.cpp#L130

crash:
09:40:44.358 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
09:40:44.392 -> Core 1 register dump:
09:40:44.392 -> PC      : 0x400de9e4  PS      : 0x00060034  A0      : 0x40081ae4  A1      : 0x3ffc0bf0  
09:40:44.392 -> A2      : 0x00000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x400874dc  
09:40:44.392 -> A6      : 0x00000000  A7      : 0xffffffff  A8      : 0x80081608  A9      : 0x3ff5f024  
09:40:44.392 -> A10     : 0x3ffc10fc  A11     : 0x20000000  A12     : 0x00000400  A13     : 0x3ffb1a10  
09:40:44.425 -> A14     : 0x00000000  A15     : 0x3ffb1a4c  SAR     : 0x00000019  EXCCAUSE: 0x00000007  
09:40:44.425 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
09:40:44.425 -> Core 1 was running in ISR context:
09:40:44.425 -> EPC1    : 0x4008a500  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400de9e4
09:40:44.425 -> 
09:40:44.458 -> Backtrace: 0x400de9e4:0x3ffc0bf0 0x40081ae1:0x3ffc0c10 0x4008a4fd:0x00000000

PC: 0x400de9e4: IRTimer() at /home/merlin/Arduino/libraries/IRremote/IRremote.cpp line 131
EXCVADDR: 0x00000000
Decoding stack results
0x400de9e4: IRTimer() at /home/merlin/Arduino/libraries/IRremote/IRremote.cpp line 131
0x4008a4fd: esp_rom_spiflash_read_data at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/spi_flash_rom_patch.c line 222

marcmerlin, 2018-08-21:
If I have the right esp-idf version, 
static esp_rom_spiflash_result_t esp_rom_spiflash_read_data(esp_rom_spiflash_chip_t *spi, uint32_t flash_addr,
        uint32_t *addr_dest, int32_t byte_length)
{
(...)
            WRITE_PERI_REG(PERIPHS_SPI_FLASH_ADDR, temp_addr << 8);
            REG_WRITE(PERIPHS_SPI_FLASH_CMD, SPI_USR);
            while (REG_READ(PERIPHS_SPI_FLASH_CMD) != 0);   << 222 here

marcmerlin, 2018-08-21:
Last bit: I can't just turn off the interrupt before doing SPIFFS, I need to turn it off enough time before doing SPIFFS (not sure what "enough" actually is, but if it's just before I still get crashes).

stickbreaker, 2018-08-21:
@marcmerlin every bit of code you call during an interrupt must be IRAM_ATTR.
When SPIFFS is updating flash, it has to DISABLE the iCache Updates because FLASH is DISABLED, so any calls to code that is not currently in iCache is not accessible.  The iCache is a 64k code page buffer, the FLASH hold your complete program, But, the CPU needs the program in RAM before it can execute it. So, there is a background process that catches cache miss calls.  Whenever program execution falls outside the current contents of iCache this background process, stalls execution, reads from FLASH, reAddresses execution, and releases execution.  If iCache is disabled because FLASH is Disabled, the Explosion happens.

stickbreaker, 2018-08-21:
@marcmerlin Do some searches on IRAM_ATTR.  This is your problem.

Chuck.

marcmerlin, 2018-08-21:
@stickbreaker thanks much, reading now.

marcmerlin, 2018-08-21:
@stickbreaker so, the ISR is defined here:
https://github.com/z3t0/Arduino-IRremote/blob/master/IRremote.cpp#L127
all I did was change the timer definition to say
void IRAM_ATTR IRTimer()
sadly, this doesn't seem to fix the problem.

The docs say that any function called from an IRAM function must be in IRAM too. Does it mean I need to manually track all the functions called in there and if they are not core functions from esp-arduino, I must track them recursively and add IRAM to all of them, otherwise I get the same problem?
Or the compiler should be nice enough to do this error prone work for me?

marcmerlin, 2018-08-21:
@stickbreaker . So IRRemote ultimately is not my code, it's an external library I'm calling given that there is still no ESP32 RMT IR reading library that I could find today.
I checked https://github.com/z3t0/Arduino-IRremote/blob/master/IRremote.cpp#L127 and all it seems to do is call digitalread which is not in my control, TIMER_RESET which is defined to nothing, and that's about it.
So I'm a bit lost, sadly.

stickbreaker, 2018-08-21:
What about the `BLINKLED_ON()`, `BLINKLED_OFF()`?

Chuck.

stickbreaker, 2018-08-21:
@marcmerlin post the code for your actual ISR.

Chuck.

stickbreaker, 2018-08-21:
I don't remember seeing any `volatile` marking on the timing buffer. Where/how does the timing buffer get decoded?

Chuck.

marcmerlin, 2018-08-21:
@stickbreaker  first thanks for sticking with me and your help.
I will look for volatile, although this should cause the library to fail working and not a crash I believe.
In the meantime, I found what piece of code to comment out
https://gist.github.com/marcmerlin/d55c600f6407c5b03cf508ba66c2cb96#file-irremote-cpp-L185
this is the actual file I'm using now, the rest of the library is posted in the upstream location: https://github.com/z3t0/Arduino-IRremote/
Basically I took out a very small piece of the state machine and it stopped the issue.
I probably didn't fix anything but just removed barely enough code not to overflow a cache I guess.
In the meantime, things work ok and the part of that state machine doesn't seem to matter for the IR decoding I need.
In other words, I made the problem go away, but in the wrong way and of course there is no way my patch will get accepted.

stickbreaker, 2018-08-21:
@marcmerlin Well, the problem was not with esp32/arduino, so I think you should close this issue.

Chuck.

stickbreaker, 2018-08-21:
I agree that the code you commented out is irrelevant. 
your back trace decode is bogus to:

09:40:44.425 -> Core 1 was running in ISR context:

Backtrace: 0x400de9e4:0x3ffc0bf0 0x40081ae1:0x3ffc0c10 0x4008a4fd:0x00000000
 PC: 0x400de9e4: IRTimer() at /home/merlin/Arduino/libraries/IRremote/IRremote.cpp line 131 EXCVADDR: 0x00000000 
Decoding stack results
0x400de9e4: IRTimer() at /home/merlin/Arduino/libraries/IRremote/IRremote.cpp line 131 
0x4008a4fd: esp_rom_spiflash_read_data at  /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/spi_flash_rom_patch.c  line 222

I don't see how this stack trace is in an interrupt context.
This is more what I would expect:
>0x400d1368: feeder() at C:\Users\dancl\Documents\Arduino\libraries\Adafruit_VS1053_Library/Adafruit_VS1053.cpp line 737 (discriminator 1)
0x40080ddd: __onPinInterrupt at C:\Users\dancl\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-gpio.c line 211
0x400817d9: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x400dbc63: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85

the `_xt_lowint1` is the interrupt dispatcher in FreeRTOS.

Your stack trace show your TimerISR() was called from SPIFFS?

Chuck.

marcmerlin, 2018-08-21:
As sad as my workaround is, it's fixed my problem.
Sadly I don't have a short program to reproduce, but I believe if checking out my entire source code, it should be reproducible without the IR or Neopixel hardware I'm using.
This is however likely to be time consuming to look into, so I understand that it may be a priority or may not happen at all, especially since I can't prove that the chip and/or the arduino-esp32 layers are at fault (although it's starting to feel like one of the 2 might be)
I'll live this up to you and @me-no-dev to decide compared to other priorities you have.

stickbreaker, 2018-08-21:
your stack trace doesn't show a return through the FreeRTOS interrupt dispatcher. When the timer interrupt happens, it generates an interrupt that is dispatched. The stack has to be cleaned up after the ISR, priority changed, interrupt cleared. 

The IRAM_ATTR move the code completely into IRAM,  the Cache failure error is related to something else.  Are you using PSRAM to store the decode buffer?

Chuck.

marcmerlin, 2018-08-21:
@stickbreaker  I have no extra RAM connected and arduino shows PSRAM disabled

stickbreaker, 2018-08-21:
The only way this can be fixed is if you provide a small test case that exhibits the problem, or others encounter this error and they provide a test case. 

Based on my experience and your work around, the actual problem locus in somewhere else in your code.

Chuck.

marcmerlin, 2018-08-21:
@stickbreaker I'll take your word for it, and we both agree that my fix/workaround is a non fix, just luck that happens not to trigger the real problem for now.
I'll see I can make a smaller example later, we both agree that my current code (which you haven't seen) is way too big anyway.
How about I close this as wontfix and we can re-open if I have more usable data.

marcmerlin, 2018-08-21:
@stickbreaker  Well, making minimal code to trigger this was trivial turns out.
Here's the new crash
PC: 0x40080e37: IRTimer() at /home/merlin/Arduino/libraries/IRremote/IRremote.cpp line 141
EXCVADDR: 0x00000000
Decoding stack results
0x40080e37: IRTimer() at /home/merlin/Arduino/libraries/IRremote/IRremote.cpp line 141
0x40081285: __timerISR at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-timer.c line 88
0x40089e81: esp_rom_spiflash_read_data at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/spi_flash_rom_patch.c line 238
14:52:59.589 -> Enabling IRin
14:52:59.590 -> Before SPIFFS begin
14:52:59.590 -> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
14:52:59.622 -> Core 1 register dump:
14:52:59.622 -> PC      : 0x40080e37  PS      : 0x00060034  A0      : 0x80081288  A1      : 0x3ffc0bd0  
14:52:59.622 -> A2      : 0xbad00bad  A3      : 0x20000000  A4      : 0x00000400  A5      : 0x3ffb1c30  
14:52:59.622 -> A6      : 0x00000000  A7      : 0x3ffb1c6c  A8      : 0x3ffc2458  A9      : 0x3f400020  
14:52:59.622 -> A10     : 0x00000001  A11     : 0x0000001e  A12     : 0x00000002  A13     : 0x3ffb1c10  
14:52:59.656 -> A14     : 0x00000000  A15     : 0x00000005  SAR     : 0x00000002  EXCCAUSE: 0x00000007  
14:52:59.656 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
14:52:59.656 -> Core 1 was running in ISR context:
14:52:59.656 -> EPC1    : 0x40089e84  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080e37
14:52:59.656 -> 
14:52:59.656 -> Backtrace: 0x40080e37:0x3ffc0bd0 0x40081285:0x3ffc0bf0 0x40081761:0x3ffc0c10 0x40089e81:0x00000000

Code:
#include <SPIFFS.h>
#include <IRremote.h>
IRrecv irrecv(34);

// Setup method runs once, when the sketch starts
void setup() {
    delay(3000);
    Serial.begin(115200);
    Serial.println("Enabling IRin");
    irrecv.enableIRIn();
    irrecv.blink13(false);

    Serial.println("Before SPIFFS begin");
    SPIFFS.begin();
    Serial.println("SPIFFS Directory listing:");
    File dir = SPIFFS.open("/");
    while (File file = dir.openNextFile()) {
	Serial.print("FS File: ");
	Serial.print(file.name());
	Serial.print(" Size: ");
	Serial.println(file.size());
    }
}

void loop() {}

IR library: https://github.com/z3t0/Arduino-IRremote/

marcmerlin, 2018-08-21:
This crashes happens even adding IRAM as discussed, and goes away if I comment out the 2 lines of code in that library:
https://gist.github.com/marcmerlin/d55c600f6407c5b03cf508ba66c2cb96#file-irremote-cpp-L185
Of course, I can't promise that the library is bug free, I didn't write the entire library but the bug seems pretty much contained in the ISR which is pretty short, so I think the rest of the lib can be ignored.
Timer is defined here:
https://github.com/z3t0/Arduino-IRremote/blob/master/irRecv.cpp#L130
the only thing that may be necessary is to have valid SPIFFS data already uploaded or the SPIFFS code may short circuit and not run anything that causes the crash.

atanisoft, 2018-08-23:
I have also encountered crashes on the ESP32 with SPIFFS and a hardware timer. I couldn't find an easy solution so I moved away from SPIFFS. In my case I'm not using any libraries and the ISRs are relatively simple (https://github.com/atanisoft/DCCppESP32/blob/master/src/SignalGenerator.cpp#L169, https://github.com/atanisoft/DCCppESP32/blob/master/src/SignalGenerator.cpp#L185, https://github.com/atanisoft/DCCppESP32/blob/master/src/SignalGenerator.cpp#L71)

By removing SPIFFS usage I haven't been able to reproduce any crashes related to the hardware timers.

closed 2019-08-15 by stale[bot]