espressif/arduino-esp32 #1784: esp32 crash core panic

moijerem59, 2018-08-22:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			20/08/2018?
IDE name:							Arduino IDE?
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Hello, when I open my serial port via the BT connection, it works, but this hatchback randomly. When the disconnection occurs the cr crash.
I can not solve this problem. It comes from my code or esp code
Thank you for your expertise

### Debug Messages:
```
ASSERT_PARAM(32768 0), in rwbt.c at line 273

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

Core 0 register dump:

PC      : 0x40086751  PS      : 0x00060034  A0      : 0x8008a1cc  A1      : 0x3ffc05b0 

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054 

A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x80086751  A9      : 0x3ffc0590 

A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff 

A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005 

EXCVADDR: 0x00000000  LBEG    : 0x40086689  LEND    : 0x40086690  LCOUNT  : 0x00000000 

Core 0 was running in ISR context:

EPC1    : 0x401ca586  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40086751

 

Backtrace: 0x40086751:0x3ffc05b0 0x4008a1c9:0x3ffc05d0 0x4008a85b:0x3ffc05f0 0x40082041:0x3ffc0610 0x401ca583:0x00000000

 

Core 1 register dump:

PC      : 0x401ca586  PS      : 0x00060734  A0      : 0x800927f2  A1      : 0x3ffd7f70 

A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffd7aa8 

A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc9b98  A9      : 0x3ffc9b5c 

A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80090c50  A13     : 0x3ffdac50 

A14     : 0x00000000  A15     : 0x3ffd7c20  SAR     : 0x00000000  EXCCAUSE: 0x00000005 

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000 

 

Backtrace: 0x401ca586:0x3ffd7f70 0x400927ef:0x3ffd7f90

 

Rebooting...
```

copercini, 2018-08-22:
If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

moijerem59, 2018-08-22:
Thank you for the answer, in the decoder I stick Integrity or only a party?``

moijerem59, 2018-08-22:
`ASSERT_PARAM(32768 0), in rwbt.c at line 273
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40086751  PS      : 0x00060034  A0      : 0x8008a1cc  A1      : 0x3ffc05b0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x80086751  A9      : 0x3ffc0590  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40086689  LEND    : 0x40086690  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x401ca586  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40086751

Backtrace: 0x40086751:0x3ffc05b0 0x4008a1c9:0x3ffc05d0 0x4008a85b:0x3ffc05f0 0x40082041:0x3ffc0610 0x401ca583:0x00000000

Core 1 register dump:
PC      : 0x401ca586  PS      : 0x00060b34  A0      : 0x800927f2  A1      : 0x3ffd7f70  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffd7aa8  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc9b98  A9      : 0x3ffc9b5c  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000029  A13     : 0x3ffc4df4  
A14     : 0x00000000  A15     : 0x3ffd7c20  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x401ca586:0x3ffd7f70 0x400927ef:0x3ffd7f90

Rebooting...
`

moijerem59, 2018-08-22:
Backtrace: 0x40086751:0x3ffc05b0 0x4008a1c9:0x3ffc05d0 0x4008a85b:0x3ffc05f0 0x40082041:0x3ffc0610 0x401ca583:0x00000000

decode 

Decoding stack results
0x401ca583: esp_vApplicationWaitiHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 66

moijerem59, 2018-08-22:
Backtrace: 0x401ca586:0x3ffd7f70 0x400927ef:0x3ffd7f90

decode 

Decoding stack results

0x401ca586: esp_vApplicationWaitiHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 66
0x400927ef: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 3492

lbernstone, 2018-08-22:
Any code called called as an interrupt callback should be in IRAM_ATTR context (see https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIOInterrupt/GPIOInterrupt.ino).  The code in interrupts needs to be quick, and all variables need to be available to the context.  Set flag variables for long running code rather than executing it in the interrupt.

no5killz, 2018-08-26:
I was facing the same issue. I believe this is related https://github.com/espressif/esp-idf/issues/2070. I build the project now as idf project with disabled modem sleep and it seems to work.

closed 2018-08-27 by moijerem59