espressif/arduino-esp32 #1785: esp32 - 50mA minimum.

Stasikiii, 2018-08-22:
Hello

I did radio off by:

#include <WiFi.h>
void setup() {
  WiFi.mode(WIFI_OFF);
  btStop();
}

but esp32 still consume 50mA of currency.
In datasheet it's described as a 10mA in modem sleep mode.

What wrong?

copercini, 2018-08-22:
1- By default the Arduino Core runs always at the maximum speed, so you'll always be in the worst case
2- This table consider only the ESP32, if you are using any dev board, there are more components than the ESP32...

Stasikiii, 2018-08-22:
1 - How to settings this on medium? :)

copercini, 2018-08-22:
https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-367867856

It's not very simple, but it's possible

lbernstone, 2018-08-22:
https://github.com/lbernstone/arduino-esp32-80MHz

Stasikiii, 2018-08-23:
You mean i can change it only in a esp-idf?

copercini, 2018-08-23:
@Stasikiii yes, it can only be change using IDF, but the result will produce the @lbernstone file, so using https://github.com/lbernstone/arduino-esp32-80MHz procedure you can skip the IDF proccess

Stasikiii, 2018-08-23:
@copercini thx you!

lbernstone, 2018-08-23:
Maybe you need to cool it to absolute zero :smile: .  The testing is done in isolation, without any flash, lights, or gpio traces.  You are working in the real world, with a real OS running.  esp32-solo-1 modules should be arriving in stores soon, which will be as efficient as you can go with esp32.  If you need less power, you get less power...

copercini, 2018-08-23:
Another tips: if you remove the USB to serial converter, LEDs, switch to a better LDO (if you are using one), remove everything except the ESP, I think you can get the 10mA