espressif/arduino-esp32 #1786: OTA Update from S3 Crash

thehellmaker, 2018-08-22:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:	        11/jul/2017
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I have a guru meditation error
Decoded Stack Trace is here:
0x400d9a34: WiFiClientRxBuffer::fillBuffer() at ?? line ?
0x400d9a34: WiFiClientRxBuffer::fillBuffer() at ?? line ?
0x400d9ad4: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at ?? line ?
0x4016ff8e: aws_iot_mqtt_internal_write_len_to_buffer at ?? line ?
0x400e0401: Print::printf(char const*, ...) at ?? line ?
0x400e042d: Print::println() at ?? line ?
0x400dba32: Preferences::getString(char const*, String) at ?? line ?
0x400d60e6: doOTA() at ?? line ?
0x400d39e0: EventsIncoming::handleOTAUpdateEvent(String) at ?? line ?
0x400d3532: mySubCallBackHandler(String, String) at ?? line ?
0x400d22f6: iot_subscribe_callback_handler(_Client*, char*, unsigned short, IoT_Publish_Message_Params*, void*) at ?? line ?
0x400d8292: aws_iot_mqtt_internal_cycle_read at ?? line ?
0x400d8d39: aws_iot_mqtt_resubscribe at ?? line ?
0x400d1e5d: _flash_cache_start at ?? line ?

The code is kinda huge to paste here so i'm gonna give the pseudo code here
1. Subscribe it AWS IOT
2. On MQTT message store S3 path and bucket
3. In loop if the message from MQTT has arrived then start OTA update similar to 
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino

lbernstone, 2018-08-22:
You need to post the error along with the backtrace.  Does that example code work properly for OTA without your code involved?

thehellmaker, 2018-08-24:
I used the code mentioned in the link below without any changes
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino

Stack Dump
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2ed0  PS      : 0x00060e30  A0      : 0x800d2f73  A1      : 0x3ffb1dc0
A2      : 0x3ffb9cd4  A3      : 0x3ffb1e0f  A4      : 0x00000001  A5      : 0x40000000
A6      : 0x00045f3c  A7      : 0x444078f8  A8      : 0x00000000  A9      : 0xffffffff
A10     : 0x02ffffff  A11     : 0x3ffba0c4  A12     : 0x00000050  A13     : 0x00000000
A14     : 0x02ffffff  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000003
EXCVADDR: 0x40000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x400d2ed0:0x3ffb1dc0 0x400d2f70:0x3ffb1de0 0x40142462:0x3ffb1e00 0x400d82cd:0x3ffb1e30 0x400d82f9:0x3ffb1e50 0x400d46f6:0x3ffb1e70 0x400d2861:0x3ffb1e90 0x400d2c52:0x3ffb
1f60 0x40126323:0x3ffb1fa0

Decoded Stack Trace:
Decoding 14 results
0x400d2ed0: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at ?? line ?
0x400d2ed0: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at ?? line ?
0x400d2f70: WiFiClient::read(unsigned char*, unsigned int) at ?? line ?
0x40142462: WiFiClient::read() at ?? line ?
0x400d82cd: Stream::timedRead() at ?? line ?
0x400d82f9: Stream::readBytes(char*, unsigned int) at ?? line ?
0x400d46f6: UpdateClass::writeStream(Stream&) at ?? line ?
0x400d2861: execOTA() at ?? line ?
0x400d2c52: setup() at ?? line ?
0x40126323: loopTask(void*) at ?? line ?

lbernstone, 2018-08-24:
Two things:
Turn on verbose core debug by setting `build_flags = -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG` in your platform.ini.
Make sure you can download the file using curl/wget (although this shouldn't cause a failure).  Also add `Serial.setDebugOutput(true);` just after your Serial.begin.

I've never seen it do that thing with the question marks instead of line numbers.  Are you using the latest version of platform.io/arduino-esp32?

arilarralde, 2018-12-05:
Hi! any solution? I have the same problem

closed 2019-08-15 by stale[bot]