espressif/arduino-esp32 #1789: Cannot turn off serial debug messages for ESP-NOW

Crypter, 2018-08-23:
The problem occurs if I try to send packets to unreachable device "too often" (is there such thing?). So, I assume I'm filling the TX Queue to its max?

If the receiving end is reachable then the sender has no problems with packets being sent every 5 ms, but if it's unreachable any delay less than ~20ms leads to this. Delay of 25 between sends doesn't snow any error.

The packets are less than 6 bytes of data.
I have set Core debug level: none
The moment the device becomes unreachable this is what I get on Serial terminal:

`E (20295) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20315) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20335) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20355) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20365) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20385) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20405) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20425) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb
E (20435) wifi: esf_buf: t=3 l=48 max:8, alloc:8 no eb`

Now, I use the USB Serial port for communication, and I need these errors suppressed. I guess these need to be pushed a level up, to ERROR level, not "None" level?

Related issue: #371
and [Also found on esp32 forum](https://www.esp32.com/viewtopic.php?t=5367)

I can provide sample code if needed, but I think this proves enough.

Regards,
Crypter

lbernstone, 2018-08-23:
This is an upstream issue, but an easy workaround would be to `Serial.end()` and use Serial1 for your device communication.

Crypter, 2018-08-23:
That's not a workaround, I need USB communications, Otherwise I'll need to hook up another usb2serial adapter to Serial1.

None the less, I'm happy this is acknowledged, I'll wait for the fix in the libraries :)

Thx!

lbernstone, 2018-08-23:
You can remap Serial1 to use pins 1&3. `Serial1.begin(115200,SERIAL_8N1,3,1)`

copercini, 2018-08-25:
This is caused due the IDF was compiled to show error logs (below arduino layer), so I think the only way to solve this elegantly is using Arduino as IDF component and disable IDF and arduino logs

closed 2019-08-15 by stale[bot]