espressif/arduino-esp32 #1790: ESP32 Deep Sleep Examples not working as

JasperRowan, 2018-08-23:
### Hardware:
Board:							ESP32 DEVKIT V1 (www.doit.am)
Core Installation/update date:			18/jul/2018
IDE name:					Arduino IDE 1.8.5 (windows store 1.8.10.0) (Windoes 10)
Flash Frequency:					        80Mhz
Upload Speed:						115200

### Description:
I am trying to use DeepSleep to help with power consumption in a battery power project where I am using esp32s to monitor a sensor and report data back to a receiving esp32 using the ESP NOW protocol.
I tried to use the 3 examples under ESP32/DeepSleep to get an understanding of how DeepSleep works. I thought the most simple example was the TimerWakeUp, but once I uploaded it and checked the serial monitor I realized that none of the serial prints were being called. The board reset successfully every five seconds and the current dropped every sleep cycle from around 50 mA to 9mA so it seems to be partially working. 

(I have 5 of these ESP32 DEVKIT V1 boards (ordered altogether from Amazon) and I got the same result on all five... was also able to get ESP NOW working on them with no problem)

Thanks for your help 

-Jasper

### Sketch:
/*
Simple Deep Sleep with Timer Wake Up
=====================================
ESP32 offers a deep sleep mode for effective power
saving as power is an important factor for IoT
applications. In this mode CPUs, most of the RAM,
and all the digital peripherals which are clocked
from APB_CLK are powered off. The only parts of
the chip which can still be powered on are:
RTC controller, RTC peripherals ,and RTC memories

This code displays the most basic deep sleep with
a timer to wake it up and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  5        /* Time ESP32 will go to sleep (in seconds) */

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case 1  : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case 2  : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case 3  : Serial.println("Wakeup caused by timer"); break;
    case 4  : Serial.println("Wakeup caused by touchpad"); break;
    case 5  : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
  " Seconds");

  /*
  Next we decide what all peripherals to shut down/keep on
  By default, ESP32 will automatically power down the peripherals
  not needed by the wakeup source, but if you want to be a poweruser
  this is for you. Read in detail at the API docs
  http://esp-idf.readthedocs.io/en/latest/api-reference/system/deep_sleep.html
  Left the line commented as an example of how to configure peripherals.
  The line below turns off all RTC peripherals in deep sleep.
  */
  //esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  //Serial.println("Configured all RTC Peripherals to be powered down in sleep");

  /*
  Now that we have setup a wake cause and if needed setup the
  peripherals state in deep sleep, we can now start going to
  deep sleep.
  In the case that no wake up sources were provided but deep
  sleep was started, it will sleep forever unless hardware
  reset occurs.
  */
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This is not going to be called
}

### Debug Messages:

ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Boot ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Boot ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Boot ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Boot ets Jun  8 2016 00:22:57

This just loops forever

tonu42, 2018-08-23:
> The board reset successfully every five seconds and the current dropped every sleep cycle from around 50 mA to 9mA so it seems to be partially working.

So the sketch is working.

> This just loops forever

This is being called in the code each time the device wakes up:
print_wakeup_reason();

Although I think the messages you are seeing over and over again are merely the initial prints that the chip always does.

> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

Shows you that the device is being reset by the deepsleep reset proccess.

JasperRowan, 2018-08-23:
Ok,

thank you for your quick response!

My understanding is that this each time the esp32 wakes up it will run the
setup function. I am still confused about why the  print_wakeup_reason();
and  Serial.println("Going to sleep now");  are not printing to the serial
monitor. In the  print_wakeup_reason(); function each serial print begins
with "Wakeup caused by..." or "Wakeup was not caused by..." but I never see
that in the serial monitor.
I'm sorry if I'm missing something very obvious, I'm new to
microcontrollers in general but especially the esp32.

Thank you so much for your time.

-Jasper

lbernstone, 2018-08-23:
Put in a delay(100) right before the sleep.  There is not enough time to print everything to serial before it goes to sleep.

JasperRowan, 2018-08-24:
Thank you so much for your fast response. That solved the problem!

closed 2018-08-24 by JasperRowan

entropia1ac, 2018-08-28:
It does not work for me-no wake up. (Board: TTGO LORA32)
........
void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));
  print_wakeup_reason();

   Serial.flush();
   delay(1000);
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
  " Seconds");
}

.....

entropia1ac, 2018-08-28:
not work -no going to sleep. (Board: TTGO LORA32 ,stable release 1.0.0)
........
void setup(){
Serial.begin(115200);
delay(1000); //Take some time to open up the Serial Monitor
++bootCount;
Serial.println("Boot number: " + String(bootCount));
print_wakeup_reason();

Serial.flush();
delay(1000);
esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
" Seconds");
}
void loop(){
  Serial.println("Boot number in loop=" + String(bootCount));
   delay(1000);
}
.....
debug:==============
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Boot number: 1
Wakeup was not caused by deep sleep
Setup ESP32 to sleep for every 5 Seconds
Boot number in loop=1
Boot number in loop=1
Boot number in loop=1
Boot number in loop=1
Boot number in loop=1
Boot number in loop=1
Boot number in loop=1
Boot number in loop=1
Boot number in loop=1

Lejaco13, 2018-11-01:
You forget to put that:

esp_deep_sleep_start();

Lejaco13, 2018-11-01:
The better board for the low power consumption "DeepSleep" (10 μA) is from the manufacturer DFRobot.
FireBeetle ESP32

https://www.dfrobot.com/product-1590.html

PR #1791 (2018-09-17): Flush serial in DeepSleep example to allow print before sleep
Fixes #1790