espressif/arduino-esp32 #1793: Arduino IDE can't configure flash size 16M of ESP32-WROOM-32D

q-bird, 2018-08-25:
Hi 
I built a board with ESP32-WROOM-32D 16M flash
But in Arduino IDE the largest memory is just 4M
[image]

Then I tried to upload firmware to the ESP (16M) and it showed error
[image]
Based on that, the tool couldn't auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB

Then how can I config 16M in arduino IDE and let it know that I'm using ESP 16M, not 4M, so that I can feel free to write firmware? 

Thank you!

lbernstone, 2018-08-25:
@me-no-dev has been talking about what the right partition scheme should be for 16M, as SPIFFS is not really intended for multi-megabytes.  This will be added shortly.  If you want to take a stab, you can look through [boards.txt](https://github.com/espressif/arduino-esp32/blob/master/boards.txt) and add a flash size and partition scheme.

q-bird, 2018-08-25:
@lbernstone 
Thank you for your support!
But I still got the same problem after I added a new definition in boards.txt

```
esp32test.name=ESP32 16M

esp32test.upload.tool=esptool
esp32test.upload.maximum_size=8388608
esp32test.upload.maximum_data_size=2097152
esp32test.upload.wait_for_upload_port=true

esp32test.serial.disableDTR=true
esp32test.serial.disableRTS=true

esp32test.build.mcu=esp32
esp32test.build.core=esp32
esp32test.build.variant=esp32
esp32test.build.board=ESP32_FCU

esp32test.build.f_cpu=240000000L
esp32test.build.flash_size=16MB
esp32test.build.flash_freq=40m
esp32test.build.flash_mode=dio
esp32test.build.boot=dio
esp32test.build.partitions=max
esp32test.build.defines=

esp32test.menu.PSRAM.disabled=Disabled
esp32test.menu.PSRAM.disabled.build.defines=
esp32test.menu.PSRAM.enabled=Enabled
esp32test.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue

esp32test.menu.PartitionScheme.default=Default
esp32test.menu.PartitionScheme.default.build.partitions=max
esp32test.menu.PartitionScheme.minimal=Minimal (2MB FLASH)
esp32test.menu.PartitionScheme.minimal.build.partitions=minimal
esp32test.menu.PartitionScheme.no_ota=No OTA (Large APP)
esp32test.menu.PartitionScheme.no_ota.build.partitions=no_ota
esp32test.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
esp32test.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (Large APPS with OTA)
esp32test.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
esp32test.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080

esp32test.menu.FlashMode.qio=QIO
esp32test.menu.FlashMode.qio.build.flash_mode=dio
esp32test.menu.FlashMode.qio.build.boot=qio
esp32test.menu.FlashMode.dio=DIO
esp32test.menu.FlashMode.dio.build.flash_mode=dio
esp32test.menu.FlashMode.dio.build.boot=dio
esp32test.menu.FlashMode.qout=QOUT
esp32test.menu.FlashMode.qout.build.flash_mode=dout
esp32test.menu.FlashMode.qout.build.boot=qout
esp32test.menu.FlashMode.dout=DOUT
esp32test.menu.FlashMode.dout.build.flash_mode=dout
esp32test.menu.FlashMode.dout.build.boot=dout

esp32test.menu.FlashFreq.80=80MHz
esp32test.menu.FlashFreq.80.build.flash_freq=80m
esp32test.menu.FlashFreq.40=40MHz
esp32test.menu.FlashFreq.40.build.flash_freq=40m

esp32test.menu.FlashSize.16M=16MB (128Mb)
esp32test.menu.FlashSize.16M.build.flash_size=16MB
esp32test.menu.FlashSize.4M=4MB (32Mb)
esp32test.menu.FlashSize.4M.build.flash_size=4MB
esp32test.menu.FlashSize.2M=2MB (16Mb)
esp32test.menu.FlashSize.2M.build.flash_size=2MB
esp32test.menu.FlashSize.2M.build.partitions=minimal

esp32test.menu.UploadSpeed.921600=921600
esp32test.menu.UploadSpeed.921600.upload.speed=921600
esp32test.menu.UploadSpeed.115200=115200
esp32test.menu.UploadSpeed.115200.upload.speed=115200
esp32test.menu.UploadSpeed.256000.windows=256000
esp32test.menu.UploadSpeed.256000.upload.speed=256000
esp32test.menu.UploadSpeed.230400.windows.upload.speed=256000
esp32test.menu.UploadSpeed.230400=230400
esp32test.menu.UploadSpeed.230400.upload.speed=230400
esp32test.menu.UploadSpeed.460800.linux=460800
esp32test.menu.UploadSpeed.460800.macosx=460800
esp32test.menu.UploadSpeed.460800.upload.speed=460800
esp32test.menu.UploadSpeed.512000.windows=512000
esp32test.menu.UploadSpeed.512000.upload.speed=512000

esp32test.menu.DebugLevel.none=None
esp32test.menu.DebugLevel.none.build.code_debug=0
esp32test.menu.DebugLevel.error=Error
esp32test.menu.DebugLevel.error.build.code_debug=1
esp32test.menu.DebugLevel.warn=Warn
esp32test.menu.DebugLevel.warn.build.code_debug=2
esp32test.menu.DebugLevel.info=Info
esp32test.menu.DebugLevel.info.build.code_debug=3
esp32test.menu.DebugLevel.debug=Debug
esp32test.menu.DebugLevel.debug.build.code_debug=4
esp32test.menu.DebugLevel.verbose=Verbose
esp32test.menu.DebugLevel.verbose.build.code_debug=5
```

I also created a new partitions "max.csv" 
[image]
[image]

Did I miss something else? Please help me to solve this problem!

Many thanks!

q-bird, 2018-08-25:
Hi,
I made it done already, jst because the GPIO12 is pulling up in my board...
it's a little inconvenient when solder the ESP to pcb board and using GPIO12 as well as GPIO15 for other purpose.
Is there anyway to avoid this problem? everytime I need to update firmware will I have to remove the pulled-up resistor on GPIO12?
What about OTA? does GPIO12 affect the upload process in OTA mode?

Thank you!

lbernstone, 2018-08-25:
We really need a wiki. https://github.com/espressif/esp-idf/tree/release/v3.1/examples/storage/sd_card#note-about-gpio12

beegee-tokyo, 2018-08-26:
[ESP32-ARDUINO WIKI](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

q-bird, 2018-08-27:
Thanks @beegee-tokyo and @lbernstone 

I removed the pull-uped resistor at pin GPIO12, and enable the internal pull-up using firmware
And it's ok now

closed 2018-08-27 by q-bird