espressif/arduino-esp32 #1797: SPI.begin crashes with certain pins

anteph, 2018-08-25:
### Description:
Hi,

I'm running into some troubles when trying to initialize SPI for some pins that are not the default ones (please check the code below). 

Basically, upon calling the SPI.begin method with those pins, the program crashes with the errors below.

I'm trying to use some custom SPI pins to try to interact with a SD card in a AnalogLamb Fish32 board, which doesn't use the default ones. However, to confirm that it is not hardware related, I've also tested the code in a generic ESP32 dev board from eBay, and it still crashes.

Error backtrace:
0x400d0b41:0x3ffb1f40 0x7ffffffd:0x3ffb1f60 0x400d0964:0x3ffb1f80 0x400e5233:0x3ffb1fa0

After decoded:
0x400d0b41: spiAttachSCK at C:\Users\N\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-spi.c line 1026
0x400d0964: setup() at C:\Users\N\AppData\Local\Temp\arduino_modified_sketch_811833/sketch_aug25d.ino line 5
0x400e5233: loopTask(void*) at C:\Users\N\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 15

Full error:
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0b80  PS      : 0x00060730  A0      : 0x800d0a51  A1      : 0x3ffb1f40  
A2      : 0x0000003f  A3      : 0x00000006  A4      : 0x00000010  A5      : 0x00000000  
A6      : 0x3ffc1054  A7      : 0x00000006  A8      : 0x800d0b82  A9      : 0x3ffb1f20  
A10     : 0x00000006  A11     : 0x0000003f  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400d0f64  LEND    : 0x400d0fc6  LCOUNT  : 0x00000002  

### Hardware:
Board:							ESP32 Dev Module and AnalogLamb Fish32
Core Installation/update date:			23/ago/2018
IDE name:							Arduino IDE 1.8.5

### Sketch:
```cpp
#include "SPI.h"

void setup() {
  SPI.begin(6,7,11,10);//SCK,MISO,MOSI,SS    
  delay(1000);
}

void loop() {}```

stickbreaker, 2018-08-25:
@anteph gpio06-11 are used byte the builtin FLASH, you can't use them for other purposes.

Chuck.

anteph, 2018-08-25:
@stickbreaker thank you very much for the feedback :)

In that case I'm marking this as resolved and try to look for an alternative to interact with the SD card, although I think there's not much to do since some of the pins attached to it are in that range.

Best regards,
Nuno Santos

closed 2018-08-25 by anteph