espressif/arduino-esp32 #1798: Memory leak in Blue tooth serial with bluedroid

usrdes, 2018-08-25:
### Hardware:
Board:							ESP32-PICO-KIT Dev. Board
Core Installation/update date:			1 Aug 2018
IDE name:							Arduino 1.8.5
Flash Frequency:					80MHz
Upload Speed:						921600

### Description:
Blue tooth serial (Bluedroid) has memory leak. The sketch below is modified from serial to serial BT example. 
The modification is as follows:
1. set up timer for 1 sec timeout
2. every sec turn on or off the bloe tooth serial.
3. every time BTSerial is turned on or off print heap size
4. observe that the heap size keeps diminishing, and eventually the core crashes. 

### Sketch: Modified serial to serial  blue tooth example as follows:

//This example code is in the Public Domain (or CC0 licensed, at your option.)
//By Evandro Copercini - 2018
//
//This example creates a bridge between Serial and Classical Bluetooth (SPP)
//and also demonstrate that SerialBT have the same functionalities of a normal Serial

#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

volatile int switch_off_Bluetooth = 0;

// Timer set up
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
#define PRESCALER 50000
#define NUM_TO_COUNT_UP_TO_1_SEC  1600
#define COUNT_OF_SECS 1  
#define TIME_OUT COUNT_OF_SECS*NUM_TO_COUNT_UP_TO_1_SEC
volatile int timeout_occured=0;

void IRAM_ATTR onTimer() {
  portENTER_CRITICAL_ISR(&timerMux);
  timeout_occured = 1;
  portEXIT_CRITICAL_ISR(&timerMux);
}

void setup_timer() {
  timer = timerBegin(0, PRESCALER, true);
  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, TIME_OUT, true);
  timerAlarmEnable(timer);
}

void reset_timeout() {
    portENTER_CRITICAL(&timerMux);
    timeout_occured = 0;
    portEXIT_CRITICAL(&timerMux);
}

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
  setup_timer();
  Serial.printf("Timer for a count of %d seconds of BTserial ON/OFF started",COUNT_OF_SECS);
  Serial.println(" --> ");
}

void loop() {

    if(timeout_occured) { // Turn on or off blue tooth based on timer time-out of COUNT_OF_SECS
      reset_timeout(); // first reset time out flag
      if (switch_off_Bluetooth) { // check if the BT serial needs to be turned off
        switch_off_Bluetooth = 0;
        SerialBT.flush();
        SerialBT.end(); 
        Serial.print("@ ");
        Serial.print(millis()/1000);
        Serial.printf(" secs > BTserial OFF Heap Size: %d ",ESP.getFreeHeap());
        Serial.println(" "); 
      } else { // else switch on Bluetooth serial
        switch_off_Bluetooth = 1;
        SerialBT.begin("ESP32test"); //Bluetooth device name
        Serial.print("@ ");
        Serial.print(millis()/1000);
        Serial.printf(" secs > BTserial ON  Heap Size: %d ",ESP.getFreeHeap());
        Serial.println(" "); 
      }
      
    }
    
}

### Debug Messages:

The device started, now you can pair it with bluetooth!
Timer for a count of 1 seconds of BTserial ON/OFF started --> 
@ 1 secs > BTserial ON  Heap Size: 100308  
@ 2 secs > BTserial OFF Heap Size: 169984  
@ 3 secs > BTserial ON  Heap Size: 95212  
@ 4 secs > BTserial OFF Heap Size: 165340  
@ 5 secs > BTserial ON  Heap Size: 90616  
@ 6 secs > BTserial OFF Heap Size: 160696  
@ 7 secs > BTserial ON  Heap Size: 86028  
@ 8 secs > BTserial OFF Heap Size: 156060  
@ 9 secs > BTserial ON  Heap Size: 81340  
@ 10 secs > BTserial OFF Heap Size: 152332  
@ 11 secs > BTserial ON  Heap Size: 77612  
@ 12 secs > BTserial OFF Heap Size: 148608  
@ 13 secs > BTserial ON  Heap Size: 73912  
@ 14 secs > BTserial OFF Heap Size: 144884  
@ 15 secs > BTserial ON  Heap Size: 70180  
@ 16 secs > BTserial OFF Heap Size: 141160  
@ 17 secs > BTserial ON  Heap Size: 66508  
@ 18 secs > BTserial OFF Heap Size: 137436  
@ 19 secs > BTserial ON  Heap Size: 62740  
@ 20 secs > BTserial OFF Heap Size: 133708  
@ 21 secs > BTserial ON  Heap Size: 59036  
@ 22 secs > BTserial OFF Heap Size: 129984  
@ 23 secs > BTserial ON  Heap Size: 55296  
@ 24 secs > BTserial OFF Heap Size: 126264  
@ 25 secs > BTserial ON  Heap Size: 51552  
@ 26 secs > BTserial OFF Heap Size: 122536  
@ 27 secs > BTserial ON  Heap Size: 47884  
@ 28 secs > BTserial OFF Heap Size: 118816  
@ 29 secs > BTserial ON  Heap Size: 44172  
@ 30 secs > BTserial OFF Heap Size: 115092  
@ 31 secs > BTserial ON  Heap Size: 40404  
@ 32 secs > BTserial OFF Heap Size: 111368  
@ 33 secs > BTserial ON  Heap Size: 36664  
@ 34 secs > BTserial OFF Heap Size: 107644  
@ 35 secs > BTserial ON  Heap Size: 32948  
@ 36 secs > BTserial OFF Heap Size: 103920  
@ 37 secs > BTserial ON  Heap Size: 29224  
@ 38 secs > BTserial OFF Heap Size: 100196  
@ 39 secs > BTserial ON  Heap Size: 25504  
@ 40 secs > BTserial OFF Heap Size: 96472  
@ 41 secs > BTserial ON  Heap Size: 21804  
@ 42 secs > BTserial OFF Heap Size: 92748  
@ 43 secs > BTserial ON  Heap Size: 18076  
@ 44 secs > BTserial OFF Heap Size: 89024  
@ 45 secs > BTserial ON  Heap Size: 14384  
@ 46 secs > BTserial OFF Heap Size: 85300  
@ 47 secs > BTserial ON  Heap Size: 10696  
@ 48 secs > BTserial OFF Heap Size: 81576  
@ 49 secs > BTserial ON  Heap Size: 6928  
@ 50 secs > BTserial OFF Heap Size: 77852  
@ 53 secs > BTserial ON  Heap Size: 3112  
@ 53 secs > BTserial OFF Heap Size: 74132  
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060e30  A0      : 0x801139de  A1      : 0x3ffdaf90 
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000001c  A5      : 0x00000000 
A6      : 0x40179f3c  A7      : 0x00000001  A8      : 0x800d70bd  A9      : 0x3ffdaf50 
A10     : 0x3ffe8ca8  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffe7370 
A14     : 0x3ffe15f4  A15     : 0x00000084  SAR     : 0x00000008  EXCCAUSE: 0x0000001d 
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000 
Backtrace: 0x4000c46c:0x3ffdaf90 0x401139db:0x3ffdafa0 0x401142b1:0x3ffdafc0 
0x400ef8f5:0x3ffdafe0 0x400ef92f:0x3ffdb000
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4

stickbreaker, 2018-08-25:
@usrdes please edit your message, add the [Markdown symbols](https://guides.github.com/features/mastering-markdown/) for code ( three back ticks then c++ on a line before your code example, and three back ticks on a separate line after your code).  You code should look like this:

```c++
void setup(){
// important stuff here
}
```

copercini, 2018-08-26:
1 problem found: - The queue is not being deleted at end, to fix it's just add
`vQueueDelete(_spp_queue);`

inside `void BluetoothSerial::end()` of BluetoothSerial.cpp

this reduces the leak to just 96 bytes each iteration

copercini, 2018-08-26:
Problem 2 found: it's missing a call to `esp_spp_deinit();`

Now it seems stable =)

```
The device started, now you can pair it with bluetooth!
 --> 
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
@ 1 secs > BTserial ON  Heap Size: 127172  
[0;31mE (2749) BT_APPL: JVenable fails[0m
[0;31mE (3750) BT_APPL: bta_jv_disable[0m
@ 2 secs > BTserial OFF Heap Size: 153888  
@ 3 secs > BTserial ON  Heap Size: 127736  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (5990) BT_APPL: bta_jv_disable[0m
@ 4 secs > BTserial OFF Heap Size: 153888  
@ 5 secs > BTserial ON  Heap Size: 127736  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (8229) BT_APPL: bta_jv_disable[0m
@ 7 secs > BTserial OFF Heap Size: 153888  
@ 8 secs > BTserial ON  Heap Size: 127736  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (10468) BT_APPL: bta_jv_disable[0m
@ 9 secs > BTserial OFF Heap Size: 153888  
@ 10 secs > BTserial ON  Heap Size: 127736  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (12707) BT_APPL: bta_jv_disable[0m
@ 11 secs > BTserial OFF Heap Size: 153740  
@ 12 secs > BTserial ON  Heap Size: 127616  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (14993) BT_APPL: bta_jv_disable[0m
@ 13 secs > BTserial OFF Heap Size: 153740  
@ 14 secs > BTserial ON  Heap Size: 127592  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (17232) BT_APPL: bta_jv_disable[0m
@ 16 secs > BTserial OFF Heap Size: 153740  
@ 17 secs > BTserial ON  Heap Size: 127616  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (19471) BT_APPL: bta_jv_disable[0m
@ 18 secs > BTserial OFF Heap Size: 153740  
@ 19 secs > BTserial ON  Heap Size: 127592  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (21710) BT_APPL: bta_jv_disable[0m
@ 20 secs > BTserial OFF Heap Size: 153740  
@ 21 secs > BTserial ON  Heap Size: 127616  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (23949) BT_APPL: bta_jv_disable[0m
@ 22 secs > BTserial OFF Heap Size: 153592  
@ 23 secs > BTserial ON  Heap Size: 127476  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (26234) BT_APPL: bta_jv_disable[0m
@ 25 secs > BTserial OFF Heap Size: 153592  
@ 26 secs > BTserial ON  Heap Size: 127476  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (28476) BT_APPL: bta_jv_disable[0m
@ 27 secs > BTserial OFF Heap Size: 153592  
@ 28 secs > BTserial ON  Heap Size: 127476  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[... 67 lines cut ...]
@ 58 secs > BTserial OFF Heap Size: 153592  
@ 59 secs > BTserial ON  Heap Size: 127448  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (62222) BT_APPL: bta_jv_disable[0m
@ 61 secs > BTserial OFF Heap Size: 153592  
@ 62 secs > BTserial ON  Heap Size: 127448  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (64462) BT_APPL: bta_jv_disable[0m
@ 63 secs > BTserial OFF Heap Size: 153592  
@ 64 secs > BTserial ON  Heap Size: 127448  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (66701) BT_APPL: bta_jv_disable[0m
@ 65 secs > BTserial OFF Heap Size: 153592  
@ 66 secs > BTserial ON  Heap Size: 127448  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
[0;31mE (68941) BT_APPL: bta_jv_disable[0m
@ 67 secs > BTserial OFF Heap Size: 153592  
@ 68 secs > BTserial ON  Heap Size: 127448  
[I][BluetoothSerial.cpp:54] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:69] esp_spp_cb(): ESP_SPP_START_EVT
```

copercini, 2018-08-26:
this `�[0;31mE (68941) BT_APPL: bta_jv_disable�[0m` log comes from here https://github.com/espressif/esp-idf/blob/b85e8e10b90ab7089ec7989ba513f7a2235f9ea4/components/bt/bluedroid/bta/jv/bta_jv_act.c#L666 and seems normal

I think it's fixed now, I'll PR the changes

usrdes, 2018-08-26:
@stickbreaker : Thank you for letting me know how to add Markdown symbols, I have updated code above.

@copercini : Thank you for fixing, but I do not understand what PR means (Pull Request?) Also how about that log output you show in the link. Should that or will that need to be fixed as well? Please update here when is a good time to re-install (or update)  the ESP32 arduino blue tooth serial library.
Again thank you very much!

copercini, 2018-08-26:
@usrdes yes, the pull request with all fixes is here: https://github.com/espressif/arduino-esp32/pull/1801

it still need some review and merge to the main code

if you really want to use it right now, you can get it here with the fixes: https://github.com/copercini/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.cpp and replace this file in your installation

usrdes, 2018-08-26:
@copercini , OK thank you, I will close this issue,

usrdes, 2018-08-26:
@copercini : Sad to re-open but after updating the library cpp file, and recompiling  I am still getting leak. I made sure that the new file was being used by deliberately introducing a compile error in the cpp, and removing the deliberate error. 

Do I need to update anything else?

Here is the console output:
The device started, now you can pair it with bluetooth!
Timer for a count of 1 seconds of BTserial ON/OFF started --> 
@ 1 secs > BTserial ON  Heap Size: 100688  
@ 2 secs > BTserial OFF Heap Size: 171572  
@ 3 secs > BTserial ON  Heap Size: 96824  
@ 4 secs > BTserial OFF Heap Size: 168300  
@ 5 secs > BTserial ON  Heap Size: 93544

copercini, 2018-08-26:
this remaining leak seem related with timer code, please test with this code (without times):
```cpp
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;
volatile int switch_off_Bluetooth = 0;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
  Serial.println(" --> ");
  delay(1000);
}

void loop() {
  if (switch_off_Bluetooth) { // check if the BT serial needs to be turned off
    switch_off_Bluetooth = 0;
    SerialBT.flush();
    SerialBT.end();
    Serial.print("@ ");
    Serial.print(millis() / 1000);
    Serial.printf(" secs > BTserial OFF Heap Size: %d ", ESP.getFreeHeap());
    Serial.println(" ");
  } else { // else switch on Bluetooth serial
    switch_off_Bluetooth = 1;
    SerialBT.begin("ESP32test"); //Bluetooth device name
    Serial.print("@ ");
    Serial.print(millis() / 1000);
    Serial.printf(" secs > BTserial ON  Heap Size: %d ", ESP.getFreeHeap());
    Serial.println(" ");
  }
  delay(1000);
}
```

usrdes, 2018-08-26:
@copercini ,

Tried your code on two devices. Results below,
In both cases, the ESP32 simply stops, not even the "Guru panic ..."

Device #1----------------------------------------

The device started, now you can pair it with bluetooth!
 --> 
@ 1 secs > BTserial ON  Heap Size: 100856  
@ 2 secs > BTserial OFF Heap Size: 171740  
@ 3 secs > BTserial ON  Heap Size: 97048  
@ 4 secs > BTserial OFF Heap Size: 168468  
@ 6 secs > BTserial ON  Heap Size: 93780  
@ 7 secs > BTserial OFF Heap Size: 165196  
@ 8 secs > BTserial ON  Heap Size: 90504  
@ 9 secs > BTserial OFF Heap Size: 161924  
@ 10 secs > BTserial ON  Heap Size: 86844  
@ 11 secs > BTserial OFF Heap Size: 158652  
@ 12 secs > BTserial ON  Heap Size: 83964  
@ 14 secs > BTserial OFF Heap Size: 155232  
@ 15 secs > BTserial ON  Heap Size: 80548  
@ 16 secs > BTserial OFF Heap Size: 151960  
@ 17 secs > BTserial ON  Heap Size: 77220  
@ 18 secs > BTserial OFF Heap Size: 148688  
@ 19 secs > BTserial ON  Heap Size: 73948  
.......
.......
@ 63 secs > BTserial ON  Heap Size: 11832  
@ 65 secs > BTserial OFF Heap Size: 83244  
@ 66 secs > BTserial ON  Heap Size: 8560  
@ 67 secs > BTserial OFF Heap Size: 79972  
@ 68 secs > BTserial ON  Heap Size: 5280  
@ 69 secs > BTserial OFF Heap Size: 76700  
@ 72 secs > BTserial ON  Heap Size: 1908  
@ 73 secs > BTserial OFF Heap Size: 73428  
@ 77 secs > BTserial ON  Heap Size: 716  

Device #2 -----------------------------

load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
E (207) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

The device started, now you can pair it with bluetooth!
 --> 
@ 1 secs > BTserial ON  Heap Size: 99076  
@ 2 secs > BTserial OFF Heap Size: 170544  
E (3527) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 3 secs > BTserial ON  Heap Size: 95396  
@ 4 secs > BTserial OFF Heap Size: 167272  
E (5842) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 6 secs > BTserial ON  Heap Size: 92072  
@ 7 secs > BTserial OFF Heap Size: 163996  
E (8158) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 8 secs > BTserial ON  Heap Size: 88800  
@ 9 secs > BTserial OFF Heap Size: 160728  
E (10473) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 10 secs > BTserial ON  Heap Size: 85528  
@ 12 secs > BTserial OFF Heap Size: 157452  
E (12789) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 13 secs > BTserial ON  Heap Size: 82256  
@ 14 secs > BTserial OFF Heap Size: 154184  
E (15104) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 15 secs > BTserial ON  Heap Size: 78984  
@ 16 secs > BTserial OFF Heap Size: 150908  
E (17419) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 17 secs > BTserial ON  Heap Size: 75712  
@ 19 secs > BTserial OFF Heap Size: 147640  
E (19735) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

 .
 .
 .
 .

@ 65 secs > BTserial ON  Heap Size: 10272  
@ 66 secs > BTserial OFF Heap Size: 82200  
E (66069) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 67 secs > BTserial ON  Heap Size: 7028  
@ 68 secs > BTserial OFF Heap Size: 78924  
E (68383) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 71 secs > BTserial ON  Heap Size: 3680  
@ 73 secs > BTserial OFF Heap Size: 75656  
E (72677) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 76 secs > BTserial ON  Heap Size: 364  
@ 77 secs > BTserial OFF Heap Size: 72380  
E (76969) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

copercini, 2018-08-27:
@usrdes what is you ESP32 core version? did you get it from arduino boards manager or directly from git?

usrdes, 2018-08-28:
@copercini 
I followed instructions at : 
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md
on 8/12/2018.
So I figure that makes it the latest core version ..??
Which file in the install contains this ESP32 core version info?

usrdes, 2018-08-28:
@me-no-dev : Hi Can you please re-open this issue? As far as I can tell, it is not yet resolved. Thank you.

copercini, 2018-08-28:
@usrdes the issue was closed automatically when code was merged...

I'm using the boards manager installation https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md and no leaks here

usrdes, 2018-08-29:
@copercini 
OK, I used the board manager in the link you posted and updated my ESP32. Confirmed that the BluetoothSerial.cpp patch was included. Then I tried with the two devices, loading the compiled code once as ESP32 pico kit and once as ESP32 dev module. The memory leak is still there.
In the beginning of the run, the heap difference every "ON" is ~5K bytes, towards the end, i.e. before the crash it is ~3.8K bytes.  The output is listed below.

Any suggestions on what to try next?
Thank you
--------------------------- Output --------------------
Device 1 loaded as ESP32 pico kit

------------------
The device started, now you can pair it with bluetooth!
 --> 
@ 1 secs > BTserial ON  Heap Size: 100800  
@ 2 secs > BTserial OFF Heap Size: 170460  
@ 3 secs > BTserial ON  Heap Size: 95720  
@ 4 secs > BTserial OFF Heap Size: 165824  

   |
   |
   |
   |

@ 56 secs > BTserial ON  Heap Size: 7112  
@ 58 secs > BTserial OFF Heap Size: 78060  
@ 61 secs > BTserial ON  Heap Size: 3296  
@ 62 secs > BTserial OFF Heap Size: 74336  
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060430  A0      : 0x80113686  A1      : 0x3ffdadb0 
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000001c  A5      : 0x00000000 
A6      : 0x4017994c  A7      : 0x00000001  A8      : 0x800d6d81  A9      : 0x3ffdad70 
A10     : 0x3ffe82d4  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffe6b4c 
A14     : 0x3ffe15f4  A15     : 0x00000084  SAR     : 0x00000008  EXCCAUSE: 0x0000001d 
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000 
Backtrace: 0x4000c46c:0x3ffdadb0 0x40113683:0x3ffdadc0 0x40113f59:0x3ffdade0 
0x400ef5b9:0x3ffdae00 0x400ef5f3:0x3ffdae20
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
------------------

Device 2 loaded as ESP32 pico kit

--------------------

E (217) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

The device started, now you can pair it with bluetooth!
 --> 
@ 1 secs > BTserial ON  Heap Size: 98948  
@ 2 secs > BTserial OFF Heap Size: 169068  
E (3532) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

@ 3 secs > BTserial ON  Heap Size: 93916  
@ 4 secs > BTserial OFF Heap Size: 164444  
E (5851) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)

   |
   |
   |
   |
[... 111 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060c30  A0      : 0x80113686  A1      : 0x3ffdb2e0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000001c  A5      : 0x00000000  
A6      : 0x4017994c  A7      : 0x00000001  A8      : 0x800d6d81  A9      : 0x3ffdb2a0  
A10     : 0x3ffe783c  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffe6768  
A14     : 0x3ffded44  A15     : 0x00000084  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4000c46c:0x3ffdb2e0 0x40113683:0x3ffdb2f0 0x40113f59:0x3ffdb310 0x400ef5b9:0x3ffdb330 0x400ef5f3:0x3ffdb350

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310

------------------------

usrdes, 2018-09-01:
@copercini 

Attempting to update my arduino install:
First I uninstalled ESP32, cleaned up any remaining folders
Uninstalled Arduino,  again cleaned up any remaining folders
Updated my Java installation, so I have the latest environment.
Re-installed Arduino,
Re-installed ESP32.

Result:
  The latest install did not get the Blue tooth serial updates. Further the Arduino boards manages shows that the ESP32 is version 1.0.0.   So I am wondering if the version numbers are getting updated at all?

Anyway, I then downloaded the "zip" folder from github, to compare release dates and find some
discrepancies. I have attached a ms-paint picture, showing some of the folder structures and file
dates etc.

Can you please tell me if my arduino install is a problem?

Thanks

[image]

lbernstone, 2018-09-01:
Boards Manager has full releases only.  It gets updated infrequently. Development happens in the git repository.  Instructions are provided on the main page for installing the dev repo.  Note that if you install both, you will have two sets of ESP32 boards in your IDE list.

usrdes, 2018-09-01:
@lbernstone 
Hi, thank you for the quick response. Also forgive me for I do not understand your statements:
Please clarify:
1. "Boards Manager has full releases only." --> is this for fresh clean complete release install ?
2.  "Instructions are provided on the main page for installing the dev repo. " --> is the "dev repo" some other place than github?

I followed the instructions below, after I completely wiped out my previous install.
So it should be up to date... right? Please help,. sorry that I do not understand your comments.
Thank you

[image]

stickbreaker, 2018-09-01:
@usrdes 
If you want to use the current development branch you have to use the instructions for the repository:
[image]

Chuck.

usrdes, 2018-09-02:
@copercini 
I cleaned out my installs, and re-installed from development repository.
Checked that your updates to blue tooth serial is included.
Still memory leak exists.

I suspect that the core blue droid code may have an issue.

IonicBob, 2018-09-13:
No update or resolution? As of Sep 12, I also see exactly the same memory leak, results as reported above (core panic after 63 iterations) using a ESP32 Dev Module and the latest dev branch and bluetoothserial.cpp update.  I got here looking for resolution a random reset  which occurs several times a day in a simple streaming bluetooth classic sketch.   Is the the ESP32 bluetooth not yet ready for prime time?  I notice a major update for bluetooth in ESP-IDF V3.1.  Does this need to ripple through Arduino-ESP32?

usrdes, 2018-09-19:
@IonicBob 
As far as I can tell, bluetoothserial.cpp is a wrapper for calls to bluedroid, as can be seen in the file:
#if defined(CONFIG_BT_ENABLED) && defined(CONFIG_BLUEDROID_ENABLED)

While copercini appears to have fixed some of the open / close of memory leaks, 
there is still some leak in the blue droid which is part of the core code that runs on esp32.

Don't know if the ESP-IDF V3.1 fix will take care of it or not, but for sure, it needs to ripple through to Arduino-ESP32.

copercini, 2018-09-22:
@usrdes @IonicBob could you guys try cloning the last git code? (instructions [here](https://github.com/espressif/arduino-esp32/tree/master/docs/arduino-ide)
I think this was fixed by me-no-dev here https://github.com/espressif/arduino-esp32/commit/96822d783f3ab6a56a69b227ba4d1a1a36c66268

IonicBob, 2018-09-24:
I tried this 3 times.  Upgrade and then a fresh install of the last git code using instructions. Same result in all cases.  This behaves a little different than before.
...
@ 71 secs > BTserial OFF Heap Size: 146812  
@ 72 secs > BTserial ON  Heap Size: 68056  
@ 74 secs > BTserial OFF Heap Size: 143540  
@ 75 secs > BTserial ON  Heap Size: 64784  
@ 76 secs > BTserial OFF Heap Size: 140264  
@ 77 secs > BTserial ON  Heap Size: 61500  
@ 78 secs > BTserial OFF Heap Size: 136992  
@ 79 secs > BTserial ON  Heap Size: 58228  
@ 80 secs > BTserial OFF Heap Size: 133720  
@ 83 secs > BTserial ON  Heap Size: 54820  
@ 85 secs > BTserial OFF Heap Size: 130448  
@ 88 secs > BTserial ON  Heap Size: 52956  
@ 89 secs > BTserial OFF Heap Size: 127176  
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4012f4a5  PS      : 0x00060230  A0      : 0x80111bbf  A1      : 0x3ffd5bf0  
A2      : 0x00000000  A3      : 0x3fffff0c  A4      : 0xffffff80  A5      : 0x3ffd5ed0  
A6      : 0x3fffff14  A7      : 0x00001800  A8      : 0x8012f494  A9      : 0x3ffd5be0  
A10     : 0x3ffd5bf4  A11     : 0x00000000  A12     : 0x00000268  A13     : 0x3ffd5e5c  
A14     : 0x00000020  A15     : 0x00000026  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4012f4a5:0x3ffd5bf0 0x40111bbc:0x3ffd5e80 0x40114b82:0x3ffd5ea0 0x400efdaa:0x3ffd5ec0 0x4008c821:0x3ffd5ef0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
...

ConcepcionMota, 2018-11-26:
I have one problem with the BluetoothSerial.h, when i compile the project arduino says "error: 'EventGroupHandle_t' does not name a type

 static EventGroupHandle_t _spp_event_group = NULL;", please someone help me

piotrkundu, 2019-06-19:
Is this still a problem? Any updates?

IonicBob, 2019-06-20:
> Is this still a problem? Any updates?

Can't say, I needed an immediate resolution and switched back to an uart attached HC5/6 and its been running nonstop ever since. Having said that I believe there's been a lota work done on the libraries and its workwhile trying again. I don't have an immediate test system at hand, and would encourage you or someone else to re-run the BT test and close this issue if it now works.

closed 2019-09-02 by stale[bot]