espressif/arduino-esp32 #1799: ESP8266 allows flash split to have 3MB of SPIFFS, ESP32 only allows up to 1.5MB of SPIFFS in the arduino GUI

marcmerlin, 2018-08-26:
Can you change the hardware configuration for ESP32 in arduino so that the compile menu options also allow splitting the flash so that more than 1.5MB out of 4MB is usable for SPIFFS? (default is 1MB, 'No OTA" boost this to only 1.5MB)
This is true as of today with arduino-esp32 from git, and arduino 1.8.6 pre release.

francescolavra, 2018-10-28:
You can change your csv partition file without blowing up things as long as you don't touch the nvs and otadata partitions and as long as the app0 partition starts at address 0x10000.

marcmerlin, 2018-12-10:
Thanks, it would however be nicer if the GUI allowed more splits or at least allowed a split as big as the lesser ESP8266 chip.

me-no-dev, 2018-12-12:
Awww... it's been a while @marcmerlin :) feel free to come up with better partition schemes and make a PR :P Surely we can have as many partition schemes as we want. Just someone with need needs to pus some time. Reason is that unless we hit a particuylar limitation that can not be handled by one of the existing ones, nobody knows what to create.

marcmerlin, 2018-12-14:
Hi @me-no-dev  yes, I work a lot on something for a while, and then disappear for a year or so as I move to one of my too many other hobbies ;)   I'll see if I can make a proposal for this.

marcmerlin, 2019-03-25:
With the latest git, "large app, no OTA" gives 2MB of spiffs, which is not bad. The only problem is how non obvious how much or little spiffs you get from each of the options.
Now, I have confirmed that reading animated gifs from that spiffs causes too many hangs for my use, so I need to switch to fatfs and hope that works better.

marcmerlin, 2019-03-28:
@me-no-dev would it be possible to simply show the amount of program flash vs spiffs room for each compile option (like minimal: 1M/1M, No OTA 2M/2M, Huge App: 3M/1M, etc...)

marcmerlin, 2019-03-30:
@me-no-dev @lbernstone better late than never I guess, this PR addresses this issue (and then some) as well as adds some ffat options to esp32-dev:
https://github.com/espressif/arduino-esp32/pull/2623

closed 2019-04-26 by marcmerlin