espressif/arduino-esp32 #1802: Recieve more than 20 Bytes from Android App with Ble Client

Tombo89, 2018-08-26:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			26/Aug/2018
IDE name:							Arduino IDE

### Description:
Hello,
It's not really an Issue I'm Posting, more a Question about the Ble Client example Code. I used this Code Succesfull but can only recieve 20 bytes. I need to recieve more from my Android app. The Documentation says I need to do a subsequent read request with shifted offsets. Maybe someone can help me how to do this? I'm not such an experienced Coder.

yoursunny, 2018-08-26:
You'll need to set a higher MTU on both BLE client and BLE server.
I had one project using ESP32 as BLE server. The maximum supported MTU on ESP32 is 517.
https://github.com/yoursunny/esp8266ndn/blob/5cc7034274f023d896fce7c1ec1ef167c489f3a4/src/transport/ble-server-transport.cpp#L26
My client is written with bluepy library. I don't know Android.
https://github.com/yoursunny/esp8266ndn/blob/5cc7034274f023d896fce7c1ec1ef167c489f3a4/extras/BlePingClient.py#L31

Tombo89, 2018-08-26:
Hello,
he arod a is rite by a company and their documentation says i should start a shiftet request. I already increased the mtu but i dont know how to form a command for the shiftet offset.

If I'm right these two parts should be the parts for getiting the data
` // Read the value of the characteristic.
    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());

    pRemoteCharacteristic->registerForNotify(notifyCallback);``

> static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}`

chegewara, 2018-09-13:
@yoursunny You are right, but from this quote i understand that its required different approach. 

@Tombo89 You have to read data from characteristic in loop until data is less than 20 bytes and concatenate all received data.

closed 2019-08-15 by stale[bot]