espressif/arduino-esp32 #1803: BluetoothSerial SerialBT excample loses data without delay(20)

D-140480, 2018-08-27:
### Hardware:
Board:							        Sparkfun ESP32 Thing
Core Installation/update date:			Jun  8 2016
IDE name:							Arduino IDE 1.8.6
Flash Frequency:					        80Mhz
Upload Speed:						921600

### Description:
When using the example code https://github.com/espressif/arduino-esp32/blob/901a341949f23afe54bc6fc5c6617974ff9b36a9/libraries/BluetoothSerial/examples/SerialToSerialBT/SerialToSerialBT.ino
if you take out the delay of 20 milliseconds, the data sent from the Arduino IDE serial monitor is sporadically truncated.  Why?

```cpp
#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
//  delay(20);
}
```

### Debug Messages:
```
I could not see any "crashes" or errors, just incorrect serial output when the delay was not there.
```

D-140480, 2018-08-28:
Lowering the delay to delay(2) still corrupts the data flow from the Arduino IDE serial monitor to the Bluetooth device (an Android phone).  The corruption most commonly seen is the loss of the LF and CR characters (plus a random number of other terminal ASCII characters), so the data seen on The Bluetooth Terminal app is on one line, rather than neat lines.
delay(5) seems to work ok, but delay(4) starts dropping characters again, interesting it dropped the 10th character from a 16 character (+LF&CR) string...

lbernstone, 2018-08-28:
@me-no-dev posted a new branch (https://github.com/espressif/arduino-esp32/tree/bt-serial-overhaul) today that specifically addresses the issues in BluetoothSerial.  Give it a try.

D-140480, 2018-08-29:
@lbernstone   and @me-no-dev , I replaced the libraries/BluetoothSerial/src/BluetoothSerial.cpp with the new version, after digging out where it was installed on my computer, re-ran the example code and it works fine with the Sparkfun ESR32 Thing.  No dropped serial data in the example code.  Thanks ever so much.

closed 2018-08-29 by D-140480