espressif/arduino-esp32 #1811: BluetoothSerial.h memory

aster94, 2018-08-28:
### Hardware:
Board:							ESP32 lolin lite
Core Installation/update date:			last from board manager
IDE name:					        vscode (arduino ide)
Flash Frequency:					80
Upload Speed:					460800

### Description:
hello

i get this memory usage from a simple sketch which uses the BluetoothSerial, is it normal?

> Using library BluetoothSerial at version 1.0 in folder: /home/vincenzo/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/BluetoothSerial 
> Sketch uses 884211 bytes (67%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 53084 bytes (16%) of dynamic memory, leaving 274596 bytes for local variables. Maximum is 327680 bytes.

while i am here i have also another question, when the BT is enabled and used there is a flag which i could use in my program? 
example: `#ifdef USE_BT`

### Sketch:
```
#include "BluetoothSerial.h"
 
BluetoothSerial SerialBT;
 
void setup() {
  SerialBT.begin("Saiken_BT");
}
 
void loop() {
 
  SerialBT.println("Hello World");
  delay(1000);
}
//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}

```

lbernstone, 2018-08-28:
This is a forum for issues with the codebase, not basic programming instruction. Bluetooth is a very heavy library.  The header files have all the defines for the library.

closed 2018-08-30 by aster94

prafullpriyam, 2018-11-11:
If someone help me out then thank you in advance........

It seems that "esp_bt_cod_t" structure has not been defined inside the BluetoothSerial.cpp file. Please help me out.

ConcepcionMota, 2018-11-26:
Hi, I have the error "fatal error: esp_spp_api.h :Not such file or directory"