espressif/arduino-esp32 #1812: Wifi doesn't like the MCP2515

dorarosi, 2018-08-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			28/jul/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
So I connected 4 MCP2515 to my ESP32 and everything was fine. Then I added OTA Updates and got some strange reading issues. The first few messages got read wrong. Sometimes there were always some wrong messages, not just at the beginning. I only send 0x30B with a DLC of 8 and zeros as content but get very diffrent results. The code is built to do a gateway to test if all messages come through. The messages get sent with a Peak System PCAN USB Pro with 500k and a cycle time of 10ms. The gateway without wifi can easily redirect 3 diffrent messages sent at 1ms cycle time.

[image]

### Sketch:
```
#include <SPI.h>
#include <esp32Mcp2515.h>
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

struct can_frame canMsg1, canMsg2, canMsg3, canMsg4;
ESP32MCP2515 can1(14,26,27,15, false);
ESP32MCP2515 can2(14,26,27,22, false);
ESP32MCP2515 can3(18,19,23, 5, true);
ESP32MCP2515 can4(18,19,23,33, true);

const char* ssid = "****";
const char* password = "****";

void isr0(){
  if (can1.readMessage(&canMsg1) == ESP32MCP2515::ERROR_OK) {
     /* 
    Serial.print("CAN1 ");
    Serial.print(canMsg1.can_id, HEX); // print ID
    Serial.print(" ");
    Serial.print(canMsg1.can_dlc, HEX); // print DLC
    Serial.print(" ");
    
    for (int i = 0; i<canMsg1.can_dlc; i++)  {  // print the data
        
      Serial.print(canMsg1.data[i],HEX);
      Serial.print(" ");
    }
      

    Serial.println();
    */
    can2.sendMessage(&canMsg1);      
  }
}

void isr1(){
  if (can2.readMessage(&canMsg2) == ESP32MCP2515::ERROR_OK) {
    
    Serial.print("CAN2 ");
    Serial.print(canMsg2.can_id, HEX); // print ID
    Serial.print(" "); 
    Serial.print(canMsg2.can_dlc, HEX); // print DLC
    Serial.print(" ");
    
    for (int i = 0; i<canMsg2.can_dlc; i++)  {  // print the data
        
      Serial.print(canMsg2.data[i],HEX);
      Serial.print(" ");

    }

    Serial.println();
    
    can4.sendMessage(&canMsg2);      
  }
}

void isr2(){
  if (can3.readMessage(&canMsg3) == ESP32MCP2515::ERROR_OK) {
    /*  
    Serial.print("CAN3 ");
    Serial.print(canMsg3.can_id, HEX); // print ID
    Serial.print(" "); 
    Serial.print(canMsg3.can_dlc, HEX); // print DLC
    Serial.print(" ");
    
    for (int i = 0; i<canMsg3.can_dlc; i++)  {  // print the data
        
      Serial.print(canMsg3.data[i],HEX);
      Serial.print(" ");

    }

    Serial.println(); 
    */
    can4.sendMessage(&canMsg3);    
  }
}

void isr3(){
  if (can4.readMessage(&canMsg4) == ESP32MCP2515::ERROR_OK) {
    /* 
    Serial.print("CAN4 ");
    Serial.print(canMsg4.can_id, HEX); // print ID
    Serial.print(" "); 
    Serial.print(canMsg4.can_dlc, HEX); // print DLC
    Serial.print(" ");
    
    for (int i = 0; i<canMsg4.can_dlc; i++)  {  // print the data
        
      Serial.print(canMsg4.data[i],HEX);
      Serial.print(" ");

    }

    Serial.println();  
      */
    can2.sendMessage(&canMsg4); 
  }
}

void setup() {
  Serial.begin(115200);
  SPI.begin();

  can1.reset();
  can1.setBitrate(CAN_500KBPS, MCP_8MHZ);
  can1.setMode(can1.CANCTRL_REQOP_NORMAL);
  
  can2.reset();
  can2.setBitrate(CAN_500KBPS, MCP_8MHZ);
  can2.setMode(can2.CANCTRL_REQOP_NORMAL);
  
  can3.reset();
  can3.setBitrate(CAN_500KBPS, MCP_8MHZ);
  can3.setMode(can3.CANCTRL_REQOP_NORMAL);
  
  can4.reset();
  can4.setBitrate(CAN_500KBPS, MCP_8MHZ);
  can4.setMode(can4.CANCTRL_REQOP_NORMAL);
    
  WiFi.mode(WIFI_STA);
  Serial.print("Connecting to ");
  Serial.print(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
  }
  Serial.println();

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  //delay(2000);
  
  pinMode(25, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(25), isr0, ONLOW);
  
  pinMode(21, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(21), isr1, ONLOW);
  
  pinMode(17, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(17), isr2, ONLOW);
  
  pinMode(32, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(32), isr3, ONLOW);
  
  Serial.println("------- CAN Read ----------");
  Serial.println("CAN  ID  DLC   DATA");
}

void loop() {
  ArduinoOTA.handle();
  //delay(10);
}
```
### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
18
Connecting to HuaweiMate10Pro[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 2 - STA_START
......[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 7 - STA_GOT_IP
.
Ready
IP address: 192.168.43.158
CAN2 0 0 
CAN2 30B 8 5 0 0 0 0 0 0 0 
------- CAN Read ----------
CAN  ID  DLC   DATA
CAN2 20A 8 0 0 6 0 0 0 0 0 
CAN2 20B 8 0 0 0 0 0 0 0 6 
CAN2 30B 8 0 0 0 0 0 0 0 0 
CAN2 30B 8 0 0 0 0 0 0 0 0 
CAN2 30B 8 0 0 0 0 0 0 0 0 
CAN2 30B 8 0 0 0 0 0 0 0 0  
```

lbernstone, 2018-08-29:
A couple suggestions:
1) Your ISRs should be marked as IRAM_ATTR.  Look at the GPIO example to see where that goes.
2) ISRs should be very quick to run, have as much of their resources localized as possible, and not be actively "doing".  Fill a buffer in the ISR, and then have a process in the main loop that prints them out and clears the buffers. Set flags for other activities that need to happen.

dorarosi, 2018-08-30:
So IRAM_ATTR doesn't change anything. And the ISRs are short and take less than 10 sec. As you can see in the console log the ESP can't read the messages right. I also found out that only CAN 1 and 2, which are on the HSPI, don't work right. CAN 3 and 4 on the VSPI are working fine.

closed 2018-09-21 by dorarosi