espressif/arduino-esp32 #1813: Sketch not executed after upload

Bassinator, 2018-08-29:
### Hardware:
Board:							        u-blox NINA W10 series (esp32)
Core Installation/update date:			29/aug/2018
IDE name:							Arduino IDE
Flash Frequency:					        40Mhz
Upload Speed:						115200

### Description:
after uploading a sketch from Arduino IDE it does not start

### Sketch:
```
///////////////////////////////////////////////////
//Project 11 Hello World!
int counter = 1;
void setup(){
  Serial.begin(115200);
}
void loop(){
  Serial.println("Hello World");
  Serial.print("Counter = ");
  Serial.println(counter);
  delay(1000);
  counter++;
}
```

### Serial Monitor output:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
user code done
```

### Workaround:
replaced the following files:
~/Documants/Arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin
~/Documents/Arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader_dio_40m.bin
with the content of this zip file
https://github.com/rac2030/IoT-conference-badge/blob/firmware/firmware/hackathon-kraftwerk/deploy/hackathon-kraftwerk-firmware-v.1.0.1.zip

menox1981, 2018-09-13:
.....And where did these files come from?

Bassinator, 2018-09-13:
I have an ESP32 from the make Zurich badge https://makezurich.ch. the files are from the make Zürich badge firmware. Sources: https://github.com/rac2030/IoT-conference-badge/tree/firmware/firmware/hackathon-kraftwerk

closed 2019-08-15 by stale[bot]