espressif/arduino-esp32 #1816: HardwareSerial bug

aster94, 2018-08-30:
## Hardware:
Board:	ESP32 lolin lite
Core Installation/update date:	last from board manager
IDE name:	vscode (arduino ide)
Flash Frequency:	80
Upload Speed:	460800

## Description:
hello,
I am 100% sure that there is something wrong in the HardwareSerial class. Today i lost about 2h trying to debug my code on an ESP32, then i decided to test the same code on an arduino mega, there it worked.

The problem is that i am not able to make a minimum working example which raises the error... i am really really sorry for this, i tried to make it but after 1h of attempts i give up....
if you want to see it you should use this library with the example sketch: https://github.com/aster94/Keyword-Protocol-2000

the code will hang exactly here:
https://github.com/aster94/Keyword-Protocol-2000/blob/master/src/KWP2000.cpp#L156

the error is raised whenever i call any function of the serial class (or at least available, read) that you will see is this:
```
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x400d25b3 on core 1

Backtrace: 0x400883f8:0x3ffb1e30 0x400885fb:0x3ffb1e50 0x400d25b3:0x3ffb1e70 0x400859f2:0x3ffb1ea0 0x4008515d:0x3ffb1ed0 0x400d17f5:0x3ffb1ef0 0x400d12b5:0x3ffb1f10 0x400d0efa:0x3ffb1f30 0x400d11f9:0x3ffb1f60 0x400d0a92:0x3ffb1f80 0x400e47fa:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
```

aster94, 2018-09-12:
this says s
```
Decoding stack results
0x400883f8: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x400885fb: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x400d3c1b: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400859f2: vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portmux_impl.inc.h line 105
0x4008515d: uxQueueMessagesWaiting at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1748
0x400d1561: uartAvailable at /home/vincenzo/.arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/esp32-hal-uart.c line 248
0x400d19ad: HardwareSerial::available() at /home/vincenzo/.arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/HardwareSerial.cpp line 65
0x400d0f1a: KWP2000::listenResponse(unsigned char*, unsigned char*, unsigned char) at /home/vincenzo/Cloud/Programmi/Arduino/libraries/Keyword-Protocol-2000/src/KWP2000.cpp line 141
0x400d1215: KWP2000::initKline() at /home/vincenzo/Cloud/Programmi/Arduino/libraries/Keyword-Protocol-2000/src/KWP2000.cpp line 428
0x400d0ab2: loop() at /tmp/arduino_modified_sketch_628531/sketch_sep12a.ino line 32
0x400e5e92: loopTask(void*) at /home/vincenzo/.arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/main.cpp line 17
```

so if i am not wrong the error seems to be caused from this line
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L248

i wasn't able to date back the errors before, since

menox1981, 2018-09-13:
Haha it looks like this could be a FreeRTOS problem. It's checking if the task is on the right core. Since this a a dual core processor it may have been moved to the other core by FreeRTOS. I would share this information with them.

closed 2019-08-15 by stale[bot]