espressif/arduino-esp32 #1817: undefined reference to `esp_derive_local_mac'

tablatronix, 2018-08-30:
esp32 doit

I can call most stuff in esp_system.h without a problem, but trying to use this and I get this.
It will give me syntax complaints and stuff, but when compiling it does this, linker issue in SDK ?

code ( probably wrong also)
```
#include <esp_system.h>
uint64_t _chipmacid = 0LL;
esp_efuse_mac_get_default((uint8_t*) (&_chipmacid));
uint64_t _localmacid = 0LL;
esp_derive_local_mac((uint8_t*)(&_localmacid),(uint8_t*)(&_chipmacid));
```

compile error
```
.pioenvs\esp32doit-devkit-v1\src\GetChipID.ino.cpp.o:(.literal._Z5setupv+0x40): undefined reference to `esp_derive_local_mac'
.pioenvs\esp32doit-devkit-v1\src\GetChipID.ino.cpp.o: In function `setup()':
GetChipID.ino.cpp:(.text._Z5setupv+0xf2): undefined reference to `esp_derive_local_mac'
```

shrug

lbernstone, 2018-08-30:
Post this issue upstream.  I see it in the header, but no actual function in ESP-IDF.

closed 2018-08-30 by tablatronix

beegee-tokyo, 2018-08-31:
Shouldn't that be
`esp_device_local_mac()` ?
DEVICE not DERIVE