espressif/arduino-esp32 #1818: Error compiling for board SparkFun ESP32 Thing (with NFC)

D-140480, 2018-08-31:
### Hardware:
Board:							       Sparkfun ESP32 Thing
Core Installation/update date:			Jun 8 2016 (of the chip)
IDE name:							Arduino IDE 1.8.6
Flash Frequency:					       80Mhz
Upload Speed:						921600

### Description:
Cannot compile sketch when PN532 library included.
This issue was touched upon by ThrowAway at:
https://www.esp32.com/viewtopic.php?t=1044
but I cannot see is this was solved or pursued and I am up against this issue as well.  I have an Adafruit PN532 RFID shield as I am trying to use NFC to write to NFC tags.  I have the code working fine on an Sparkfun Redboard, but can't even get started on the SparkFun ESP32 Thing as it will not compile.  The verbose output from the compile logs has 129 mentions of the word error :-( and it be fair, most of them are in the PN532 library such as:
C:\Users\[user]\Documents\Arduino\libraries\PN532\PN532.cpp:626:56: error: narrowing conversion of '(((int)len) + 5)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Werror=narrowing].

But as this sketch does upload to an Arduino Uno clone but not to the SparkFun ESP32 Thing, it makes me think the issue is with the ESP32 core code.

The very much cut down code is below.

### Sketch:
```cpp

//Change the code below by your sketch
#include <PN532_I2C.h>
#include <PN532.h>   // The following files are included in the libraries Installed
#include <NfcAdapter.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-08-31:
You will really need to have the library maintainers work on this.  FWIW, the [Adafruit library](https://github.com/adafruit/Adafruit-PN532) does compile, but they indicate it does not work on esp8266. It is worth giving it a try.

D-140480, 2018-09-02:
Thanks @lbernstone , I'll give them a go, I see there is already an aging pull request for "Added ESP32 as supported target".  They don't seem very active...

closed 2018-09-02 by D-140480

D-140480, 2018-09-02:
@lbernstone , you are lucky you can get any code compiled with the [Adafruit library](https://github.com/adafruit/Adafruit-PN532) as I can even get their examples to compile.

beegee-tokyo, 2018-09-02:
Having  a quick look into `Adafruit_PN532.cpp` shows
```
#include <Wire.h>
#if defined(__AVR__) || defined(__i386__) || defined(ARDUINO_ARCH_SAMD) || defined(ESP8266) || defined(ARDUINO_ARCH_STM32)
 #define WIRE Wire
#else // Arduino Due
 #define WIRE Wire1
#endif
```
If I am not wrong that leads on the ESP32 to
`#define WIRE Wire1`
which might be cause for the compilation errors.
Try to replace all this #ifdef/#else stuff with a simple `#define WIRE Wire` and see if it compiles.

D-140480, 2018-09-03:
Hi @beegee-tokyo , there is a [pull request](https://github.com/adafruit/Adafruit-PN532/pull/44) (Made on 1 April...) which adds the ESP32
#if defined(__AVR__) || defined(__i386__) || defined(ARDUINO_ARCH_SAMD) || defined(ESP8266) || defined(ARDUINO_ARCH_STM32) || defined(ESP32)
which seems to be a complicated (but perhaps more proper?) way of doing what you suggest.  I changed the .cpp file to align with the Adafruit pull request, and it still does not compile.  I then changed it directly with what you suggested, still no joy.

Some of the errors I get, when the Board Manager is set to Sparkfun ESP32 Thing, are"
In file included from C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.cpp:60:0:

C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.h: In constructor 'Adafruit_PN532::Adafruit_PN532(uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.h:199:29: error: 'Adafruit_PN532::_miso' will be initialized after [-Werror=reorder]

   uint8_t _ss, _clk, _mosi, _miso;

                             ^

C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.h:199:22: error:   'uint8_t Adafruit_PN532::_mosi' [-Werror=reorder]

   uint8_t _ss, _clk, _mosi, _miso;

But it is interesting that these **errors** get changed to mere **warnings** when the board Manager is changed to an Arduino UNO:
In file included from C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.cpp:60:0:

C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.h: In constructor 'Adafruit_PN532::Adafruit_PN532(uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.h:199:29: warning: 'Adafruit_PN532::_miso' will be initialized after [-Wreorder]

   uint8_t _ss, _clk, _mosi, _miso;

                             ^

C:\Users\[user]\Documents\Arduino\libraries\Adafruit-PN532\Adafruit_PN532.h:199:22: warning:   'uint8_t Adafruit_PN532::_mosi' [-Wreorder]

   uint8_t _ss, _clk, _mosi, _miso;

And if you got this far, thanks.  This is the .cpp file modified as you suggested.
#include <Wire.h>
//#if defined(__AVR__) || defined(__i386__) || defined(ARDUINO_ARCH_SAMD) || defined(ESP8266) || defined(ARDUINO_ARCH_STM32) || defined(ESP32)
 #define WIRE Wire
//#else // Arduino Due
// #define WIRE Wire1
//#endif

D-140480, 2018-09-21:
> @lbernstone

Well, what do you know.  I installed Arduino IDE on another computer and now, I too, can get the PN532 Adafruit library to compile.  Boy, that was a frustrating exercise.  Now I will see how I get on actually writing the code!