espressif/arduino-esp32 #1819: BLE status

sensboston, 2018-08-31:
It's a feature request or just a help request. **Arduino core for the ESP32** is working perfectly for me, right out of box. The only issue I'm experiencing, it's a lack of full functional BLE device with GATT profile example :( I looked at all open and close issues related to BLE; unfortunately there is no useful information at all, just a common words or non-working projects.

If anyone already implemented a simple BLE device with generic attribute, could you please share your code with me (if you can't share with community)? I definitely wanna get rid from SPP (yes, it works perfectly fine and stable but requires pairing, what I'm trying to avoid).

Any **real helpful** information is very welcome but please try to avoid to providing me a links to low-level BT functions or library for the different hardware etc. Thank you!

sensboston, 2018-08-31:
Thanks for reply, but I already found an answer on my question, 5 minutes after posting (what's why I closed it immediately)  :) I was messed up with the SimpleBLE sketch (what is completely useless) but found another great ESP32_BLE samples (ble_server etc) which 100% fulfill my needs and works out of the box.

karansoi, 2018-08-31:
@Sensboston, it will be great you shared these examples with us too @github for benefit to all.

sensboston, 2018-09-01:
@karansoi, these examples [already available](https://github.com/nkolban/ESP32_BLE_Arduino/tree/7951347ed68313d75c367e1f2cce763cb56d1eb2/examples) in this repository! Working (at least for me) right out of box, with two different ESP32 clones.