espressif/arduino-esp32 #1820: SPIFFS - FSBrowser

q-bird, 2018-08-31:
Hi there,

I want to upload a webserver folder to Flash using FSBrowser.ino example, everything is working fine.
Then I want to use a non-standard font with the extension `.woff2`.
I copied a font file `.woff2` to "data" folder and tried to upload using "ESP32 Sketch Data Upload" Tool...
But it showed an error:
```
/KFOkCnqEu92Fr1MmgVxEIzIFKw.woff2
SPIFFS_write error(-10010): unknown
error adding file!
SPIFFS Create Failed!
```

I was trying to add something in the code like:
```
String getContentType(String filename) {
  if (server.hasArg("download")) {
    return "application/octet-stream";
  } else if (filename.endsWith(".htm")) {
    return "text/html";
  } else if (filename.endsWith(".html")) {
    return "text/html";
  } else if (filename.endsWith(".woff2")) {
    return "font/woff2 woff2";
  } ...
```
I tried with  `return "font/woff2 woff2";`,  `return "application/font-woff2";`, `return "font/x-woff2";`...
But I still got the same error.

Is there any way to upload `.woff2` font-file to Flash?

Thanks in advance!

MarkyAD, 2018-08-31:
@q-bird this area is for reporting issues with the esp32 core code. You should close your issues here so the developers dont have to weed through posts that are not core issues. Use a forum like esp32.com for "how do I..." type questions. SPIFFS filenames are limited to 31 characters. There are no paths or special characters in SPIFFS filenames, so every character counts toward that limit. Your filename as shown above is 33 characters.

q-bird, 2018-08-31:
@MarkyAD 
Thanks for your advice :)
And I also resolved this problem by changing the file name!

closed 2018-08-31 by q-bird