espressif/arduino-esp32 #1822: ESP32 - Modbus over TCP

rfaloisio, 2018-08-31:
Hey Guys!

I've implemented my own modbus protocol inside my own code. I wanted to do it over TCP. Everything works out, except when the values change.
My value of intTemperatura changes all the time but on the other end the value received is not changing.

Here is my code:
`void setup(){

  //Definição da Serial e do Modbus
  /////////////////////////////////
  Serial.begin(115200);
//  pinMode(SSerialTxControl, OUTPUT);    
//  digitalWrite(SSerialTxControl, RS485Receive);  // Inicia comuicação com RS-485 aguardando comunicação    
//  RS485Serial.begin(19200);

  delay(1000);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED){
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
   Serial.println("Conectado");
   Serial.println(WiFi.localIP());
   

  //Criação das Tabelas Modbus
  /////////////////////////////
  MAddress3X[0] = 3001; //Código do Produto
  MAddress3X[1] = 1807; //Número de Série parte 1
  MAddress3X[2] = 1001; //Número de Séria parte 2
  MAddress3X[3] = 102; // Versão de Firmware
  
  //Entradas Digitais
  ///////////////////
  pinMode(pinED1, INPUT);  
  
  //Leitura de Temperatura e Umidade
  //////////////////////////////////
  dht.begin();
  wifiServer.begin();

  //Timers
  ////////
  timer_DHT11.every(2000, Read_DHT11); // configura função timer para ser chamada a cada 1 segundo
  timer_Console.every(2000, AtualizaConsole); // configura função timer para ser chamada a cada 1 segundo
  
}

void  loop(){
  client = wifiServer.available();
  timer_DHT11.update();  // chama atualização da leitura do sensor de temperatura e umidade
  timer_Console.update();  // chama atualização da leitura do sensor de temperatura e umidade
  valED1 = digitalRead(pinED1); // leitura da entrada digital 0
  byteED18 = ConvertValToByte(byteED18, valED1, 0); // coloca o valor do pino na posição do byte correspondente
  strStatus = ConvertByteToText(byteStatus); // converte byte para texto para mostrar no console de depuração
  strByteED18 = ConvertByteToText(byteED18); // converte byte para texto para mostrar no console de depuração
  MAddress3X[4] = byteStatus;
  MAddress3X[5] = byteED18;
  MAddress3X[6] = intTemperatura;
  MAddress3X[7] = intUmidade;

  //COMUNICAÇÃO WIFI
  //////////////////// 
  
    while(client.connected()){ 
  if (client.available() > 0){
    if (dataComplete == true){
    char data[255];
    dataComplete = false;
  }
    
    if (dataComplete == false){
      while (client.available()>0){
        char character = client.read();
        data[count] = character;
        count++;
      }
     client.flush();
     dataComplete = true;

     // VERIFICA CRC
     CRC_CHECK(data, count, 0, 5);
     if (bol_CRC == true){
      // CRC está OK verifica se a comunicação é para o device
      int_SlaveId = (int)data[0];
      if (int_SlaveId == ModbusSlaveId){
        // CRC OK e solicitação é para o device
        TRATA_COMUNICACAO(data);
      }
     }
     count=0;
    } // dataComplete = false
    
  } // RS485Serial.available<> > 0
  
 }
      
}
`

And here is the function that sends the modbus frame:

` int_ModbusCommand = (int)RxData[1];
   if (int_ModbusCommand == 4)
    {
    int_ModbusAddressHigh = (int)RxData[2] * 256;
    int_ModbusAddressLow = (int)RxData[3];
    int_ModbusAddress = int_ModbusAddressHigh + int_ModbusAddressLow;
    int_ModbusDataLengthHigh = (int)RxData[4] * 256;
    int_ModbusDataLengthLow = (int)RxData[5];
    int_ModbusDataLength = int_ModbusDataLengthHigh + int_ModbusDataLengthLow;

ByteTX[0] = ModbusSlaveId;
        ByteTX[1] = 4;
        ByteTX[2] = int_ModbusDataLength * 2;
        int_FinalAddress = int_ModbusAddress + int_ModbusDataLength;
        int_Cont = 0;
        for (i = int_ModbusAddress; i < int_FinalAddress; i++)
          {
          high = highByte(MAddress3X[i]);
          low = lowByte(MAddress3X[i]);
          ByteTX[3 + int_Cont] = high;
          ByteTX[4 + int_Cont] = low;
          int_Cont = int_Cont + 2;
          }
        for (i = 0; i < (3 + int_Cont); i++)
          {
          CharTX[i] = char(ByteTX[i]);
          }
        CRC_CHECK(CharTX, (3 + int_Cont), 0, (2 + int_Cont));
        ByteTX[3 + int_Cont] = int_LSB;
        ByteTX[4 + int_Cont] = int_MSB;
//        RS485Serial.write(ByteTX,(5 + int_Cont));
        client.write(ByteTX, (5 + int_Cont));   
        client.stop();   
//        delay(100);`

Hope someone can help me.

Thanks

rfaloisio, 2018-08-31:
Sorry about the format of the code, I'm having some trouble with github on my machine.

malbrook, 2018-09-11:
There are too many functions whch are not provided to be able to decode exactly where you are going wrong, for example where does  intTemperatura; get updated, what does highByte() do, what does lowByte() do. I can see from their names what they are supposed to be doing but without the code it is not possible to see if they are working correctly. another possible area is your calculation for the modbus crc16 (function CRC_CHECK()) which seems quite complex in its call, usually you would only need the buffer pointer and the length of the buffer to calculate the crc. Also ModbusTCP which is a standard Modbus protocol exists and that does not use the CRC unlike Modbus RTU which you appear to be copying into TCP transmissions.

closed 2019-08-15 by stale[bot]