espressif/arduino-esp32 #1824: serial cannot receive the last byte

piaoyezju, 2018-09-01:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			latest
IDE name:							Arduino IDE

### Description:
A few minutes after booting, the serial port will not receive the last byte unless there is new data behind it, and then the new data is still missing the last byte.
The code is as follows, connect P16, P17
please help me ,thanks!

### Sketch:
```cpp
//#include "SPIFFS.h"
#include <WiFi.h>
HardwareSerial USE_SERIAL1(1);
HardwareSerial USE_SERIAL2(2);
#define DEBUG_SERIAL Serial
#define UART1_RX          19
#define UART1_TX          18
#define UART2_RX          16
#define UART2_TX          17
#define UART0_RX          23
#define UART0_TX          22

byte state = 0x00;
unsigned long expireTime = 0;

void startQuery() {
  if (state == 0x00) {   // new query
    if(USE_SERIAL2.available() > 0) {
      DEBUG_SERIAL.println("new query, clear data:" + String(USE_SERIAL2.available()));
      while (USE_SERIAL2.available() > 0) {
        USE_SERIAL2.read();
      }
    }
    expireTime = millis() + 300;
    state = 0x01;
    byte cmd[9] = {0x2b, 0x00, 0x01, 0x00, 0x01, 0xa1, 0x00, 0x00, 0x7e};
    USE_SERIAL2.write(cmd, 9);
  } else if (state == 0x01) {
    if(USE_SERIAL2.available() >= 9) {
      while (USE_SERIAL2.available() >= 9) {
        byte head = USE_SERIAL2.read();
        if (head == 0x2b) {
          byte buf[8] = {0};
          USE_SERIAL2.readBytes(buf, 8);
          state = 0x00;
          DEBUG_SERIAL.print(".");
        }else{
          DEBUG_SERIAL.print("error first byte:");
          DEBUG_SERIAL.print(head, HEX );
          DEBUG_SERIAL.println("");
        }
      }
    }else if (expireTime < millis()) { // timeout
      DEBUG_SERIAL.println("timeout, clear data:" + String(USE_SERIAL2.available()));
      while (USE_SERIAL2.available() > 0) {
        USE_SERIAL2.read();
      }
      state = 0x00;
    }
  }
}

void setup() {
//  SPIFFS.begin(true);
  WiFi.begin();
  USE_SERIAL1.begin(38400, SERIAL_8N1, UART1_RX, UART1_TX);
  USE_SERIAL2.begin(38400, SERIAL_8N1, UART2_RX, UART2_TX);
  DEBUG_SERIAL.begin(115200);
}

void loop() {
    startQuery();
}
```

### Debug Messages:
```
.....................
error first byte:7E
timeout, clear data: 8 
error first byte:7E
timeout, clear data: 8 
error first byte:7E
timeout, clear data: 8 
```

piaoyezju, 2018-09-06:
i check out the version to commit 7141e3e29dc0c5758313bf4e0ac93fcca15dcecf then it works fine

piaoyezju, 2018-09-06:
Also,If I comment out `WiFi.begin();` this error will not appear.

MarkyAD, 2018-09-07:
I had an older core version on another machine and compiled on that, and sure enough its been running for hours now with no problems. Thanks.

lbernstone, 2018-09-07:
Possible fix: https://github.com/espressif/arduino-esp32/pull/1849

MarkyAD, 2018-09-08:
@lbernstone yes indeed that patch fixes it.

closed 2018-09-09 by piaoyezju