espressif/arduino-esp32 #1832: WiFi SoftAP not working?

beegee-tokyo, 2018-09-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			04.September.2018
IDE name:						Arduino IDE **_1.8.5_**
Flash Frequency:					80Mhz
Upload Speed:						115200

**ESP32 core V1.0.0** installed over boards manager from 
**`https://dl.espressif.com/dl/package_esp32_index.json`**

### Description:
Simple examples for WiFi SoftAP seem to work (no error indicated), but the SoftAP is not showing (Tried several Android devices, PC Wifi). 
The SSID is not hidden, even if I try to add the AP manually, none of the devices can see or connect to the SoftAP.

I tried on 2 different ESP32 modules, but get the same result.
Both modules WiFi are working, I can connect to any of my WiFi AP's here with **`WiFi.begin("sta","pw");`*. Just the SoftAP seems not to work at all.

Tried as well with 
**`WiFi.mode(WIFI_AP_STA);`** or **`WiFi.mode(WIFI_AP);`** 
before calling WiFi.softAP, but no change.

Tried as well with  	
```
IPAddress softApIp = {10, 0, 1, 1};
IPAddress softApGw = {10, 0, 1, 1};
IPAddress softApMa = {255, 255, 255, 0};
WiFi.softAPConfig(softApIp, softApGw, softApMa);
```
before calling WiFi.softAP, but no change.

Tried as well **`WiFi.softAP(ssid, password, 5, 0, 4);`** but no change.

Tried as well with the developer version of the ESP32 core (**`https://dl.espressif.com/dl/package_esp32_dev_index.json`** V1.0.0-rc4), but no change.

### Sketch:
```cpp
#include <WiFi.h>
 
const char *ssid = "MyESP32AP";
const char *password = "testpassword";
 
void setup(){
  Serial.begin(115200);
// WiFi.mode(WIFI_AP_STA);
  WiFi.softAP(ssid, password);
//  WiFi.softAP(ssid, password, 5, 0, 4);
 
  Serial.println();
  Serial.print("IP address: ");
  Serial.println(WiFi.softAPIP());
  Serial.print("[WIF] SoftAP MAC: ");
  Serial.println(WiFi.softAPmacAddress());
}
 
void loop(){}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 13 - AP_START

IP address: 192.168.4.1
[WIF] SoftAP MAC: 24:0A:C4:81:CE:9D
```

MarkyAD, 2018-09-04:
AP works here, but trying WiFi.softAP(ssid, password, 5, 0, 4); it always shows up as channel 1, no encryption, ssid = ESP_###### (last 3 bytes of mac address). Tried using WiFi.disconnect(true); and also did an erase_flash, same results. Before the erase_flash it was always stuck on the channel it was previously associated with as a station.

beegee-tokyo, 2018-09-04:
@MarkyAD @Sauro98 

Thanks for the quick reply.
Erasing the whole flash and then reprogramming did the trick. But as you said, whatever is used as SSID, it ends up with ESP_######.

closed 2018-09-04 by beegee-tokyo

Sauro98, 2018-09-04:
Weird, I set my AP ssid to be the whole mac address and that is actually what I get

beegee-tokyo, 2018-09-04:
I try this:
```
// Start local AP portal
uint8_t resultAP = 0;
if (WiFi.softAPConfig(
		IPAddress(10, 0, 1, 1),
		IPAddress(10, 0, 1, 1),
		IPAddress(255, 255, 255, 0)))
{
	char softApName[] = "RS-xx:xx:xx:xx:xx:xx";
	sprintf(softApName, "RS-%s", WiFi.macAddress().c_str());
#ifdef WIF_DBG
	SerialCMD.print("[WIFI] Starting SoftAP: ");
	SerialCMD.println(softApName);
#endif
	WiFi.softAP(softApName, "evlaser");
}
```
And get ESP_81CE9D as AP name.

MarkyAD, 2018-09-04:
@beegee-tokyo I still think there is some issue with WiFi.softAP(), no matter what channel I choose its always on ch.1 and it never lets me set my own ssid. Perhaps something introduced in the last couple months of commits. I'm using core from only a couple commits back from current git.

beegee-tokyo, 2018-09-04:
@MarkyAD 
For the channel, I don't care, because I use AP and STA mode in parallel. So the AP channel MUST be the same as the STA.
But for the AP name it is annoying.

Sauro98, 2018-09-04:
It works for me, but I was at commit `e157ec06a78d6abeaed1aaa9745cea6d7e155490`. I pulled the latest changes and it is still working.

beegee-tokyo, 2018-09-04:
Hahaha, found the same just now:

`strlen(passphrase) < 8` in WiFiAP.cpp

MarkyAD, 2018-09-04:
Maybe it shouldn't just fail and return there like that. I can see WPA2 depending on a decent passphrase, but maybe falling back to no encryption like it does a couple lines down from there would be better. At any rate, we should've been checking the return value of softAP() to see it was returning false. ;-P

Muny, 2021-12-11:
I think there may still be an issue with WiFi.softAP.  I just encountered an issue where WiFi.softAP() would never advertise itself and clients could not connect, even directly through BSSID.  I erased the flash of the esp32 and it started working again.  I'm not really sure how to replicate the issue though, unfortunately.

AC-Mike, 2023-06-14:
> This is the code I am using:
> 
> ```c++
> const IPAddress apIP = IPAddress(192, 168, 4, 1);
> static const byte DNS_PORT = 53;
> 
> DNSServer dnsServer;
> 
> WiFi.mode(WIFI_AP);
> WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
> WiFi.softAP(mac_name, "password");
> 
> dnsServer.start(DNS_PORT, "*", apIP);
> ```

This worked for me.  The order of the calls seems to be important.