espressif/arduino-esp32 #1838: HardwareSerial multiple definition

atsteich, 2018-09-05:
I'm not really familar with the esp32, and I'm getting a real strange error...
I installed the esp32 in an local arduino ide, no other packages installed. 

when running the following code:

`HardwareSerial Serial1(1);

#define RXD2 16
#define TXD2 17

void setup() {
  Serial.begin(115200);
  Serial1.begin(9600, SERIAL_8N1, RXD2, TXD2);
  Serial.println("Serial Txd is on pin: "+String(TX));
  Serial.println("Serial Rxd is on pin: "+String(RX));
}

void loop() {
  while (Serial2.available()) {
    Serial.print(char(Serial2.read()));
  }
}`

I get an error message saying that there is a multiple definition of Serial1. I have absolutely no idea what I am doing wrong, since there are lot of examples using exactly this code...

When I remove the line " HardwareSerial Serial2(2) " the script can be compiled and uploaded, but still it is not working when I run some traffic on the specified pins.

lbernstone, 2018-09-05:
The HardwareSerials should all be pre-allocated, so the objects are already there.  The default pins for Serial1 are 9&10, for Serial2 they are 16&17.  These pins are used for other hardware in some configurations (e.g. wrover modules).  The serial ports can be assigned to any [open output pins](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap), so maybe try using 21&22 or others that aren't in use on your board.  Make sure your code is consistent, as what you have posted here initializes Serial1 in one place, and then uses Serial2 later.

atsteich, 2018-09-05:
Thanks for responding!

You are right, there shouldn't be any Serial2.xxx, but only Serial1.xxx. 
Correcting this doesn't help, I still get the same error message. Changing the RX and TX pin is also not working...

Is nobody other having this problem?

MarkyAD, 2018-09-06:
@atsteich @lbernstone kindly made clear that the names Serial/Serial1/etc are already defined, so to define them again is an error. You can refer to the uart as something else, however:
```c++
HardwareSerial GPS(1); // uart 1
void setup() {
  GPS.begin(9600, SERIAL_8N1, RXD2, TXD2);
}
```

MarkyAD, 2018-09-06:
As @lbernstone said, you can use Serial1 without having to declare anything other than the Serial1.begin() line. If you still dont get any serial data, you've probably got the TX/RX lines swapped around. TX on the device should go to RX on the esp32 and RX on the device to TX on the esp32.

larryn46, 2019-07-05:
I had a similar problem. My code was almost the same as yours. Just comment out the line 
HardwareSerial Serial1(1);
to 
// HardwareSerial Serial1(1);
Serial1 is already defined.

closed 2019-09-17 by stale[bot]