espressif/arduino-esp32 #1840: Arduino as an esp-idf Component ... build issue

RA5040, 2018-09-05:
### Hardware:
Board:							ESP32 Wrover Module
Core Installation/update date:			4/9/2018
IDE name:						ESP32 with Arduino component
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I am trying to run Arduino code as a component in the esp-idf initially, and then in Eclipse.

I've followed the instructions here carefully :
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

I got some compilation errors so I referenced this link:
https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-418740315

and followed the instructions there carefully (both the 23rd Feb and 14th May posts) and I THINK I've done things correctly.

However I get a build error, as follows:
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp: In member function 'bool fs::SDMMCFS::begin(const char*, bool)':
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:56:26: error: 'sdmmc_host_io_int_enable' was not declared in this scope
.io_int_enable = sdmmc_host_io_int_enable,
^
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:57:24: error: 'sdmmc_host_io_int_wait' was not declared in this scope
.io_int_wait = sdmmc_host_io_int_wait,
^
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:59:5: error: 'sdmmc_host_t' has no non-static data member named 'io_int_enable'
};
^
/home/robert/esp/esp-idf/make/component_wrapper.mk:273: recipe for target 'libraries/SD_MMC/src//SD_MMC.o' failed
make[1]: *** [libraries/SD_MMC/src//SD_MMC.o] Error 1
/home/robert/esp/esp-idf/make/project.mk:453: recipe for target 'component-arduino-build' failed
make: *** [component-arduino-build] Error 2

I have deselected "Autostart Arduino setup and loop on boot" (first link above) so that the main.cpp should be as below.

I'm running on Ubuntu.

### Sketch:

extern "C" void app_main()
#include <stdio.h>
#include "sdkconfig.h"
#include "Arduino.h"
extern "C" void app_main()
{
initArduino();
}

lbernstone, 2018-09-05:
At the moment, there is no need to go through the "matching" routine- ESP-IDF and arduino-esp32 still seem to be in sync.  So, make sure you have the latest versions of each development repo.
extern "C" is a marker for the compiler to indicate obligate C code.  It is not needed if the C code will compile properly in C++, so use it sparingly.  In fact, you can compile the basic sketch you have there with no extern calls at all.

RA5040, 2018-09-05:
Thank you ... however I have a different problem right now. I've had to reinstall Ubuntu because of a disk crash and for some reason that I can't figure out Make is failing on not finding the xtensa toolchain. As fas as I can see everything seems right:

[image]

robert@robert-ubuntu:~/esp/blink$ make
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/main/bootloader_start.o
/home/robert/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c: In function 'call_start_cpu0':
/home/robert/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:45:5: error: missing braces around initializer [-Werror=missing-braces]
     bootloader_state_t bs = { 0 };
     ^
/home/robert/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:45:5: error: (near initialization for 'bs.ota_info') [-Werror=missing-braces]
/home/robert/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:45:5: warning: missing initializer for field 'factory' of 'bootloader_state_t' [-Wmissing-field-initializers]
In file included from /home/robert/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:21:0:
/home/robert/esp/esp-idf/components/bootloader_support/include_bootloader/bootloader_config.h:35:25: note: 'factory' declared here
     esp_partition_pos_t factory;
                         ^
cc1: some warnings being treated as errors
/home/robert/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'bootloader_start.o' failed
make[2]: *** [bootloader_start.o] Error 1
/home/robert/esp/esp-idf/make/project.mk:481: recipe for target 'component-main-build' failed
make[1]: *** [component-main-build] Error 2
/home/robert/esp/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/robert/esp/blink/build/bootloader/bootloader.bin' failed
make: *** [/home/robert/esp/blink/build/bootloader/bootloader.bin] Error 2

If I try to run xtensa-esp32-elf-gcc for example it does find it:
robert@robert-ubuntu:~/esp/blink$ xtensa-esp32-elf-gcc
xtensa-esp32-elf-gcc: fatal error: no input files
compilation terminated.

And the IDF_PATH and PATH seems to be set up correctly:
robert@robert-ubuntu:~$ printenv IDF_PATH
/home/robert/esp/esp-idf

robert@robert-ubuntu:~$ printenv PATH
/home/robert/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/robert/esp/xtensa-esp32-elf/bin

And the bin folder seems to be in the right place with the right content:

robert@robert-ubuntu:~$ cd /home/robert/esp/xtensa-esp32-elf/bin

robert@robert-ubuntu:~/esp/xtensa-esp32-elf/bin$ ls

xtensa-esp32-elf-addr2line  xtensa-esp32-elf-cc            xtensa-esp32-elf-elfedit    xtensa-esp32-elf-gcc-ar      xtensa-esp32-elf-gdb     xtensa-esp32-elf-nm       xtensa-esp32-elf-readelf
xtensa-esp32-elf-ar         xtensa-esp32-elf-c++filt       xtensa-esp32-elf-g++        xtensa-esp32-elf-gcc-nm      xtensa-esp32-elf-gprof   xtensa-esp32-elf-objcopy  xtensa-esp32-elf-size
xtensa-esp32-elf-as         xtensa-esp32-elf-cpp           xtensa-esp32-elf-gcc        xtensa-esp32-elf-gcc-ranlib  xtensa-esp32-elf-ld      xtensa-esp32-elf-objdump  xtensa-esp32-elf-strings
xtensa-esp32-elf-c++        xtensa-esp32-elf-ct-ng.config  xtensa-esp32-elf-gcc-4.8.5  xtensa-esp32-elf-gcov        xtensa-esp32-elf-ld.bfd  xtensa-esp32-elf-ranlib   xtensa-esp32-elf-strip

Can you see anything wrong?  I'm kind of stuck. I have re-installed the toolchain and even moved it further up in case the path length was too long, but I get the same error.

MarkyAD, 2018-09-06:
@RA5040 from your project directory (~/esp/blink) try:

rm -rf build
make menuconfig
make

In menuconfig under "SDK tool configuration->Compiler toolchain path/prefix" should be set to
"xtensa-esp32-elf-" (without the quotes). menuconfig needs ncurses-dev libraries installed.

RA5040, 2018-09-06:
Copied blink example and built and flashed it: OK

Followed instructions in https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md ("Autostart Arduino setup and loop on boot" off)

1. Make on blink.c before any changes
WARNING: Toolchain version is not supported: 1.22.0-73-ge28a011
Expected to see version: 1.22.0-80-g6c4433a
...
Python requirements from /home/robert/esp/esp-idf/requirements.txt are satisfied.
/home/robert/esp/esp-idf/components/partition_table/Makefile.projbuild:60: warning: undefined variable 'quote'
Building partitions from /home/robert/esp/blink/components/arduino/tools/partitions/default.csv...
Partitions defined in '/home/robert/esp/blink/components/arduino/tools/partitions/default.csv' occupy 4.0MB of flash (4194304 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.
/home/robert/esp/esp-idf/components/partition_table/Makefile.projbuild:60: recipe for target '/home/robert/esp/blink/build/default.bin' failed
make: *** [/home/robert/esp/blink/build/default.bin] Error 2

2. Make on blink.cpp before any code changes
Same error

3. Modified code to have format

//#include "Arduino.h"
extern "C" void app_main()
{
    initArduino();
    ...
}

4. Make on modified code:
Same error

5. rm -rf build etc as per @MarkyAD 
same error

6. @MarkyAD : Installed ncurses-dev
sudo apt-get update
sudo apt-get install ncurses-dev

got
Note, selecting 'libncurses5-dev' instead of 'ncurses-dev'
libncurses5-dev is already the newest version (6.1-1ubuntu1.18.04).
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.

7. Ran menuconfig and make again
same errro

MarkyAD, 2018-09-06:
Youve selected a 2M flash size, in menuconfig under "Serial flasher config" set "Flash Size 4M".

MarkyAD, 2018-09-06:
>WARNING: Toolchain version is not supported: 1.22.0-73-ge28a011
>Expected to see version: 1.22.0-80-g6c4433a

Also its finding an old version of the toolchain, you should use the version it wants.

RA5040, 2018-09-06:
Thanks for your help!  I went right back to the Espressif installation instructions, which is why I have an older version of the toolchain ... I guess the documentation is not up to date.
At any rate, I've made your corrections and I now get the following errors:

CXX build/arduino/libraries/SD_MMC/src//SD_MMC.o
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp: In member function 'bool fs::SDMMCFS::begin(const char*, bool)':
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp:59:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp:59:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp:59:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp:59:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp:59:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp:59:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/robert/esp/blink/components/arduino/libraries/SD_MMC/src/SD_MMC.cpp:59:5: warning: missing initializer for member 'sdmmc_host_t::command_timeout_ms' [-Wmissing-field-initializers]
/home/robert/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'libraries/SD_MMC/src//SD_MMC.o' failed
make[1]: *** [libraries/SD_MMC/src//SD_MMC.o] Error 1
/home/robert/esp/esp-idf/make/project.mk:481: recipe for target 'component-arduino-build' failed
make: *** [component-arduino-build] Error 2

MarkyAD, 2018-09-06:
In menuconfig under "Arduino Configuration" tick "Include only specific arduino libraries" and untick "Enable SD_MMC". I had the same problem here.

MarkyAD, 2018-09-06:
I dunno eclipse I just use makefiles and vi, maybe @lbernstone can help with that. Be sure to thank him as well as he also took the time to help you.

RA5040, 2018-09-06:
Absolutely ... a lot of thanks to @ibernstone & @MarkyAD.
I'll do a write-up on this as it may be of use to others.

RA5040, 2018-09-06:
Here are instructions to build Arduino code in esp-idf:

MAKING ARDUINO COMPONENT

1. Install latest toolchain and esp-idf into ~/esp

2. Make blink project from esp-idf: 
cd ~/esp && \
rm -r -f blink  && \
cp -r esp-idf/examples/get-started/blink blink && \
cd blink

3. Create a components folder in the project folder and clone this repository inside it:
mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. 

4. make menuconfig
 - Go into Arduino Configuration:
   - Autostart Arduino setup and loop on boot: off
   - Disable mutex locks for HAL: off
   - Include only specific Arduino libraries: on
     -  Enable SD_MMC: off (if this is causing a problem in Make)
  - SDK tool configuration: make sure Compiler toolchain path/prefix is 'xtensa-esp32-elf-'

5. Rename blink.c to blink.cpp

6. Edit blink.cpp to be like:
#include "Arduino.h"

extern "C" void app_main()
{
    initArduino();
    pinMode(4, OUTPUT);
    digitalWrite(4, HIGH);
    //do your own thing
}

7. Make & flash
If make fails, clear the build folders and try again:
rm -rf build

RA5040, 2018-09-06:
Here are instructions to move the application to Eclipse:

BUILDING AN ARDUINO APPLICATION IN ECLIPSE 
LINUX

A. MAKE C++ TEMPLATE APPLICATION

    1. Install latest toolchain and esp-idf into ~/esp
       
    2. Make blink project from esp-idf:
cd ~/esp &&
rm -r -f blink &&
cp -r esp-idf/examples/get-started/blink blink &&
cd blink
       
    3. Create a components folder in the project folder and clone this repository inside it:
mkdir -p components &&
cd components &&
git clone https://github.com/espressif/arduino-esp32.git arduino && 
cd arduino &&
git submodule update --init --recursive &&
cd ../..
       
    4. make menuconfig
    • Go into Arduino Configuration:
        ◦ Autostart Arduino setup and loop on boot: off
        ◦ Disable mutex locks for HAL: off
        ◦ Include only specific Arduino libraries: on
            ▪ Enable SD_MMC: off (if this is causing a problem in Make)
    • SDK tool configuration: make sure Compiler toolchain path/prefix is 'xtensa-esp32-elf-'
      
    5. Rename blink.c to blink.cpp
       
    6. Edit blink.cpp to be like
       
#include "Arduino.h"
extern "C" void app_main()
{
initArduino();
pinMode(4, OUTPUT);
digitalWrite(4, HIGH);
//do your own thing
}

    7. Make & flash
If make fails, clear the build folders and try again:
rm -rf build

B. EDIT CODE TO MAKE IT INTO AN ARDUINO/C++ APPLICATION

1. Copy the blink application to your Eclipse workspace

2. In Eclipse do:
    • File/Import/CC++/Existing Code as a Makefile Project
    • Next
    • Browse to the blink folder in your workspace (the Project Name will automatically be put in).
    • Under Toolchain for Indexer Settings select Cross GCC
    • Click Finish
      
3. Rename blink.cpp file to your own application name and replace the contents with your own Arduino/C++ sketch. 

4. Add "#include "Arduino.h" at the top and make sure you have  extern ‘C’ void app_main and initArduino() as in A. Above. Also make calls to setup() and loop() if you have these in your sketch.

C. SET UP BUILD ENVIRONMENT TO BUILD A C++ APP

1. Right-click the project in the Project Explorer tab
    • Choose Properties
    • Choose C/C++ Build
    • Choose Environment
    • Choose Add
        ◦ Name: v
        ◦ Value: 1
        ◦ Select Add to all configurations
        ◦ OK
    • Choose Add again
        ◦ Name: IDF_PATH
        ◦ Value: enter the full path where you have installed the EPS-IDF (see 1. above). For instance: C:/msys32/home/Robert/esp/esp-idf (make sure to use '/', not '\') in Windows or ~/esp/esp-idf in Linux (note: the path may already be set)
        ◦ Select Add to all configurations
        ◦ OK
    • Select the PATH variable and click on Edit
        ◦ Replace the Value with C:\msys32\usr\bin;C:\msys32\mingw32\bin;C:\msys32\opt\xtensa-esp32-elf\bin in Windows or /bin:/home/robert/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/home/robert/esp/xtensa-esp32-elf/bin (or the corresponding path if you have installed msys32 somewhere else ... obviously with your own username inserted instead of ‘robert’).
        ◦ Apply
    • Select C/C++ General
        ◦ Preprocessor Include Paths
        ◦ Providers tab
        ◦ Click on CDT Cross GCC Built-in Compiler Settings
        ◦ Replace the Command to get compiler specs with
            ▪ xtensa-esp32-elf-gcc ${FLAGS} -E -P -v -dD "${INPUTS}
        ◦ Apply
        ◦ Click on CDT GCC Build Output Parser
        ◦ Replace the Compiler command pattern with
            ▪ xtensa-esp32-elf-(g?cc)|([gc]\+\+)|(clang)
            ▪ Apply
        ◦ Apply and Close

D. BUILD TARGET

1. Right-click on project
    • Index/Rebuild
    • Clean Project
      
2. Build Targets/Create
    • Target name: all (this will be used to build the app)
      
3. Build Targets/Create
    • Target name: flash (this will be used to build and flash the app)
      
4. Connect to EPS32

5. Check COM/tty port (for example using Device Manager/Ports in Windows, or ‘dmesg | grep tty’ in Linux)

6. Double-check that COM/tty port has been correctly set:
    • Open sdkconfig and check/correct the entry in:
    • CONFIG_ESPTOOLPY_PORT="COMx" / “/dev/ttyUSBx”

7. Click on the project
    • Build Targets
    • Flash (double-click)
      
8. Open a Terminal (you may need to go to Window/Show View/Terminal)
    • Click on Terminal icon
    • Choose terminal: Serial Terminal
    • Set up values
    • OK
	If your program terminated at flash, restart the ESP32 to see the output (if any).

NOTE: if you program shows errors even though it built without any errors, rebuild the index (right-click on project , Index, Rebuild). If that doesn't fix the problem try cleaning and rebuilding the project. If that doesn't work, go to Project/ Properties/ C/C++ General/ Code Analysis and unclick/apply and click back on/apply Syntax and Semantic Errors: you can also turn off/apply individual entries until you find the one that is causing a problem. In some cases it's simply a question of overriding the 'error' in the project editor).

lbernstone, 2018-09-06:
I was not building a new enough version of IDF, I guess.
https://github.com/espressif/arduino-esp32/pull/1836

RA5040, 2018-09-06:
I have another question, which I also posted here https://github.com/espressif/arduino-esp32/issues/1449 (but I think that thread is probably dead ... also I think it's relevant to this thread).

How do I add a library to the project?  https://github.com/espressif/arduino-esp32/issues/1449 says to just put the library in the project/components/arduino/libraries, but that doesn't work for me.  

The library is not found if I select 'Include only specific Arduino libraries' (which I have to do so that I can deselect 'Enable SD_MMC' ). If I do not select 'Include only specific Arduino libraries' then the U8g2 is built, but the build fails on the sd_mmc.

So it's a catch-22 situation until the sd_mmc issue is resolved ... or is there another way of adding the U8g2 library so that it becomes part of the Arduino libraries?

RA5040, 2018-09-06:
Well, I've found a workaround ... which is to comment out the SD_MMC.cpp file.

But is there a better way, by adding the library to the Arduino libraries?  Or having the library as a separate component (or something ...)?

neilpanchal, 2018-09-10:
I just followed the instructions and I am having a hard time building the component. I am using latest ESP-IDF, tool chain and arduinoesp32. 
```
/home/neil/esp/projects/hello_world/components/arduino/cores/esp32/esp32-hal-time.c:16:28: fatal error: apps/sntp/sntp.h: No such file or directory
compilation terminated.
```

lbernstone, 2018-09-10:
@neilpanchal: Please do not hijack issues.  This has nothing to do with the OP, and explains exactly what the problem is.

neilpanchal, 2018-09-10:
@lbernstone I was going to start another one, but thought the title of this issue was sufficiently vague to just post it here. No worries, I will open a new one. I did it out of courtesy to not inundate issue list with redundancy.

menox1981, 2018-09-13:
U8g2 relies on the Arduino SD library. I'll need to look at the source but I think this can be disabled with a #define

qt1, 2018-09-17:
Same problem, started after update.

got a few 
    sorry, unimplemented: non-trivial designated initializers not supported

Code looks like C initialization, which should work but doesn't.

closed 2019-08-15 by stale[bot]

Suraj-singh999898, 2020-02-24:
can anybody explain to me why the latest version on esp-idf (v4.0) does not support the Arduino-esp32 component?

atanisoft, 2020-02-24:
@Suraj-singh999898 it does if you use the correct branch of arduino-esp32, look at the idf v4 branch for support.

Suraj-singh999898, 2020-02-24:
hi atanisoft,

can you please elaborate on this method, I am trying this for a week.
its will be very help full.

sorry if it's a silly question, I just a beginner

atanisoft, 2020-02-24:
@Suraj-singh999898 make sure you are using https://github.com/espressif/arduino-esp32/tree/idf-release/v4.0 and it should work.

Suraj-singh999898, 2020-02-24:
Ok I try it now just wait an hour, I respond you soon,
And thanks.

Suraj-singh999898, 2020-02-28:
@atanisoft Sir,
worked.
Thanks

QGB, 2022-04-24:
```
[87/88] Generating binary image from built executable
esptool.py v3.2-dev
Merged 1 ELF section
Generated C:/Users/qgb/.platformio/platforms/espressif32/examples/espidf-aws-iot/build/bootloader/bootloader.bin
[88/88] cmd.exe /C "cd /D C:\Users\qgb\.platformio\platforms\espressif32\examples\espidf-aws-iot\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/qgb/.platformio/platforms/espressif32/examples/espidf-aws-iot/build/bootloader/bootloader.bin"
Bootloader binary size 0x4d80 bytes. 0x3280 bytes (39%) free.
[995/997] Linking CXX executable subscribe_publish.elf
FAILED: subscribe_publish.elf
cmd.exe /C "cd . && C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc -nostartfiles -march=rv32imc --specs=nosys.specs @CMakeFiles\subscribe_publish.elf.rsp -o subscribe_publish.elf  && cd ."
c:/espressif/tools/riscv32-esp-elf/esp-2021r2-patch2-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/port_common.c:117: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

atanisoft, 2022-04-24:
@QGB This usually happens when compiling a CPP file without declaring the `app_main` function as:
```
extern "C" void app_main()
{
...
}
```
The `extern "C"` piece is the critical piece of that in that it declares the method to use C style linkage rather than C++.

If you have further issues please provide a code example in a *NEW* issue so people can assist.

QGB, 2022-04-27:
@atanisoft  I got some compiling error in Arduino ide . old version libraries conflict with Arduino latest libraries ,how to resolve?

```
0.2\\variants\\esp32c3" "-IC:\\Users\\qgb\\Documents\\Arduino\\libraries\\TFT_eSPI" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\SPI\\src" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\FS\\src" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\SPIFFS\\src" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\Wire\\src" "-IC:\\Users\\qgb\\Documents\\Arduino\\libraries\\SdFat\\src" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\Preferences\\src" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\BluetoothSerial\\src" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\WiFi\\src" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2\\libraries\\WebServer\\src" "C:\\Users\\qgb\\AppData\\Local\\Temp\\arduino_build_507321\\sketch\\draw.cpp" -o "C:\\Users\\qgb\\AppData\\Local\\Temp\\arduino_build_507321\\sketch\\draw.cpp.o"

使用已经编译的文件：C:\Users\qgb\AppData\Local\Temp\arduino_build_507321\sketch\grbl_file.cpp.o

使用已经编译的文件：C:\Users\qgb\AppData\Local\Temp\arduino_build_507321\sketch\com.cpp.o

"C:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\riscv32-esp-elf-gcc\\gcc8_4_0-esp-2021r2/bin/riscv32-esp-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4-beta1-189-ga79dc75f0a\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/config" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/newlib/platform_include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/freertos/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/freertos/port/riscv/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/freertos/include/esp_additions" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/include/soc" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/include/soc/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3/private_include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/heap/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/log/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/lwip/include/apps" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/lwip/include/apps/sntp" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/lwip/lwip/src/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/lwip/port/esp32/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/lwip/port/esp32/include/arch" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/soc/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/soc/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/soc/esp32c3/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/hal/esp32c3/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/hal/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/hal/platform_port/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_rom/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_rom/include/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.2/tools/sdk/esp32c3/include/esp_rom/esp32c [... cut ...]
使用 2.0.0  版本的库 Preferences 在文件夹： C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Preferences 

使用 2.0.0  版本的库 BluetoothSerial 在文件夹： C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\BluetoothSerial 

使用 2.0.0  版本的库 WiFi 在文件夹： C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi 

使用 2.0.0  版本的库 WebServer 在文件夹： C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WebServer 

exit status 1

no matching function for call to 'File32::openNext(SdBaseFile*)'

```

atanisoft, 2022-04-27:
@QGB Your issue doesn't seem related to this one, please file a new issue and perhaps someone will be able to help you.

menox1981, 2022-04-27:
How in God's name did I get involved in this? I'm receiving emails. Unfortunately, I'm not a developer on this project..

BrugerX, 2023-03-17:
> How in God's name did I get involved in this? I'm receiving emails. Unfortunately, I'm not a developer on this project..

Because you're swag money baby, you deserve it [U+1F970](https://unicode.org/emoji/charts/full-emoji-list.html#1f970) [U+1F60D](https://unicode.org/emoji/charts/full-emoji-list.html#1f60d)