espressif/arduino-esp32 #1841: MPR121 Not Found

J-Arnold, 2018-09-05:
### Hardware:
Board:							NodeMCU32s
Core Installation/update date:			esp32 1.0
IDE name:							Arduino IDE 1.8.5 
Flash Frequency:					80Mhz?
Upload Speed:						921600

### Description:
I try to use a MPR121 with ESP32 by using Adafruit MPR121 and always not connecting.
I tryed on different ESP32 and I tryed the MPR121 on a ESP8266.
For better testing I extrakted the code untill the first error fron the library and tested it on ESP32 and ESP8266

### Sketch:
#include <Wire.h>
#define MPR121_CONFIG2 0x5D
#define MPR121_SOFTRESET 0x80
#define MPR121_ECR 0x5E

/*
 * ESP32
 * GPIO22 <-> SCL
 * GPIO21 <-> SDA
 * 
 * esp8266 NodeMCU
 * 
 * D1(GPIO5) <-> SCL
 * D2(GPIO4) <-> SDA
 */

int8_t _i2caddr;

uint8_t readRegister8(uint8_t reg) {
    Wire.beginTransmission(_i2caddr);
    Wire.write(reg);
    Wire.endTransmission(false);
    while (Wire.requestFrom(_i2caddr, 1) != 1);
    return ( Wire.read());
}

void writeRegister(uint8_t reg, uint8_t value) {
    Wire.beginTransmission(_i2caddr);
    Wire.write((uint8_t)reg);
    Wire.write((uint8_t)(value));
    Wire.endTransmission();
}
void Test()
{
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  
  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16){
        Serial.print("0");
      }
      Serial.print(address,HEX);
      Serial.println("  !");
      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
//      else
//    Serial.println("done\n");
}

void setup() {
  Serial.begin(115200);
  Wire.begin();

 Test();
 
  _i2caddr = 0x5A;

  // soft reset
  writeRegister(MPR121_SOFTRESET, 0x63);
  delay(1);
  
  for (uint8_t i=0; i<0x7F; i++) {
    Serial.print("$"); Serial.print(i, HEX);
    Serial.print(": 0x"); Serial.println(readRegister8(i));
  }

  writeRegister(MPR121_ECR, 0x0);

  uint8_t c = readRegister8(MPR121_CONFIG2);
  Serial.print("readRegister8(MPR121_CONFIG2); = 0x");
  Serial.println(c, HEX);
  if (c != 0x24){
    Serial.print("MPR121 0x"); Serial.print(_i2caddr, HEX); Serial.println(" not found, check wiring");
  } else {
    Serial.print("MPR121 0x"); Serial.print(_i2caddr, HEX); Serial.println(" found");
  }

}

void loop() {
  // put your main code here, to run repeatedly:

}

### Debug Messages:
Output on ESP32
Scanning...
I2C device found at address 0x5A  !
$0: 0x0
$1: 0x0
$2: 0x0
$3: 0x0
$4: 0x0
$5: 0x0
$6: 0x0
$7: 0x0
$8: 0x0
$9: 0x0
$A: 0x0
$B: 0x0
$C: 0x0
$D: 0x0
$E: 0x0
$F: 0x0
$10: 0x0
$11: 0x0
$12: 0x0
$13: 0x0
$14: 0x0
$15: 0x0
$16: 0x0
$17: 0x0
$18: 0x0
$19: 0x0
$1A: 0x0
$1B: 0x0
$1C: 0x0
$1D: 0x0
$1E: 0x0
$1F: 0x0
$20: 0x0
$21: 0x0
$22: 0x0
$23: 0x0
$24: 0x0
$25: 0x0
$26: 0x0
$27: 0x0
$28: 0x0
$29: 0x0
$2A: 0x0
$2B: 0x0
$2C: 0x0
$2D: 0x0
$2E: 0x0
$2F: 0x0
$30: 0x0
$31: 0x0
$32: 0x0
$33: 0x0
$34: 0x0
$35: 0x0
$36: 0x0
$37: 0x0
$38: 0x0
$39: 0x0
$3A: 0x0
$3B: 0x0
$3C: 0x0
$3D: 0x0
$3E: 0x0
$3F: 0x0
$40: 0x0
$41: 0x0
$42: 0x0
$43: 0x0
$44: 0x0
$45: 0x0
$46: 0x0
$47: 0x0
$48: 0x0
$49: 0x0
$4A: 0x0
$4B: 0x0
$4C: 0x0
$4D: 0x0
$4E: 0x0
$4F: 0x0
$50: 0x0
$51: 0x0
$52: 0x0
$53: 0x0
$54: 0x0
$55: 0x0
$56: 0x0
$57: 0x0
$58: 0x0
$59: 0x0
$5A: 0x0
$5B: 0x0
$5C: 0x0
$5D: 0x0
$5E: 0x0
$5F: 0x0
$60: 0x0
$61: 0x0
$62: 0x0
$63: 0x0
$64: 0x0
$65: 0x0
$66: 0x0
$67: 0x0
$68: 0x0
$69: 0x0
$6A: 0x0
$6B: 0x0
$6C: 0x0
$6D: 0x0
$6E: 0x0
$6F: 0x0
$70: 0x0
$71: 0x0
$72: 0x0
$73: 0x0
$74: 0x0
$75: 0x0
$76: 0x0
$77: 0x0
$78: 0x0
$79: 0x0
$7A: 0x0
$7B: 0x0
$7C: 0x0
$7D: 0x0
$7E: 0x0
readRegister8(MPR121_CONFIG2); = 0x0
MPR121 0x5A not found, check wiring

Ouput from ESP8266
Scanning...
I2C device found at address 0x5A  !
$0: 0x0
$1: 0x0
$2: 0x0
$3: 0x0
$4: 0x0
$5: 0x0
$6: 0x0
$7: 0x0
$8: 0x0
$9: 0x0
$A: 0x0
$B: 0x0
$C: 0x0
$D: 0x0
$E: 0x0
$F: 0x0
$10: 0x0
$11: 0x0
$12: 0x0
$13: 0x0
$14: 0x0
$15: 0x0
$16: 0x0
$17: 0x0
$18: 0x0
$19: 0x0
$1A: 0x0
$1B: 0x0
$1C: 0x0
$1D: 0x0
$1E: 0x0
$1F: 0x0
$20: 0x0
$21: 0x0
$22: 0x0
$23: 0x0
$24: 0x0
$25: 0x0
$26: 0x0
$27: 0x0
$28: 0x0
$29: 0x0
$2A: 0x0
$2B: 0x0
$2C: 0x0
$2D: 0x0
$2E: 0x0
$2F: 0x0
$30: 0x0
$31: 0x0
$32: 0x0
$33: 0x0
$34: 0x0
$35: 0x0
$36: 0x0
$37: 0x0
$38: 0x0
$39: 0x0
$3A: 0x0
$3B: 0x0
$3C: 0x0
$3D: 0x0
$3E: 0x0
$3F: 0x0
$40: 0x0
$41: 0x0
$42: 0x0
$43: 0x0
$44: 0x0
$45: 0x0
$46: 0x0
$47: 0x0
$48: 0x0
$49: 0x0
$4A: 0x0
$4B: 0x0
$4C: 0x0
$4D: 0x0
$4E: 0x0
$4F: 0x0
$50: 0x0
$51: 0x0
$52: 0x0
$53: 0x0
$54: 0x0
$55: 0x0
$56: 0x0
$57: 0x0
$58: 0x0
$59: 0x0
$5A: 0x0
$5B: 0x0
$5C: 0x16
$5D: 0x36
$5E: 0x0
$5F: 0x0
$60: 0x0
$61: 0x0
$62: 0x0
$63: 0x0
$64: 0x0
$65: 0x0
$66: 0x0
$67: 0x0
$68: 0x0
$69: 0x0
$6A: 0x0
$6B: 0x0
$6C: 0x0
$6D: 0x0
$6E: 0x0
$6F: 0x0
$70: 0x0
$71: 0x0
$72: 0x0
$73: 0x0
$74: 0x0
$75: 0x0
$76: 0x0
$77: 0x0
$78: 0x0
$79: 0x0
$7A: 0x0
$7B: 0x0
$7C: 0x0
$7D: 0x0
$7E: 0x0
readRegister8(MPR121_CONFIG2); = 0x24
MPR121 0x5A found

stickbreaker, 2018-09-06:
@J-Arnold the release 1.0.0 has a problem with ReSTART, it has been fixed in the github version.

You can either install the github dev branch by following these instructions:
[follow the development repository instructions](https://github.com/espressif/arduino-esp32#installation-instructions)

or, you just replace these four files with the github versions
 
>cores\esp32\esp32-hal-i2c.h
>cores\esp32\esp32-hal-i2c.c
>libraries\Wire\src\Wire.h
>libraries\Wire\src\Wire.cpp

Chuck.

fabiuz7, 2018-09-07:
I'm experiencing a similar problem with Adafruit MPR121. After few minutes the i2c gets stucked. I have connected a DS3231, a MPU6050 a Wemos Motor shield and the MPR121 to Lolin32. Disconnecting it seems to avoid the stucking.

stickbreaker, 2018-09-07:
@fabiuz7 you'll have to post more information before I can diagnose your problem.
* Set the Core Debug Level to Debug,
* in `cores\esp32\esp32-hal-i2c.c` line 45 [here](https://github.com/espressif/arduino-esp32/blob/a0f0bd930cfd2d607bf3d3288f46e2d265dd2e11/cores/esp32/esp32-hal-i2c.c#L45), uncomment this line.
* post the smallest sketch that duplicates your problem
* post the Serial monitor that shows the error messages
* describe your circuit

Chuck.

fabiuz7, 2018-09-07:
Sorry for the poor info, could you tell me how to enable Core Debug Level to Debug? I've read some docs but I can't see any log...

stickbreaker, 2018-09-07:
Under the Tools menu in Arduino there is a pulldown menu.

fabiuz7, 2018-09-07:
Sorry, I was using the Wemos D32 instead LolinD32. Now I have set the debug level. I will let you know if I found out something, thank you

J-Arnold, 2018-09-07:
Thank you for the suport after changing the 4 files all work perfect,

lbernstone, 2018-09-07:
Please close the issue if it is working for you.

closed 2018-09-09 by J-Arnold

nthao113, 2024-05-29:
HELP MEEEEEEEEEEEEEEE
I use the code below: 

#include <Wire.h>
#include "Adafruit_MPR121.h"

#ifndef _BV
#define _BV(bit) (1 << (bit)) 
#endif

// You can have up to 4 on one i2c bus but one is enough for testing!
Adafruit_MPR121 cap = Adafruit_MPR121();

// Keeps track of the last pins touched
// so we know when buttons are 'released'
uint16_t lasttouched = 0;
uint16_t currtouched = 0;

void setup() {
  Serial.begin(9600);

  while (!Serial) { // needed to keep leonardo/micro from starting too fast!
    delay(10);
  }
  
  Serial.println("Adafruit MPR121 Capacitive Touch sensor test"); 
  
  // Default address is 0x5A, if tied to 3.3V its 0x5B
  // If tied to SDA its 0x5C and if SCL then 0x5D
  if (!cap.begin(0x5A)) {
    Serial.println("MPR121 not found, check wiring?");
    while (1);
  }
  Serial.println("MPR121 found!");
}

void loop() {
  // Get the currently touched pads
  currtouched = cap.touched();
  
  for (uint8_t i=0; i<12; i++) {
    // it if *is* touched and *wasnt* touched before, alert!
    if ((currtouched & _BV(i)) && !(lasttouched & _BV(i)) ) {
      Serial.print(i); Serial.println(" touched");
    }
    // if it *was* touched and now *isnt*, alert!
    if (!(currtouched & _BV(i)) && (lasttouched & _BV(i)) ) {
      Serial.print(i); Serial.println(" released");
    }
  }

  // reset our state
  lasttouched = currtouched;

  // comment out this line for detailed data from the sensor!
  return;
  
  // debugging info, what
  Serial.print("\t\t\t\t\t\t\t\t\t\t\t\t\t 0x"); Serial.println(cap.touched(), HEX);
  Serial.print("Filt: ");
  for (uint8_t i=0; i<12; i++) {
    Serial.print(cap.filteredData(i)); Serial.print("\t");
  }
  Serial.println();
  Serial.print("Base: ");
  for (uint8_t i=0; i<12; i++) {
    Serial.print(cap.baselineData(i)); Serial.print("\t");
  }
  Serial.println();
  
  // put a delay so it isn't overwhelming
  delay(100);
}
ESP32
GPIO22 <-> SCL
GPIO21 <-> SDA

I get the following error and cannot touch and receive results as usual
#䂏��V(d���Adafruit MPR121 Capacitive Touch sensor test
MPR121 found!
0 touched
1 touched
2 touched
3 touched
4 touched
5 touched
6 touched
7 touched
8 touched
9 touched
10 touched
11 touched
A�Z���V)�!�#��Adafruit MPR121 Capacitive Touch sensor test
MPR121 found!
5 touched
0 touched
1 touched
2 touched
3 touched
4 touched
6 touched
7 touched
8 touched
9 touched
10 touched
11 touched