espressif/arduino-esp32 #1844: Location of The esp tools/configs in arduino - installing via board manager

Netoperz, 2018-09-06:
Hello,
Can't find the answer - so asking here, where the hell is the espressif folder in linux and windows, when You install the esp in arduino using https://dl is installed using mhall119 from snap shop in ubuntu.
I'm looking for boards.tt and default.csv for partition map.

MarkyAD, 2018-09-06:
In unix thats what the "find" command is for. For boards manager its ~/.arduino15/packages

Netoperz, 2018-09-06:
[image]
Not really there
And whereis and find does not show any output,

system:
n37@n37--Stellar:~$ uname -a
Linux n37--Stellar 4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
[image]

Thank You for Your support, but in my case it is in /snaps, as this version of arduino comes from snap store.

And it does not show any output when using whereis or find . That's why i couldn't find it.
But after Your response i have looked deeper and everything is there, This case might help others, Thank You.
Closing

closed 2018-09-06 by Netoperz