espressif/arduino-esp32 #1847: Preference logs not found keys as errors

trycoon, 2018-09-07:
When fetching a value from Preference.h we get a log_e() to the console if the key could not be found in preference store. https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/src/Preferences.cpp#L423

This is part of my application that wraps preference:
auto mqtt_port = Configuration::getInt("MQTT_PORT", 1883);

And it will log this message:
[E][Preferences.cpp:342] getInt(): nvs_get_i32 fail: MQTT_PORT NOT_FOUND

I feel that giving an ERROR just because the requested key is not found is not the right thing to do, most of the times the application work perfect anyway. And in my example I fall back to a default value, so the "MQTT_PORT" is only if the users want to override the default port.

I think DEBUG would be a more suitable log level for this kind of errors. Now my console is cluttered with alot of these false errors, possible hiding the real ones.

closed 2019-08-15 by stale[bot]

thehellmaker, 2020-01-01:
Can this be changed to info or error logging. It is bugging me as well?

hpsaturn, 2020-05-04:
@thehellmaker I agree, this "error" maybe should be a simple warning, or maybe we should have a option of set a "debug" mode flag?

lbernstone, 2020-05-04:
It is an error in the underlying function, so it should be reflected as an error in arduino.  How about a check_key method that you can use as a boundary check?

CelliesProjects, 2020-05-04:
I would like something like 
```c++
preferences.getUint(keyValue, defaultValue, NVS_NF_IGNORE);
```
Where `NVS_NF_IGNORE` is a bool and defaults to FALSE. 
This would be least disruptive.

lbernstone, 2020-05-04:
There is only one way to check if a key exists, and that is to try to retrieve it.  This causes an underlying error in nvs if there is no key.  The checkKey method will put the onus on the programmer if you don't want this error:
```
if (prefs.checkKey("testkey")) {
   prefs.putString("testkey", "default value");
   value = "default value";
} else {
   value = prefs.getString("testkey");
}
```

hpsaturn, 2020-09-23:
Another alternative for this issue or for reducing a lot of error messages, is set the `CORE_DEBUG_LEVEL` to 0 for production for example, in `platformio.ini` or ESP32 compiling you need set it something like that:

```python
build_flags =
    -D CORE_DEBUG_LEVEL=0
```

brightproject, 2023-07-05:
@espressif  I have same error also
`[V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: Brightness NOT_FOUND`
Code
```
  preferences.begin ("Test-app");
  Brightness = preferences.getUInt ("Brightness", 1023);
  preferences.end ();
```

hpsaturn, 2023-07-05:
it is not a error, is like a warning, because this preference wasn't initialized. After the first set, should disappear.

Xylopyrographer, 2023-07-05:
Check out the official docs:

https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/preferences.html￼

https://docs.espressif.com/projects/arduino-esp32/en/latest/api/preferences.html

brightproject, 2023-07-06:
> it is not a error, is like a warning, because this preference wasn't initialized. After the first set, should disappear.

No, this warning is present all the time, as soon as I access the HBC store for the data that is stored in it, I get a string of messages about the absence of these variables.

brightproject, 2023-07-06:
> also when you re-compiling with `CODE_DEBUG_LEVEL=0` ?

I specifically included the verbose log tag to debug the device
`DebugLevel=verbose`
and you advise me to disable it.
I have a fairly old ESP32 chip, one of the first series.
And the platform version is new `2.0.9`, and probably in the first chips the work with the NBC was not sufficiently debugged, so such errors happen.
Otherwise I can't explain.