espressif/arduino-esp32 #1848: Arduino as component: Cannot exclude Arduino library without excluding added libraries

RA5040, 2018-09-07:
Core Installation/update date:			Current
IDE name:						IDF component: Arduino

Referring to: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

It isn't possible to exclude an Arduino library (menuconfig/Arduino Configuration/Include only specific Arduino libraries) without at the same time excluding user-added libraries.  This has come to light with the current need to exclude the SD_MMC library as this has a compilation error.  By selecting the option I also exclude the U8g2 library that I have added to the /components/arduino/libraries.

I would like to suggest that: 
- the option 'Include only specific Arduino libraries' should be changed to 'Exclude specific Arduino libraries'
- an additional option be added: 'Include specified third-party/user libraries' (or that the documentation should be changed to explain how to add third-party libraries)

better-adapted, 2018-09-14:
I bodged my way through adding a 3rd party library  (MQTT PubSubClient ) buy letting ALL Arduino library's be compiled - but deleting the SD_MMC directory first. so everything in the libs folder is getting compiled - i did not try too to edit component.mk

lbernstone, 2018-09-14:
OP is probably correct.  Exclude rather than include seems a better way to manage this. Workaround: just add the one line needed to fix SD_MMC. https://github.com/espressif/arduino-esp32/pull/1836

closed 2019-08-15 by stale[bot]