espressif/arduino-esp32 #1850: ESP32 Wifi and ESP8266 Wifi libraries don't provide the same response from an HTTP request

theDontKnowGuy, 2018-09-09:
Hardware: ESP32
Board:	ESP32 Dev Module 
Core Installation/update date: Sep 9, 2018
IDE name:	Arduino IDE
Flash Frequency:	40/80 (happens in both)
Upload Speed: 921600

### Description:
Has my little project which, at start, connect to a local home RPi apache webserver to get a JSON file which contains configuration for the app. 
Worked on ESP8266 (Wemos D1R2) without issues.

Migrated to ESP32. 
moved from using ESP8266WiFi.h and WiFiClientSecure.h .  to: ESP32's WiFiClientSecure.h.
some minor updates as well (8266 libs can take String for host name, ESP32 can't...) nothing major.

The few specific code lines in focus are the usual:

`  while (client.connected()) {
    String line = client.readStringUntil('\n');
      Serial.println(line);
      line = client.readStringUntil('\r');
        Serial.println(line);
      break;
    if (line == "") {
      if (DEBUGLEVEL > 2) Serial.println("headers received");
      break;
    }
  }
  const size_t capacity = JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(2) + 60;
  DynamicJsonDocument doc(capacity);
`
The full code (before migration) is here: https://github.com/theDontKnowGuy/myRFDevicesHub

Yet, all HTTP requests aren't processed as before (hence fail to be processed).
All works except of the Wifi response is not identical to the response in the Wifi8266 library used. 
it has something to do i think with getting \r\n differently although I couldn't nail it.

Headers are "done" after the first line (HTTP 1.1 200 OK).

Any Ideas?

Sauro98, 2018-09-13:
readStringUntil is inclusive, so you should read until `\r` instead of `\n`. Check out [this example](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino#L61)

Sauro98, 2018-09-13:
Here's what I use:
```c++
String currentLine, response;
while (client.connected()) {
    if (client.available()) {
        currentLine = client.readStringUntil('\r');
        response += currentLine;
        response += client.read(); // the \n
        if (currentLine.length() == 0) {
            // Here the full response was read
        } else {
            Serial.printf("(%d) ", currentLine.length());
            Serial.println(currentLine);
            currentLine = "";
         }   
    }
}
```

closed 2019-08-15 by stale[bot]