espressif/arduino-esp32 #1858: is there an example for using pppos with gsm?

jhonoryza, 2018-09-12:
there is example, but in esp-idf sdk
https://github.com/loboris/ESP32-PPPOS-EXAMPLE

can u provide pppos example for arduino ide usage?
i am using esp32 with sim800l

xvinny, 2019-06-28:
You can use [https://github.com/vshymanskyy/TinyGSM](url)

vshymanskyy, 2019-07-08:
As an author of TinyGSM, thank you for mentioning it here ;)
However PPPoS library is conceptually/technically better way to work with Cellular modems.
I would be very happy to see an example of PPPoS library usage with Arduino.

vshymanskyy, 2019-07-08:
Especially interested in this since `TTGO T-Call` board is available:
https://github.com/Xinyuan-LilyGO/TTGO-T-Call

vshymanskyy, 2019-09-06:
Hey bot, do not close this please... ;)

xvinny, 2019-09-18:
@vshymanskyy some new updates was risen today for ESP-IDF (pppos) and Arduino-ESP32 (Http Client).. I believe that these updates will help the evolution of TinyGSM :-)

closed 2019-12-01 by stale[bot]

vshymanskyy, 2021-06-29:
Here's a working example of HTTP/HTTPS clients working on top of PPPoS:
[pppos_client.zip](https://github.com/espressif/arduino-esp32/files/6736140/pppos_client.zip)

Ruzk-innovation, 2021-07-05:
Hi @vshymanskyy, thank you very much for your example. I would like to be able to ask you a question regarding the operation, since it does not work for me... I am using an ESP32 Devkit v4 and a SIM800L module, in which I made some modifications to the code to change the TX, RX and RST pins. The problem I have is that nothing happens when I write the "ppp", "htpp or https" and "Stop" commands through the serial port. In fact he finally answers me with a "NO CARRIER". Is it possible that you can help me please? Here I show what appears on the serial monitor, thank you very much in advance.

- Start GSM Communication: AT
- Configure GSM APN: AT+CGDCONT=1,"IP","bam.clarochile.cl"
- Set GSM to PPP Mode: ATD*99***1#\n or AT+CGDATA="PPP",1
- Set ESP32 PPP: ppp
- Try to download file: http, https
- Stop PPP Mode and return to AT command: stop

AT

OK

AT+CGDCONT=1,"IP","bam.clarochile.cl" 

OK

AT+CGDATA="PPP",1 

ERROR

ATD*99***1#

CONNECT

~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
ppp

~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
http

~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~

~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
ppp

~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~
~⸮}#⸮!}!}!} }2}"}&} }*} } }#}$⸮#}'}"}(}"U⸮~

NO CARRIER

vshymanskyy, 2021-07-06:
The log looks good, but after `ppp` command it should pick up the connection.
You can also try this (I created a library based on the same example): https://github.com/vshymanskyy/ESP32_PPPoS

Ruzk-innovation, 2021-07-07:
Hi @vshymanskyy, I wanted to tell you that with that library you sent me it works perfect. Thank you very much for your help. Greetings

Ruzk-innovation, 2021-07-10:
Hi @vshymanskyy, by chance is it possible to use this library through a 3G or 4G connection?  for example using a SIM7600 ?.  Thank you so much

vshymanskyy, 2021-07-10:
why not?

Ruzk-innovation, 2021-07-11:
Hi @vshymanskyy, sorry, how can I get the IP address to which it connects through this library?

vshymanskyy, 2021-07-11:
sorry, you'll have to figure it out. PRs are welcome!

lbernstone, 2021-07-11:
> Hi @vshymanskyy, sorry, how can I get the IP address to which it connects through this library?

I'm sure your device has some AT command for that, but rather than relying on something proprietary, you could http query a whatismyip or geolocation site and parse the results.

kieennt13, 2024-11-06:
Hi @vshymanskyy ,
Can TinyGSM helps create a PPP interface in order to establish a LTE based mesh network? 
I was building a no router mesh network to handle  Wifi disconnected scenario.
I've connected to LTE through modem.waitForNetwork(), modem.gprsConnect() and both modem.isNetworkConnected() && modem.isGprsConnected() return true, but mesh still in the network disnocnnected state.
I've been stucked in this point for weeks so I will greatly appriciated if you could help me figured out this problem!

me-no-dev, 2024-11-06:
Starting with v3.0.0, we have implemented PPP library that connects to GSM modems and is fully integrated with the network stack

me-no-dev, 2024-11-06:
here is the example: https://github.com/espressif/arduino-esp32/blob/master/libraries/PPP/examples/PPP_Basic/PPP_Basic.ino

kieennt13, 2024-11-06:
Thanks for the example!
But I could see it totally different from the way I did before (with modem.waitForNetwork() and modem.gprsConnect()).
The problem is I shall not to change the LTE connection way.
Is there any way to keeps connecting LTE by those modem functions but still create a PPP interface?