espressif/arduino-esp32 #1859: Core crahing when boots, trying to use Ledc for Pwm

esperancaleonardo, 2018-09-12:
### Hardware:
Board:							Lolin 32 Lite
Core Installation/update date:			5/Sep/018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Two months ago, i had tested a lec Pwm for motor control and it worked fine. Then, after some time, I had to reinstall the Core in my laptop, now, everything works fine, less pwm. The esp32 cpu crashes and reboots when I try do create an object using a class the uses ledc. It appears to be on AttachPin, but i have no idea how to solve that.

### Sketch:
```cpp

#include <Arduino.h>
#include "Motors.h"

Motor motorA = Motor(INA_1, INA_2, PWM_A);  // it crashes in this initialization.

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: en:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d3620  PS      : 0x00060330  A0      : 0x800d1505  A1      : 0x3ffe3bb0  
A2      : 0x3ffc253c  A3      : 0x00000008  A4      : 0x00000007  A5      : 0x00000006  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0d3a  A9      : 0x3ffe3b90  
A10     : 0x00000006  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xfffffffa  

Backtrace: 0x400d3620:0x3ffe3bb0 0x400d1502:0x3ffe3bd0 0x400d6e2f:0x3ffe3bf0 0x40082ffc:0x3ffe3c10 0x400831c9:0x3ffe3c40 0x4007892f:0x3ffe3c60 0x40078995:0x3ffe3c90 0x400789a0:0x3ffe3cc0 0x40078b33:0x3ffe3ce0 0x40080309:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...

```
```
PC: 0x400d3620: ledcAttachPin at /home/leonardo/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-ledc.c line 230
EXCVADDR: 0x00000000

Decoding stack results
0x400d3620: ledcAttachPin at /home/leonardo/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-ledc.c line 230
0x400d1502: _GLOBAL__sub_I__ZN7MPU605013dmpInitializeEv() at /home/leonardo/Documents/Reps/verysmall-firmware/src/principal_esp/principal_esp.ino line 20
0x400d6e2f: do_global_ctors at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 430
0x40082ffc: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 343
0x400831c9: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 213

```
This is the lib i'm using:
[Motors.zip](https://github.com/espressif/arduino-esp32/files/2375213/Motors.zip)

lbernstone, 2018-09-12:
Do ledcSetup before you attach a pin. Your minimal sketch calls variables that don't exist, so it is impossible to actually reproduce your problem.

esperancaleonardo, 2018-09-12:
> Do ledcSetup before you attach a pin. Your minimal sketch calls variables that don't exist, so it is impossible to actually reproduce your problem.

yeah, sorry, INA_1, INA_2 e PWMA are defined in a .h file.

Actually i have found the problem, that whas about types, ledcAttachPin uses uint8_t and my PWM_A isn't, how can i cast that?

lbernstone, 2018-09-12:
This forum is not for programming advice.  Google is very good for that.

closed 2019-08-15 by stale[bot]