espressif/arduino-esp32 #1861: WiFiClientSecure GoDaddy SSL image upload - (004E),  MbedTLS -78

flashgordonmjg, 2018-09-13:
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation/update date:			09/Sep/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
I can successfully upload an image using the WiFiClientSecure with a cert when I set Core Debug Level: Verbose.  When I change the Core Debug Level to Debug, I always get UNKNOWN ERROR CODE (004E) , MbedTLS -78.  It is interesting that all I have to do is to change the Core Debug Level and I get different behavior.  I have worked on this for several days and I am not sure how to further debug. 

### Sketch:
```cpp
WiFiClientSecure secureClient;

void uploadImage(ArduCAM myCAM)
{   

    int total_time = 0;
    total_time = millis();

    Serial.println("start: uploadCapturedImage()");
    time_t tnow = time(nullptr);
    String fileName = "ESP_380460-" + String(tnow) + String(".jpg");

    Serial.println("Filename: " + fileName);

    secureClient.setCACert(cert_root_ca);
    if (secureClient.connect("mydomain.com",443))
    {
        Serial.println("Connected to Server");

        secureClient.setNoDelay(true);

        size_t len = myCAM.read_fifo_length();
        secureClient.println("POST /uploadfiles/PROTO1 HTTP/1.1");
        Serial.println("POST /uploadfiles/PROTO1 HTTP/1.1");
        
        secureClient.println("Host: mydomain.com");
        Serial.println("Host: mydomain.com");

        secureClient.println("Connection: keep-alive");
        Serial.println("Connection: keep-alive");

        secureClient.println("Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryjg2qVIUS8teOAbN3");
        Serial.println("Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryjg2qVIUS8teOAbN3");

        secureClient.print("Content-Length: ");
        secureClient.println(len+191+fileName.length());  
        
        Serial.print("Content-Length: ");
        Serial.println(len+191+fileName.length());  

        secureClient.println();
        Serial.println();

        secureClient.println("------WebKitFormBoundaryjg2qVIUS8teOAbN3");
        Serial.println("------WebKitFormBoundaryjg2qVIUS8teOAbN3");

        secureClient.println("Content-Disposition: form-data; name=\"files\"; filename=\"" + fileName + "\"");
        Serial.println("Content-Disposition: form-data; name=\"files\"; filename=\"" + fileName + "\"");

        secureClient.println("Content-Type: application/octet-stream");
        Serial.println("Content-Type: application/octet-stream");

        secureClient.println();
        Serial.println();
   
        if (len >= 0x07ffff) {
            Serial.println("Over size.");
            return;
[... 141 lines cut ...]
[V][ssl_client.cpp:242] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:242] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:242] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:242] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:242] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:242] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:242] send_ssl_data(): Writing HTTP request...
Done
In eSRcv()
HTTP/1.1 200 OK
Transfer-Encoding: chunked
Content-Type: application/json; charset=utf-8
Server: Kestrel
Request-Context: appId=cid-v1:bdd2acc6-c5e3-49a2-a18e-2f129779db62
X-Powered-By: ASP.NET
Set-Cookie: ARRAffinity=81f2e5aa377fe9c0b219583b51cdb36e6cde71e63e64a15d047622753d7e8a72;Path=/;HttpOnly;Domain=mydomain.com
Date: Thu, 13 Sep 2018 01:02:48 GMT

26
{"count":1,"size":91584,"filePath":""}
0

upload blob total_time used (in miliseconds):5052

Thank you.

stickbreaker, 2018-09-13:
@flashgordonmjg When VERBOSE is logging all those messages, the Serial object is causing your task execution to be delayed. The output is 'pacing', slowing down the through put.  

as a test for my hypothesis add a simple delay to this `while()`:
```c++
    while (len) 
        {
            size_t will_copy = (len < bufferSize) ? len : bufferSize;
            SPI.transferBytes(&buffer[0], &buffer[0], will_copy);
            if (!secureClient.connected()) break;
            secureClient.write(&buffer[0], will_copy);
            len -= will_copy;
        }
```
Maybe a `delay(50)` 

See if that changes the behavior. 

Another question, why do you declare your transfer buffer this way?
```c++
        static const size_t buffSize = 4096;
        static uint8_t buffer[buffSize] = {0xFF};
```
The static attribute creates a global allocation but, you can only use it in this scope? So you have permanently consumed 4096+size_t(4bytes) of RAM that is only used during this `while()`?

Chuck.

flashgordonmjg, 2018-09-13:
Thank you @stickbreaker  for your suggestions. When I went to implement the delay, i noticed that the code was never executed. So after reviewing my initial code above, i noticed the client is being disconnected at the first 
```cpp
if (!secureClient.connected()) return;
```

I decided to create a new endpoint and a simplified project.  When I run this in Verbose, it works and successfully disconnected.  When I run in Debug, i get the same error as above in my last example.

Thank you for your help.

```cpp
#include <WiFiClientSecure.h>
#include "time.h"
#include <WiFi.h>

const char* cert_root_azure = 
"-----BEGIN CERTIFICATE-----\n" \
"MIIDdzCCAl+gAwIBAgIEAgAAuTANBgkqhkiG9w0BAQUFADBaMQswCQYDVQQGEwJJ\n" \
"RTESMBAGA1UEChMJQmFsdGltb3JlMRMwEQYDVQQLEwpDeWJlclRydXN0MSIwIAYD\n" \
"VQQDExlCYWx0aW1vcmUgQ3liZXJUcnVzdCBSb290MB4XDTAwMDUxMjE4NDYwMFoX\n" \
"DTI1MDUxMjIzNTkwMFowWjELMAkGA1UEBhMCSUUxEjAQBgNVBAoTCUJhbHRpbW9y\n" \
"ZTETMBEGA1UECxMKQ3liZXJUcnVzdDEiMCAGA1UEAxMZQmFsdGltb3JlIEN5YmVy\n" \
"VHJ1c3QgUm9vdDCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAKMEuyKr\n" \
"mD1X6CZymrV51Cni4eiVgLGw41uOKymaZN+hXe2wCQVt2yguzmKiYv60iNoS6zjr\n" \
"IZ3AQSsBUnuId9Mcj8e6uYi1agnnc+gRQKfRzMpijS3ljwumUNKoUMMo6vWrJYeK\n" \
"mpYcqWe4PwzV9/lSEy/CG9VwcPCPwBLKBsua4dnKM3p31vjsufFoREJIE9LAwqSu\n" \
"XmD+tqYF/LTdB1kC1FkYmGP1pWPgkAx9XbIGevOF6uvUA65ehD5f/xXtabz5OTZy\n" \
"dc93Uk3zyZAsuT3lySNTPx8kmCFcB5kpvcY67Oduhjprl3RjM71oGDHweI12v/ye\n" \
"jl0qhqdNkNwnGjkCAwEAAaNFMEMwHQYDVR0OBBYEFOWdWTCCR1jMrPoIVDaGezq1\n" \
"BE3wMBIGA1UdEwEB/wQIMAYBAf8CAQMwDgYDVR0PAQH/BAQDAgEGMA0GCSqGSIb3\n" \
"DQEBBQUAA4IBAQCFDF2O5G9RaEIFoN27TyclhAO992T9Ldcw46QQF+vaKSm2eT92\n" \
"9hkTI7gQCvlYpNRhcL0EYWoSihfVCr3FvDB81ukMJY2GQE/szKN+OMY3EU/t3Wgx\n" \
"jkzSswF07r51XgdIGn9w/xZchMB5hbgF/X++ZRGjD8ACtPhSNzkE1akxehi/oCr0\n" \
"Epn3o0WC4zxe9Z2etciefC7IpJ5OCBRLbf1wbWsaY71k5h+3zvDyny67G7fyUIhz\n" \
"ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS\n" \
"R9I4LtD+gdwyah617jzV/OeBHRnDJELqYzmp\n" \
"-----END CERTIFICATE-----\n";

//Station mode
const char *ssid = "_ssid_"; 
const char *password = "__password__";

WiFiClientSecure secureClient;

void setup()
{
    Serial.begin(115200);

    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) 
    {
      delay(500);
      Serial.print(".");
    }

    // Synchronize time using SNTP. This is necessary to verify that
    // the TLS certificates offered by the server are currently valid.
    Serial.print("Setting time using SNTP");
    configTime(8 * 3600, 0, "pool.ntp.org", "time.nist.gov");
    time_t now = time(nullptr);
    while (now < 8 * 3600 * 2) 
    {
      delay(500);
      Serial.print(".");
      now = time(nullptr);
    }
    Serial.println("");
    struct tm timeinfo;
    gmtime_r(&now, &timeinfo);
    Serial.print("Current time: ");
    Serial.print(asctime(&timeinfo));
    
    uploadSimpleSecureAzure();
}

void loop()
{

}

void uploadSimpleSecureAzure()
{   

    Serial.println("start: uploadSimpleSecure()");
 
    String fileName = "ESP_380460" + String(".jpg");
    Serial.println("Filename: " + fileName);

    secureClient.setCACert(cert_root_azure);
    if (secureClient.connect("espssltest.azurewebsites.net",443))
    {
        Serial.println("Connected to Server");

        size_t len = 1100;
        secureClient.println("POST /uploadfiles/PROTO1 HTTP/1.1");
        Serial.println("POST /uploadfiles/PROTO1 HTTP/1.1");
        
        secureClient.println("Host: espssltest.azurewebsites.net");
        Serial.println("Host: espssltest.azurewebsites.net");

        secureClient.println("Connection: keep-alive");
        Serial.println("Connection: keep-alive");

        secureClient.println("Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryjg2qVIUS8teOAbN3");
        Serial.println("Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryjg2qVIUS8teOAbN3");

        secureClient.print("Content-Length: ");
        secureClient.println(len+191+fileName.length());  
  
 
        Serial.print("Content-Length: ");
        Serial.println(len+191+fileName.length());   

        secureClient.println();
        Serial.println();

        secureClient.println("------WebKitFormBoundaryjg2qVIUS8teOAbN3");
        Serial.println("------WebKitFormBoundaryjg2qVIUS8teOAbN3");

        secureClient.println("Content-Disposition: form-data; name=\"files\"; filename=\"" + fileName + "\"");
        Serial.println("Content-Disposition: form-data; name=\"files\"; filename=\"" + fileName + "\"");

        secureClient.println("Content-Type: application/octet-stream");
        Serial.println("Content-Type: application/octet-stream");

        secureClient.println();
        Serial.println();

        if (!secureClient.connected()) 
        {
          Serial.println("secureClient not connected");
          return;
        }
        else
        {
          Serial.println("secureClient connected");
          secureClient.stop();
        }
    }
}
```

Mark

flashgordonmjg, 2018-09-13:
Hi @stickbreaker , FYI - I have been able to get past the MBedTLS -78 error.  I noticed that when i changed secureClient.println to secureClient.print, it would connect in Debug mode. I don't understand why there is a difference though between Debug and Verbose.  

FROM
```cpp
        secureClient.println("Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryjg2qVIUS8teOAbN3");
```
TO
```cpp
        secureClient.print("Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryjg2qVIUS8teOAbN3\r\n");
```

I changed all of my secureClient.println to secureClient.print and used \r\n.

Thank you, Mark.

stickbreaker, 2018-09-13:
It could be that `println()` is using the standard UNIX `'\n'` as a line termination. 

A couples Questions about your coding choices:
*  When you coded the `content-length:` attribute are you counting all of the `MIME 1.0` overhead?
* Why are you recreating the boundry marker multiple times? I would just define it as a char[] and reuse it
```c++
        secureClient.println("Connection: keep-alive");
        Serial.println("Connection: keep-alive");

        char boundary[]="----WebKitFormBoundaryjg2qVIUS8teOAbN3";

        #define LINEBUFFLENGTH 128
        char lineBuff[LINEBUFFLENGTH];
        snprintf(lineBuf,LINEBUFFLENGTH,"Content-Type: multipart/form-data; boundary=%s\r\n",boundary);

        secureClient.print(lineBuff);
        Serial.print(lineBuff);

        snprintf(lineBuff,LINEBUFFLENGTH,"Content-Length: %d\r\n",(len+191+fileName.length());  
        secureClient.print(lineBuff);
        Serial.print(lineBuff);

        secureClient.print("\r\n");
        Serial.print("\r\n");

        secureClient.printf("--%s\r\n",boundary);
        Serial.printf("--%s\r\n",boundary);
```
Every time I have to have multiple copies of the exactly the same text, I usually miss-copy one of them. :grimacing:  

Chuck.

closed 2019-08-15 by stale[bot]