espressif/arduino-esp32 #1862: BLE and power consumption

sensboston, 2018-09-13:
Hello. I'm looking for advices how to achieve a really low-power consumption mode with enabled BLE advertising for ESP32.

I like ESP32 and this project very much, my SOCs (different models) are working exact as supposed to be, right out of box. The only one real issue left - it's a missing "standby" mode with BLE enabled. Currently, my ESP32 "eats" 4 AA batteries overnight.

Any suggestions and advices are very appreciated!

P.S. I ordered Adafruit Feather nRF52 dev board (according to my research this chip can do a low-power consumption with BLE) but definitely prefer to stay with ESP32.

lbernstone, 2018-09-13:
This forum is for issues with the arduino-esp32 codebase.  Please take general discussion to https://www.esp32.com

sensboston, 2018-09-13:
I'm using arduino-esp32 now and would like to know what can I do for power consumption with arduino-esp32.

closed 2019-08-15 by stale[bot]